The-OpenROAD-Project/OpenLane #1286: Magic Antenna Checks Fails with segmentation violation

dineshannayya, 2022-08-26:
### Description
My One of the submacro  openlane Magic Antenna Checks Fails with segmentation violation

### Environment
<pre>python3 ./env.py issue-survey
Kernel: Linux v5.15.0-46-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: 6ab944bc23688cae6dc6fa32444891a1e57715c8
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

6ab944b 2022-08-09T14:51:35+02:00 IR Drop Reports in the Signoff Stage (#1255) - Mohamed Gaber -  (HEAD -&gt; master)
4476a58 2022-07-29T16:03:10+02:00 Sky130 Enhancements (#1240) - Mohamed Gaber -  ()
4f63c8c 2022-07-27T17:26:27+02:00 Various CI Fixes (#1235) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/efabless/OpenLane.git (fetch)
origin	https://github.com/efabless/OpenLane.git (push)
</pre>

### Reproduction Material

[pinmux.issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9429835/pinmux.issue_reproducible.tar.gz)

### Expected behavior

### Logs
<pre>[INFO]: Running Magic DRC (log: pinmux_top/runs/pinmux_top/logs/signoff/41-drc.log)...
[INFO]: Converting Magic DRC Violations to Magic Readable Format...
[INFO]: Converting Magic DRC Violations to Klayout XML Database...
[INFO]: No DRC violations after GDS streaming out.
[STEP 42]
[INFO]: Running Magic Antenna Checks (log: pinmux_top/runs/pinmux_top/logs/signoff/42-antenna.log)...
[ERROR]: during executing magic script /openlane/scripts/magic/antenna_check.tcl
[ERROR]: Log: pinmux_top/runs/pinmux_top/logs/signoff/42-antenna.log
[ERROR]: Last 10 lines:
   7700 gates analyzed.
   7800 gates analyzed.
   7900 gates analyzed.
   8000 gates analyzed.
   8100 gates analyzed.
   8200 gates analyzed.
   8300 gates analyzed.
kill: no such node _11568_/A1
output: no such node _11568_/A1
child killed: segmentation violation

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager
</pre>

vijayank88, 2022-08-26:
@dineshannayya 
Is that `reproducible` tested locally? Is generating same log with the test case?

dineshannayya, 2022-08-26:
I don't know how to validate the reproducible locally . Design wise, I have done good amount of changes in this Pinmux block

The flow gone through fine with
set ::env(USE_ARC_ANTENNA_CHECK) "1"

vijayank88, 2022-08-26:
Just get into `reproducible` directory and execute `./run.sh` it should create same segmentation violation.
I ran the test case at my end, getting different logs.

dineshannayya, 2022-08-26:
When run locally, I am getting different error, But still see a segmentation fault.

<pre>Cell sky130_fd_sc_hd__mux2_1 couldn&apos;t be read
Cell sky130_fd_sc_hd__mux2_1 couldn&apos;t be read
Cell sky130_fd_sc_hd__mux2_1 couldn&apos;t be read
Cell sky130_fd_sc_hd__mux2_1 couldn&apos;t be read
Cell sky130_fd_sc_hd__mux2_1 couldn&apos;t be read
Cell sky130_fd_sc_hd__mux2_1 couldn&apos;t be read
Cell sky130_fd_sc_hd__mux2_1 couldn&apos;t be read
Cell sky130_fd_sc_hd__mux2_1 couldn&apos;t be read
  Processed 61252 subcell instances total.
Segmentation fault (core dumped)
</pre>

vijayank88, 2022-08-26:
The tail of log i got with reproducible:
```
Moving label "vssd1" from (null) to space in cell pinmux_top.
Moving label "vssd1" from (null) to space in cell pinmux_top.
Moving label "wbd_clk_int" from (null) to space in cell pinmux_top.
Moving label "wbd_clk_pinmux" from (null) to space in cell pinmux_top.
Processing pinmux_top
freeMagic called with NULL argument.
Extracting pinmux_top into pinmux_top.ext:
Can't open file ./reports/signoff/42-antenna.feedback.txt.
Reading extract file.
Building flattened netlist.
Running antenna checks.
antennacheck finished.
```

donn, 2022-08-26:
@RTimothyEdwards can you download the reproducible file and take a look?

RTimothyEdwards, 2022-08-27:
@donn :  I cannot reproduce the error.  The test case generates a segfault due to a missing `.magicrc` startup file.  When the correct startup file is provided, then the test case runs to completion, correctly.

RTimothyEdwards, 2022-08-27:
For the record, I just fixed the unrelated segfault problem in magic.

donn, 2022-08-27:
Thanks for the info. I'm on it.

closed 2022-11-11 by dineshannayya