The-OpenROAD-Project/OpenLane #1287: macro placement issue

vinayrayapati, 2022-08-27:
I tried introducing a custom macro into the example design of the picorv32a, my synthesis was successful , but when I am trying to run the floorpan , the flow is failed due to the following errors
```
[ERROR]: during executing openroad script /openlane/scripts/openroad/basic_mp.tcl
[ERROR]: Log: designs/picorv32a/runs/run/logs/placement/6-basic_mp.log
[ERROR]: Last 10 lines:
[INFO ODB-0226] Finished LEF file:  /openlane/designs/picorv32a/runs/run/tmp/merged.nom.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/picorv32a/runs/run/tmp/placement/5-global.def
[INFO ODB-0128] Design: picorv32
[INFO ODB-0130]     Created 409 pins.
[INFO ODB-0131]     Created 9659 components and 71327 component-terminals.
[INFO ODB-0133]     Created 9761 nets and 32721 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/picorv32a/runs/run/tmp/placement/5-global.def
[ERROR MPL-0004] No macros found.
Error: basic_mp.tcl, 36 MPL-0004
child process exited abnormally
```
how should I resolve the issue?

vijayank88, 2022-08-27:
@vinayrayapati 
Study `issue_template` and file github issue accordingly.
Need `issue_reproducible` to debug the issue.
https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/using_or_issue.md#manually refer for how create  reproducible test.

donn, 2023-04-13:
Invalid and Staled Out.

closed 2023-04-13 by donn