The-OpenROAD-Project/OpenLane #1290: [OpenPdk Issue] - sky130_fd_io folder missing in latest openpdk release 44a43c23c81b45b8e774ae7a84899a5a778b6b0b

dineshannayya, 2022-08-28:
With latest Openlane, OpenPDK has updated to 44a43c23c81b45b8e774ae7a84899a5a778b6b0b,  With this I see sky130_fd_io folder are missing and this folder need it for Simulation.

Without this file simulation compile fails
<pre>iverilog -g2005-sv -Ttyp -DFUNCTIONAL -DSIM -DUSE_POWER_PINS -DUNIT_DELAY=#0.1 \
        -f/home/dinesha/workarea/efabless/MPW-7/caravel/mgmt_core_wrapper/verilog/includes/includes.rtl.caravel \
        -f../../../verilog/includes/includes.rtl.caravel_user_project -o uart_master.vvp uart_master_tb.v
/opt/pdk_mpw7/sky130B/libs.ref/sky130_fd_io/verilog/sky130_fd_io.v: No such file or directory
Preprocessor failed with 1 errors.
</pre>

In the previous OpenPdk Version: e8294524e5f67c533c5d0c3afa0bcc5b2a5fa066 I see these folder as follow
<pre>/opt/pdk_mpw7/volare/sky130/versions/e8294524e5f67c533c5d0c3afa0bcc5b2a5fa066/sky130B/libs.ref/<font color="#CC0000"><b>sky130_fd_io/v</b></font>erilog/<font color="#CC0000"><b>sky130_fd_io.v</b></font>
</pre>

<pre>Kernel: Linux v5.15.0-46-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: b6bacc9d1ab469917fda7ceea61ea3a18984b818
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

b6bacc9 2022-08-24T21:21:29+02:00 Build Script Fixes (#1283) - Mohamed Gaber -  (HEAD -&gt; master)
f0f6055 2022-08-24T19:36:18+02:00 Remove `-echo` flags from `read_sdc` (#1284) - Anton Blanchard -  ()
9452129 2022-08-20T13:41:59+02:00 Fix &apos;run_yosys&apos; invocation during -synth_explore (#1278) - meta -  ()
---
Git Remotes

origin	https://github.com/efabless/OpenLane.git (fetch)
origin	https://github.com/efabless/OpenLane.git (push)
</pre>

dineshannayya, 2022-08-28:
Locally I have copied  folder from e8294524e5f67c533c5d0c3afa0bcc5b2a5fa066/sky130B/libs.ref/sky130_fd_io/  to 44a43c23c81b45b8e774ae7a84899a5a778b6b0b/sky130B/libs.ref/sky130_fd_io/  to move ahead with simulation

donn, 2022-08-29:
Fixed by re-uploading: Sorry about that

closed 2022-08-29 by donn