The-OpenROAD-Project/OpenLane #1294: Use YAMP instead of JSON for config.json

armleo, 2022-08-29:
### Prompt
[Most of JSON is valid YAML file](https://til.codeinthehole.com/posts/json-is-valid-yaml/). Currently the less readable JSON is used instead of yaml.

### Proposal
Replace JSON with YAML since JSON is valid YAML. But YAML is much more readable.

donn, 2022-08-29:
No can do, at least as far as I'm concerned. While valid JSON is valid YAML, [YAML is a much, much larger grammar that can be interpreted in very many ways](https://github.com/cblp/yaml-sucks).

We already have enough problems with typing as-is. YAML is ambiguous, and we're already running away screaming from Tcl.

For internal manifests and the like, it's fine, because we can enforce do's and dont's. That's something we entirely have control over. But configuration files are user-written. We can't control whether users will type "False", "false", "NO, "no" or "No". There are subsets such as [strictyaml](https://github.com/crdoconnor/strictyaml), but these also have the potential to confuse users who will wonder why their otherwise valid YAML file is not getting accepted.

I thought very hard before deciding that JSON was the way going forward.

armleo, 2022-08-29:
Understandable.

closed 2022-08-29 by armleo