The-OpenROAD-Project/OpenLane #1299: Flow consistently fails regardless of configurations

WebKingdom, 2022-08-31:
## Description

When hardening a SHA1 design inside the caravel user project, flow fails with FP_SIZING=relative and FP_CORE_UTIL=30%. Other configurations were also tried but hardening failed for all. Configuration trials can be seen in openlane/sha1_top/config.tcl (https://github.com/WebKingdom/bitcoin_asic/blob/055aed80953a4996cf8b931a32cf2a102ebe4ca6/openlane/sha1_top/config.tcl).

## Expected behavior

The design should be able to harden and go through layout and placement.

## Environment

```
Kernel: Darwin v21.6.0
Distribution: macOS 10.16
Python: v3.9.0 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)
```

## Reproduction Material

issue_reproducible attached as zip. 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9463417/issue_reproducible.zip)

## Logs

```
Console logs:
[STEP 18]
[INFO]: Running Detailed Placement...
[ERROR]: during executing openroad script /openlane/scripts/openroad/opendp.tcl
[ERROR]: Exit code: 1
...
[ERROR]: Last 10 lines:
[INFO DPL-0035] ANTENNA__14155__A1
[INFO DPL-0035] ANTENNA__14155__A1
[INFO DPL-0035] ANTENNA__20474__A2
[INFO DPL-0035] ANTENNA__20930__A
[INFO DPL-0035] ANTENNA__20930__A
[INFO DPL-0035] ANTENNA__16103__A0
[INFO DPL-0035] message limit reached, this message will no longer print
[ERROR DPL-0036] Detailed placement failed.
Error: opendp.tcl, 32 DPL-0036
```

vijayank88, 2022-09-01:
@WebKingdom 
Update your `config.tcl` with following configuration resolve the issue:
```
set ::env(CELL_PAD) 2
```

vijayank88, 2022-09-01:
Also you can use  `set ::env(PL_TARGET_DENSITY) 40` to utilize more core area.
[image]

WebKingdom, 2022-09-01:
I set those 2 variables but set PL_TARGET_DENSITY = 0.4 (not 40). I was able to get further to [STEP 39] with the following **openlane.log**:
[INFO]: Writing Verilog...
[INFO]: Running LEF LVS...
[INFO]: Running Magic DRC...
[INFO]: Converting Magic DRC Violations to Magic Readable Format...
[INFO]: Converting Magic DRC Violations to Klayout XML Database...
[ERROR]: There are violations in the design after Magic DRC.
[ERROR]: Total Number of violations is 5
[INFO]: Saving current set of views in '../Users/somasz/Documents/GitHub/mpw_6c/caravel_design/caravel_bitcoin_asic/openlane/sha1_top/runs/22_09_01_01_54/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at '../Users/somasz/Documents/GitHub/mpw_6c/caravel_design/caravel_bitcoin_asic/openlane/sha1_top/runs/22_09_01_01_54/reports/manufacturability.rpt'.
[INFO]: Created metrics report at '../Users/somasz/Documents/GitHub/mpw_6c/caravel_design/caravel_bitcoin_asic/openlane/sha1_top/runs/22_09_01_01_54/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

**The manufacturability.rpt contained:** 
Design Name: sha1_top
Run Directory: /Users/somasz/Documents/GitHub/mpw_6c/caravel_design/caravel_bitcoin_asic/openlane/sha1_top/runs/22_09_01_01_54
----------------------------------------

Magic DRC Summary:
Source: /Users/somasz/Documents/GitHub/mpw_6c/caravel_design/caravel_bitcoin_asic/openlane/sha1_top/runs/22_09_01_01_54/reports/signoff/drc.rpt
Violation Message "Min area of metal1 holes > 0.14um^2 (met1.7) "found 5 Times.
Total Magic DRC violations is 5
----------------------------------------

LVS Summary:
Source: /Users/somasz/Documents/GitHub/mpw_6c/caravel_design/caravel_bitcoin_asic/openlane/sha1_top/runs/22_09_01_01_54/logs/signoff/38-sha1_top.lvs.lef.log
LVS reports no net, device, pin, or property mismatches.
Total errors = 0
----------------------------------------

Antenna Summary:
No antenna report found.

vijayank88, 2022-09-01:
If you update `open_pdks` to latest it may resolve this error.

WebKingdom, 2022-09-02:
I do have the latest version of OpenLane and still get the error. Any suggestions? Thanks!

vijayank88, 2022-09-03:
`OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2`
While creating this issue, you shared above OpenLane version.
Can you post latest OpenLane version do you tried?

vijayank88, 2022-09-03:
`sha1_top` design tested at my end and am not seening magic drc error.
```
[INFO]: Running LEF LVS...
[STEP 43]
[INFO]: Running Magic DRC (log: logs/signoff/43-drc.log)...
[INFO]: Converting Magic DRC Violations to Magic Readable Format...
[INFO]: Converting Magic DRC Violations to Klayout XML Database...
[INFO]: No DRC violations after GDS streaming out.
[STEP 44]
[INFO]: Running OpenROAD Antenna Rule Checker (log: logs/signoff/44-antenna.log)...
[STEP 45]
[INFO]: Running CVC (log: logs/signoff/45-erc_screen.log)...
[INFO]: Saving current set of views in 'results/final'...
[INFO]: Saving runtime environment...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'reports/metrics.csv'.
[WARNING]: There are max fanout violations in the design at the typical corner. Please refer to 'reports/signoff/33-rcx_sta.slew.rpt'.
[INFO]: There are no hold violations in the design at the typical corner.
[INFO]: There are no setup violations in the design at the typical corner.
[SUCCESS]: Flow complete.
[INFO]: Note that the following warnings have been generated:
[WARNING]: There are max fanout violations in the design at the typical corner. Please refer to 'reports/signoff/33-rcx_sta.slew.rpt'.
```

donn, 2022-11-30:
I think this is fixed now?

closed 2022-11-30 by donn