The-OpenROAD-Project/OpenLane #1304: Flow fails [86 DPL-0036]

mohamedleithy, 2022-09-01:
## Description
Flow fails in cts [86 DPL-0036]

## Expected behavior
Flow to complete 

## Environment

```
Kernel: Darwin v21.2.0
Distribution: macOS 12.1
Python: v3.9.13 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: b6bacc9d1ab469917fda7ceea61ea3a18984b818
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

b6bacc9d 2022-08-24T21:21:29+02:00 Build Script Fixes (#1283) - Mohamed Gaber -  (HEAD -> master, tag: 2022.08.25, origin/master, origin/HEAD)
f0f60552 2022-08-24T19:36:18+02:00 Remove `-echo` flags from `read_sdc` (#1284) - Anton Blanchard -  ()
94521291 2022-08-20T13:41:59+02:00 Fix 'run_yosys' invocation during -synth_explore (#1278) - meta -  (tag: 2022.08.21)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

## Reproduction Material
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9473540/issue_reproducible.zip)

vijayank88, 2022-09-02:
@mohamedleithy 
Update your `config.tcl` with following variable will resolve the issue:
```
set ::env(DPL_CELL_PADDING) {2}
```

mohamedleithy, 2022-09-02:
Still failing @vijayank88 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9475915/issue_reproducible.zip)

vijayank88, 2022-09-02:
can you share `config.tcl`?

donn, 2022-09-02:
@mohamedleithy To clarify- this is running the handcrafted netlist through OpenLane, right? Not the DFFRAM custom placer?

vijayank88, 2022-09-02:
@mohamedleithy 
Just checked your `issue_reproducible`. Now its failed during detail routing.
So DPL issue got fixed. Please create new issue as per guidelines.

donn, 2022-09-02:
Sorry- we know this is annoying but it's a useful practice because a lot of people Google errors to fix them.

closed 2022-09-02 by donn

JiaDYuan, 2022-10-26:
> DFFRAM custom placer

what is "DFFRAM custom placer"