The-OpenROAD-Project/OpenLane #1306: Flow failed

mohamedleithy, 2022-09-02:
## Description
Flow fails in detailed placement. 

## Environment
```
Kernel: Darwin v21.2.0
Distribution: macOS 12.1
Python: v3.9.13 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: b6bacc9d1ab469917fda7ceea61ea3a18984b818
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

b6bacc9d 2022-08-24T21:21:29+02:00 Build Script Fixes (#1283) - Mohamed Gaber -  (HEAD -> master, tag: 2022.08.25, origin/master, origin/HEAD)
f0f60552 2022-08-24T19:36:18+02:00 Remove `-echo` flags from `read_sdc` (#1284) - Anton Blanchard -  ()
94521291 2022-08-20T13:41:59+02:00 Fix 'run_yosys' invocation during -synth_explore (#1278) - meta -  (tag: 2022.08.21)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)```

## Reproduction Material
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9477518/issue_reproducible.zip)

## config.tcl
```
set ::env(DESIGN_NAME) "RAM256"
set ::env(CLOCK_PORT) "CLK"
set ::env(CLOCK_PERIOD) "10.49"
set ::env(VERILOG_FILES) "./designs/ram/pdk_generics.v ./designs/ram/model.v"
set ::env(PDK) "sky130A"
set ::env(SYNTH_READ_BLACKBOX_LIB) "1"
set ::env(PL_RESIZER_DESIGN_OPTIMIZATIONS) "0"
set ::env(PL_RESIZER_TIMING_OPTIMIZATIONS) "0"
set ::env(GLB_RESIZER_DESIGN_OPTIMIZATIONS) "0"
set ::env(GLB_RESIZER_TIMING_OPTIMIZATIONS) "0"
set ::env(CELL_PAD) "2"
set ::env(FP_CORE_UTIL) "30"
set ::env(DPL_CELL_PADDING) {2}

```

vijayank88, 2022-09-02:
@mohamedleithy 
Copy & paste your failed log here

vijayank88, 2022-09-02:
With latest OpenROAD am not facing any issue with Detail routing. Routing completed successfully.
[image]

vijayank88, 2022-09-02:
The OpenLane commit you're using is `9 days back commit`. Please update to latest commit and try again.

donn, 2022-09-03:
So I've privately discussed this with @mohamedleithy: seems to be RAM. He's running it on a VM with 4 gigs of RAM.

So, I believe that's the issue.

closed 2022-09-03 by donn