The-OpenROAD-Project/OpenLane #1307: Floorplan failed to work with custom cell in interactive mode

AyaseErii, 2022-09-02:
## Description
Hi, I am trying to put a custom cell but still in sky130 tech into a simple example design, i.e. using a custom inverter cell instead of the standard cell inverter to run the design flow of an inverter. The interactive mode is required to complete the design flow. For putting a custom cell into a design, the corresponding lef file (basically cell lef) should be included as well, e.g. adding `set lefs [path]` and `add_lefs -src $lefs` is needed. The `run_synthesis` works well, however, the `run_floorplan` pops out an error says `No macros found`. The error message can be found in the log section. Did I miss something? Could anyone point me in the right direction? I will appreciate it!

## Expected behavior
The custom cell can be added to the design, i.e. the tool will use the custom cell to run the flow, the interactive mode can be completed without errors.

## Environment
```
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v4.15.0-191-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: c91abed14f73b142b0e0b5e7f43425471256397f
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

c91abed 2022-08-31T13:14:54+02:00 Streamlined PDK/SCL Configuration (#1297) - Mohamed Gaber -  (HEAD -> master, origin/master, origin/HEAD)
ae089d8 2022-08-31T11:38:39+02:00 Update OpenROAD, Magic + DEF DRC Enhancement (#1296) - Mohamed Gaber -  ()
e14b1fd 2022-08-29T11:43:42+02:00 Add vertical and horizontal spacing to `add_pdn_stripe` calls (#1289) - Dinesh Annayya -  (tag: 2022.08.30)
---
Git Remotes

origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

## Reproduction Material
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9480446/issue_reproducible.zip)

## Logs
```
[STEP 6]
[INFO]: Running basic macro placement (log: designs/inverter/runs/RUN_2022.09.02_18.35.04/logs/placement/6-basic_mp.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/basic_mp.tcl
[ERROR]: Log: designs/inverter/runs/RUN_2022.09.02_18.35.04/logs/placement/6-basic_mp.log
[ERROR]: Last 10 lines:
[INFO ODB-0226] Finished LEF file:  /openlane/designs/inverter/runs/RUN_2022.09.02_18.35.04/tmp/merged.nom.lef
[INFO ODB-0127] Reading DEF file: /openlane/designs/inverter/runs/RUN_2022.09.02_18.35.04/tmp/placement/5-global.def
[INFO ODB-0128] Design: inverter
[INFO ODB-0130]     Created 2 pins.
[INFO ODB-0131]     Created 1 components and 6 component-terminals.
[INFO ODB-0133]     Created 2 nets and 2 connections.
[INFO ODB-0134] Finished DEF file: /openlane/designs/inverter/runs/RUN_2022.09.02_18.35.04/tmp/placement/5-global.def
[ERROR MPL-0004] No macros found.
Error: basic_mp.tcl, 36 MPL-0004
child process exited abnormally
```

maliberty, 2022-09-02:
It seems that OL asssumes that EXTRA_LEF implies macros which isn't true here.

AyaseErii, 2022-09-03:
> It seems that OL asssumes that EXTRA_LEF implies macros which isn't true here.

I see. Do you think it works when I do not add `EXTRA_LEF` in the config.tcl? But how the OL can config the custom cell lef?

maliberty, 2022-09-03:
I think the OL team will need to fix this.  In the short term you could comment out the call to basic_macro_placement in /scripts/tcl_commands/floorplan.tcl

vijayank88, 2022-09-03:
@AyaseErii 
can you share your `config.tcl` used to run this flow?

donn, 2022-09-03:
Yeah: `EXTRA_LEFS` is for macros. We'd need to consider a way to properly support custom cells, which isn't as simple as it sounds sort of adding a variable like `DONT_ASSUME_MACROS` or something similarly cryptic.

I'd ask you to use the manual workaround provided by @maliberty for now until we can address this problem.

AyaseErii, 2022-09-03:
Thanks! Commenting out the `basic_macro_placement` helps to pass the floorplan. However, another error pops out when I was running `write_powered_verilog`. The error shows as below:
```
% write_powered_verilog
[ERROR]: write_powered_verilog missing required -output_def
```
Then I use `write_powered_verilog -output_def -output_verilog`, the tool pass one step of Writing Powered Verilog but pops out `can't read "arg_values(-output_verilog)": no such element in array`. Since it was not an error, I continued to the last command `run_antenna_check`. However, the final .gds does not include the custom cell at all. Does anything go wrong?

donn, 2022-09-04:
We just never tried custom cells, frankly, so there are a whole bunch of tools that are likely to misbehave.

Would you mind sharing the original design folder (and the files for the custom cell)? I can try to come up with a workaround until we implement proper support for custom cells.

AyaseErii, 2022-09-04:
> We just never tried custom cells, frankly, so there are a whole bunch of tools that are likely to misbehave.
> 
> Would you mind sharing the original design folder (and the files for the custom cell)? I can try to come up with a workaround until we implement proper support for custom cells.

Thanks for the reply! And sure, please find the zip file below:
[inverter.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9485260/inverter.zip)

It is a very simple design, i.e. using the design flow to generate an inverter. This file includes a custom inverter `inv_sc.mag` and a cell lef file `inv_sc.lef` as well as config.tcl required to run the flow. The zip file should include all the files to run the flow.

Right now, using the interactive mode, the tool still uses the inverter from standard cell lib rather than my custom one.

Goal: the tool should use my custom inverter to complete the flow, and my custom inverter should appear in the final gdsii rather than standard inverter.

Feel free to let me know if you need other info. 

Thanks!

maliberty, 2022-09-13:
Note that I recently changed MPL-0004 to a warning

maliberty, 2022-09-13:
I'm curious why you need a custom inverter...

AyaseErii, 2022-09-13:
Thanks for the question! Right now, we are trying to make a custom inverter cell that meets our expected specs (e.g. power) for an ASIC design while probably sacrificing a few other specs (e.g. area) that we accept. A standard inverter cannot meet our expected specs.

> Note that I recently changed MPL-0004 to a warning
> I'm curious why you need a custom inverter...

maliberty, 2022-09-13:
inv_sc5 is the same area as sky130_fd_sc_hd__inv_1 - does it achieve lower power somehow?

AyaseErii, 2022-09-13:
I would say I am trying to use another device flavor (hvt) to build a custom cell, starting from the simplest change.

> inv_sc5 is the same area as sky130_fd_sc_hd__inv_1 - does it achieve lower power somehow?

maliberty, 2022-09-13:
sky130_fd_sc_hdll claims to be low leakage.  We do need to support custom cells so its mostly just curiousity.

AyaseErii, 2022-09-13:
Thanks for bringing that suggestion! We probably want the leakage even lower than the cell in `hdll` library.

> sky130_fd_sc_hdll claims to be low leakage. We do need to support custom cells so its mostly just curiousity.