The-OpenROAD-Project/OpenLane #1309: error: pathspec 'master' did not match any file(s) known to git

Shrihari01, 2022-09-03:
## Description

## Expected behavior

## Environment

```
YOUR SURVEY HERE
```

## Reproduction Material

## Logs

```
YOUR LOGS HERE
```

vijayank88, 2022-09-03:
@Shrihari01 
What will be the issue here?

Shrihari01, 2022-09-03:
> @Shrihari01 What will be the issue here?

When make pdk run on the terminal this was found

vijayank88, 2022-09-03:
Still this is not sufficient to debug.
Provide link you followed for installation:
Copy-paste complete terminal log

Shrihari01, 2022-09-03:
git clone https://github.com/efabless/openlane.git --branch rc7
    cd openlane/
    export PDK_ROOT=<absolute path to where skywater-pdk and open_pdks will reside>
    make openlane
    make pdk
    make test # This is to test that the flow and the pdk were properly installed

Shrihari01, 2022-09-03:
https://openlane-docs.readthedocs.io/en/rtd-develop/

vijayank88, 2022-09-03:
@Shrihari01 
https://openlane-docs.readthedocs.io/en/rtd-develop/ this links has out-dated steps.

Follow instruction from here for latest OpenLane installation: https://github.com/The-OpenROAD-Project/OpenLane
If any issues again feel free to re-open this.

vijayank88, 2022-09-03:
If you want submit project MPW shuttle follow instruction from here for caravel setup: https://github.com/efabless/caravel_user_project/blob/main/docs/source/quickstart.rst

closed 2022-09-03 by vijayank88

Shrihari01, 2022-09-03:
I installed it from this link....I think it's working.I don't know how to
check it...
Can you tell me the steps on how to test the verilog code in openlane??

vijayank88, 2022-09-03:
are you looking for training on how to use caravel flow?

Shrihari01, 2022-09-03:
No...I got a task to install openlane  and do a example in that...