The-OpenROAD-Project/OpenLane #1310: Error: flow.tcl command not found

curiousJion, 2022-09-03:
Hello,
I just installed caravel user project on a remote server (centos 7). %Make setup was successful but make user_proj_example gives me:
Flow.tcl command not found error.

And Openlane and pdk roots are correctly exported.
Could you please tell me what is causing this error?
Thank you.

% pwd
/local/data0/Students/AS/CARAVEL/caravel_user_project

% make openlane
cd openlane && make openlane
make[1]: Entering directory '/local/data0/Students/AS/CARAVEL/caravel_user_project/openlane'
if [ -d "/local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/openlane_src" ]; then
echo "Deleting exisiting /local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/openlane_src" &&
rm -rf /local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/openlane_src && sleep 2;
fi
Deleting exisiting /local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/openlane_src
git clone https://github.com/The-OpenROAD-Project/OpenLane --branch=2022.07.02_01.38.08 --depth=1 /local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/openlane_src &&
cd /local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/openlane_src &&
export OPENLANE_IMAGE_NAME=efabless/openlane:2022.07.02_01.38.08 &&
export IMAGE_NAME=efabless/openlane:2022.07.02_01.38.08 &&
make pull-openlane
Cloning into '/local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/openlane_src'...
remote: Enumerating objects: 508, done.
remote: Counting objects: 100% (508/508), done.
remote: Compressing objects: 100% (428/428), done.
remote: Total 508 (delta 101), reused 254 (delta 35), pack-reused 0
Receiving objects: 100% (508/508), 2.77 MiB | 13.49 MiB/s, done.
Resolving deltas: 100% (101/101), done.
Note: switching to 'f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2'.
You are in 'detached HEAD' state. You can look around, make experimental
changes and commit them, and you can discard any commits you make in this
state without impacting any branches by switching back to a branch.
If you want to create a new branch to retain commits you create, you may
do so (now or later) by using -c with the switch command. Example:
git switch -c
Or undo this operation with:
git switch -
Turn off this advice by setting config variable advice.detachedHead to false
make[2]: Entering directory '/local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/openlane_src'
Makefile:23: warning: undefined variable 'DOCKER_SWAP'
Makefile:26: warning: undefined variable 'DOCKER_MEMORY'
Makefile:42: warning: undefined variable 'ROUTING_CORES'
Trying to pull registry.access.redhat.com/efabless/openlane:2022.07.02_01.38.08...
name unknown: Repo not found
Trying to pull registry.redhat.io/efabless/openlane:2022.07.02_01.38.08...
unable to retrieve auth token: invalid username/password: unauthorized: Please login to the Red Hat Registry using your Customer Portal credentials. Further instructions can be found here: https://access.redhat.com/RegistryAuthentication
Trying to pull docker.io/efabless/openlane:2022.07.02_01.38.08...
Getting image source signatures
Copying blob e23e0ddb585d [--------------------------------------] 0b / 0b
Copying blob d4d650a92807 [--------------------------------------] 0b / 0b
Copying blob 00da99f6da93 [--------------------------------------] 0b / 0b
Copying blob 6288ff22c326 [--------------------------------------] 0b / 0b
Copying blob 4273e1cebc47 [--------------------------------------] 0b / 0b
Copying blob 2d473b07cdd5 [--------------------------------------] 0b / 0b
Copying blob 0bb3fd49f294 [--------------------------------------] 0b / 0b
Copying blob e23e0ddb585d skipped: already exists
Copying blob d4d650a92807 skipped: already exists
Copying blob 00da99f6da93 skipped: already exists
Copying blob 6288ff22c326 skipped: already exists
Copying blob 4273e1cebc47 skipped: already exists
Copying blob 2d473b07cdd5 skipped: already exists
Copying blob 0bb3fd49f294 skipped: already exists
Copying blob 0046c99aea39 skipped: already exists
Copying blob 5c2567a679fb skipped: already exists
Copying blob a7cc087578da skipped: already exists
Copying blob 2fe7c73cb171 skipped: already exists
Copying blob 4a454693240e skipped: already exists
Copying blob bdfdcd774996 skipped: already exists
Copying blob b8f4e67850a8 skipped: already exists
Copying blob c8f71bfe3963 skipped: already exists
Copying blob 89f0b07d31a4 skipped: already exists
Copying blob a4337c6d6c3b skipped: already exists
Copying blob d6266388c01d skipped: already exists
Copying blob dbc522c54251 skipped: already exists
Copying blob 7d94296f04fd skipped: already exists
Copying blob 2c637aa86889 skipped: already exists
Copying blob 9000a623b0a2 skipped: already exists
Copying blob f4886a0ced97 skipped: already exists
Copying blob e3e05bf88f0f skipped: already exists
Copying blob ecd6b35221ba skipped: already exists
Copying blob de892deb34c8 skipped: already exists
Copying blob 75688db7a987 skipped: already exists
Copying blob bcd756ce1ebf skipped: already exists
Copying blob 72ee0d5243a6 skipped: already exists
Copying blob 4f4fb700ef54 skipped: already exists
Copying config 195c88fbb8 done
Writing manifest to image destination
Storing signatures
195c88fbb8ec91aa12e2182070c62744ab03d47ffb2fde61745e5bb8ad31f05a
make[2]: Leaving directory '/local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/openlane_src'
make[1]: Leaving directory '/local/data0/Students/AS/CARAVEL/caravel_user_project/openlane'
% make user_proj_example
make -C openlane user_proj_example
make[1]: Entering directory '/local/data0/Students/AS/CARAVEL/caravel_user_project/openlane'
user_proj_example

mkdir -p ./user_proj_example/runs/22_09_01_13_15
rm -rf ./user_proj_example/runs/user_proj_example
ln -s $(realpath ./user_proj_example/runs/22_09_01_13_15) ./user_proj_example/runs/user_proj_example
docker run -it -v
(realpath /local/data0/Students/AS/CARAVEL/caravel_user_project/..) -v /local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/pdks:/local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/pdks -v /local/data0/Students/AS/CARAVEL/caravel_user_project/caravel:/local/data0/Students/AS/CARAVEL/caravel_user_project/caravel -v /local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/openlane_src:/openlane -v /local/data0/Students/AS/CARAVEL/caravel_user_project/mgmt_core_wrapper:/local/data0/Students/AS/CARAVEL/caravel_user_project/mgmt_core_wrapper -e PDK_ROOT=/local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/pdks -e PDK=sky130A -e MISMATCHES_OK=1 -e CARAVEL_ROOT=/local/data0/Students/AS/CARAVEL/caravel_user_project/caravel -e OPENLANE_RUN_TAG=22_09_01_13_15 -e MCW_ROOT=/local/data0/Students/AS/CARAVEL/caravel_user_project/mgmt_core_wrapper -u 24484:3000
efabless/openlane:2022.07.02_01.38.08 sh -c "flow.tcl -design

(realpath ..) -save -tag 22_09_01_13_15 -overwrite"
sh: flow.tcl: command not found
make[1]: *** [Makefile:72: user_proj_example] Error 127
make[1]: Leaving directory '/local/data0/Students/AS/CARAVEL/caravel_user_project/openlane'
make: *** [Makefile:73: user_proj_example] Error 2

#--------------------------------------------------------------------------------------------------------------

And the caravel/dependencies/ includes these:
% ls dependencies/*
dependencies/openlane_src:
AUTHORS.md docker Makefile run_designs.py
configuration docs README.md run_issue_regressions.py
conf.py env.py regression_results scripts
CONTRIBUTING.md flow.tcl requirements_dev.txt tests
dependencies Jenkinsfile requirements_lint.txt
designs LICENSE requirements.txt
dependencies/pdks:
sky130A sky130B volare

vijayank88, 2022-09-03:
@curiousJion 
Its difficult to understand logs and command executed.
Can you please use markdown style
```

```
and place logs inside?

donn, 2022-09-03:
@curiousJion There is a mandatory issue template that helps us better diagnose your problem. Please update your issue to follow said template or file a new one.

curiousJion, 2022-09-03:
@donn Thanks for your message. I think I did what @vijayank88 said, but I don't know how to make it reproducible. Just tried the following steps:
1. clonned caravel repo
2. added openlane & pdk root variables to .cshrc (I am using this for all the commands)
3. ran make setup
4. ran make user_proj_example ==> sh: flow.tcl command not found

vijayank88, 2022-09-03:
@curiousJion 
 Please run the following shell command in the OpenLane root folder: in your case: `dependencies/openlane_src/`

      `  python3 ./env.py issue-survey`
`echo $OPENLANE_ROOT`
`echo $PDK_ROOT`
execute all above commands and paste your outputs here

curiousJion, 2022-09-03:
@vijayank88 thanks.
% echo $OPENLANE_ROOT
/local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/openlane_src
% echo $PDK_ROOT
/local/data0/Students/AS/CARAVEL/caravel_user_project/dependencies/pdks

% python3 ./env.py issue-survey
Error: template parsing error: template: image:1: function "json" not defined
Kernel: Linux v3.10.0-1160.66.1.el7.x86_64
Distribution: centos 7
Python: v3.9.12 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)
BUT, docker is already installed:

% docker run hello-world
Trying to pull [registry.access.redhat.com/hello-world.](http://registry.access.redhat.com/hello-world.)..
  name unknown: Repo not found
Trying to pull [registry.redhat.io/hello-world.](http://registry.redhat.io/hello-world.)..
  unable to retrieve auth token: invalid username/password: unauthorized: Please login to the Red Hat Registry using your Customer Portal credentials. Further instructions can be found here: https://access.redhat.com/RegistryAuthentication
Trying to pull [docker.io/library/hello-world.](http://docker.io/library/hello-world.)..
Getting image source signatures
Copying blob 2db29710123e done
Copying config feb5d9fea6 done
Writing manifest to image destination
Storing signatures

Hello from Docker!
This message shows that your installation appears to be working correctly.

donn, 2022-09-04:
@curiousJion

A. That's not the issue template. The issue template was the thing you deleted when creating the issue.
B. Using OpenLane with caravel is supported in the caravel repo, not here. https://github.com/efabless/caravel_user_project

closed 2022-09-04 by donn