The-OpenROAD-Project/OpenLane #1311: Failed at Routing congestion too high.

whutddk, 2022-09-04:
## Description

I got Routing congestion too high. error at `[INFO]: Running Global Routing Resizer Timing Optimizations...`.

I have try different `CELL_PAD`, because flow may failed at `Running Placement Resizer Design Optimizations` for `[ERROR DPL-0036] Detailed placement failed.`

I have try to reduce the area cost by deleting my cache, which will use `sky130_sram_1kbyte_1rw1r_32x256_8`. And the area have reduced from about 75% to 30%, and I believe the design is small enough.

The design is generated from chisel3 by the way.

## Expected behavior

## Environment

WSL2 Ubuntu22.04  memory:64GB memory CPU:5950x

https://github.com/efabless/caravel_user_project

## Reproduction Material

https://github.com/whutddk/Rift2Go_2330
commit:f20a84a79b6f4c09912288111236692d2a24fbe6

## Logs

```
###############################################################################
# Design Rules
###############################################################################
set_max_fanout 5.0000 [current_design]
[INFO]: Setting signal min routing layer to: met1 and clock min routing layer to met1. 
[INFO]: Setting signal max routing layer to: met4 and clock max routing layer to met4. 
-congestion_iterations 500 -verbose
[INFO GRT-0020] Min routing layer: met1
[INFO GRT-0021] Max routing layer: met4
[INFO GRT-0022] Global adjustment: 30%
[INFO GRT-0023] Grid origin: (0, 0)
[INFO GRT-0043] No OR_DEFAULT vias defined.
[INFO GRT-0088] Layer li1     Track-Pitch = 0.4600  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met1    Track-Pitch = 0.3400  line-2-Via Pitch: 0.3400
[INFO GRT-0088] Layer met2    Track-Pitch = 0.4600  line-2-Via Pitch: 0.3500
[INFO GRT-0088] Layer met3    Track-Pitch = 0.6800  line-2-Via Pitch: 0.6150
[INFO GRT-0088] Layer met4    Track-Pitch = 0.9200  line-2-Via Pitch: 0.6150
[INFO GRT-0019] Found 7322 clock nets.
[INFO GRT-0001] Minimum degree: 2
[INFO GRT-0002] Maximum degree: 21
[INFO GRT-0003] Macros: 0
[INFO GRT-0004] Blockages: 555977

[INFO GRT-0053] Routing resources analysis:
          Routing      Original      Derated      Resource
Layer     Direction    Resources     Resources    Reduction (%)
---------------------------------------------------------------
li1        Vertical            0             0          0.00%
met1       Horizontal    4314600       2240323          48.08%
met2       Vertical      3235950       2153568          33.45%
met3       Horizontal    2157300       1507384          30.13%
met4       Vertical      1510110        863264          42.83%
---------------------------------------------------------------

[INFO GRT-0101] Running extra iterations to remove overflow.
[WARNING GRT-0227] Reached 20 congestion iterations with less than 15% of reduction between iterations.
[INFO GRT-0197] Via related to pin nodes: 2031525
[INFO GRT-0198] Via related Steiner nodes: 103541
[INFO GRT-0199] Via filling finished.
[INFO GRT-0111] Final number of vias: 3213759
[INFO GRT-0112] Final usage 3D: 16188314
[ERROR GRT-0118] Routing congestion too high.
Error: resizer_routing_timing.tcl, 53 GRT-0118
```

donn, 2022-09-04:
@whutddk - You should've seen a message along the lines of "Issue reproducible created at <path>." Could you please share said issue reproducible? It would really speed things up.

whutddk, 2022-09-04:
hi, it was my first time running the ASIC flow. I am just following the step that the readme told.
First, clone my project https://github.com/whutddk/Rift2Go_2330, 

then 
```
cd <project_name>
mkdir dependencies
export OPENLANE_ROOT=$(pwd)/dependencies/openlane_src
export PDK_ROOT=$(pwd)/dependencies/pdks # you need to export this whenever you start a new shell
export PDK=sky130B
make setup
```
After that, I got an OpenLand-base env. And I run the flow.

```
make user_proj_example
```

whutddk, 2022-09-04:
It seems that the RTL is at `$(PROJECT_BASE)/verilog/rtl/***.v`
And All setting that allow user to modify is at `$(PROJECT_BASE)/openlane/user_proj_example/config.tcl`
And the run is based on the scrip at `$(PROJECT_BASE)/dependencies/openlane_src/flow.tcl`

donn, 2022-09-04:
That is not what I asked.

When routing failed, you should've seen a message saying that a "reproducible" was created. Please upload this reproducible.

whutddk, 2022-09-04:
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9484395/issue_reproducible.zip)
sorry, my english is poor

donn, 2022-09-04:
No problem! Thank you.

@maliberty Can you take a look?

vijayank88, 2022-09-05:
@whutddk 
https://github.com/whutddk/Rift2Go_2330/blob/main/openlane/user_proj_example/config.tcl I just went through your `config.tcl`.
`config.tcl` you created completely not fit for  caravel MPW shuttle for macro hardening.

Try with following `config.tcl` and let us know status for `user_proj_example`

```
# SPDX-FileCopyrightText: 2020 Efabless Corporation
#
# Licensed under the Apache License, Version 2.0 (the "License");
# you may not use this file except in compliance with the License.
# You may obtain a copy of the License at
#
#      http://www.apache.org/licenses/LICENSE-2.0
#
# Unless required by applicable law or agreed to in writing, software
# distributed under the License is distributed on an "AS IS" BASIS,
# WITHOUT WARRANTIES OR CONDITIONS OF ANY KIND, either express or implied.
# See the License for the specific language governing permissions and
# limitations under the License.
# SPDX-License-Identifier: Apache-2.0

set ::env(PDK) $::env(PDK)
set ::env(STD_CELL_LIBRARY) "sky130_fd_sc_hd"

set script_dir [file dirname [file normalize [info script]]]

set ::env(DESIGN_NAME) rift2Wrap

set ::env(VERILOG_FILES) "\
	$::env(CARAVEL_ROOT)/verilog/rtl/defines.v \
	$script_dir/../../verilog/rtl/rift2Wrap.v \
	$script_dir/../../verilog/rtl/Rift2LinkA.v \
	$script_dir/../../verilog/rtl/plusarg_reader.v \
	$script_dir/../../verilog/rtl/bim_P_ext.v \
	$script_dir/../../verilog/rtl/btb_table_target_ext.v \
	$script_dir/../../verilog/rtl/tagMem_ext.v \
	$script_dir/../../verilog/rtl/datMem_0_ext.v"

set ::env(SYNTH_NO_FLAT) 1
# set ::env(SYNTH_STRATEGY) "AREA 3"
set ::env(SYNTH_MAX_FANOUT) 5

set ::env(DESIGN_IS_CORE) 0

set ::env(CLOCK_PORT) "user_clock2"
set ::env(CLOCK_NET) "i_Rift2LinkA.clock"
set ::env(CLOCK_PERIOD) "50"

#set ::env(FP_SIZING) absolute
#set ::env(DIE_AREA) "0 0 2920 3520"

set ::env(FP_SIZING) relative
set ::env(FP_CORE_UTIL) 30

set ::env(FP_PIN_ORDER_CFG) $script_dir/pin_order.cfg

set ::env(PL_BASIC_PLACEMENT) 0
set ::env(PL_TARGET_DENSITY) 0.35
set ::env(CELL_PAD) 3

#set ::env(GLB_RT_OVERFLOW_ITERS) 500
# Maximum layer used for routing is metal 4.
# This is because this macro will be inserted in a top level (user_project_wrapper) 
# where the PDN is planned on metal 5. So, to avoid having shorts between routes
# in this macro and the top level metal 5 stripes, we have to restrict routes to metal4.  
# 
set ::env(RT_MAX_LAYER) {met4}

# You can draw more power domains if you need to 
set ::env(VDD_NETS) [list {vccd1}]
set ::env(GND_NETS) [list {vssd1}]

set ::env(DIODE_INSERTION_STRATEGY) 4 
# If you're going to use multiple power domains, then disable cvc run.
set ::env(RUN_CVC) 1
```

Just make sure using `CLOCK_PORT` and `CLOCK_NET`. Refer [README.md](https://github.com/The-OpenROAD-Project/OpenLane/blob/57f3f21d6e1d0403a88d7a7973f7447597aae160/configuration/README.md#required-variables)

whutddk, 2022-09-05:
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9489219/issue_reproducible.zip)
hi, @vijayank88  . I have tried with the `config.tcl` you provided. And that is the result. Seems the same trouble.

vijayank88, 2022-09-05:
@whutddk 
Found reason why its taking large die size. 
Can you blackbox your memory and use pre-built openRAM macros?
Otherwise it can't meet die size required by caravel flow.

whutddk, 2022-09-05:
I have ever used pre-built openRAM macros to blackbox my memory: L2cache, L1-Dcache, L1-ICache, PTW-Cache, BTB. But I found it cost too much die size, the smallest pre-built openRAM is `32bits x 256`. So I remove all memory except L1Dcache and BTB. And reduce their depth without balckbox. OK, I will blackbox them again.

By the way, can you provide some smaller pre-built openRAM macros in next mpw shuttle?

vijayank88, 2022-09-05:
Check your `pdks`
```
% more pdks/sky130B/libs.ref/sky130_sram_macros/verilog/s
sky130_sram_1kbyte_1rw1r_32x256_8.v  sky130_sram_1kbyte_1rw1r_8x1024_8.v  sky130_sram_2kbyte_1rw1r_32x512_8.v  sram_1rw1r_32_256_8_sky130.v
```
If required size not exist, you can generate from [here](https://github.com/VLSIDA/OpenRAM)

vijayank88, 2022-09-05:
Routing congestion view:
[image]

whutddk, 2022-09-05:
how can I open this view? Which command and load which files?

vijayank88, 2022-09-05:
```
openroad -gui
read_lef tmp/merged.nom.lef
read_def results/routing/user_proj_example.def
```

whutddk, 2022-09-08:
Thank you, I have passed the route stage. However, I failed at `LEF DEF` with mismatch now.
[39lef.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9531072/39lef.zip)

closed 2022-09-10 by whutddk

maliberty, 2022-09-10:
You should open a new issue if you have a different problem.  This one is already closed.