The-OpenROAD-Project/OpenLane #1313: Flow failed during LEF LVS

vijayank88, 2022-09-05:
## Description

`yonga_modbus_controller` is a MPW shuttle design. Its running part of regression in CI [here]( https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI)
Latest regression run here: https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI/pull/77/checks?check_run_id=8172372160

Till OpenLane [2022.09.02](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.02) flow completes successfully.
From OpenLane tag [2022.09.03](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.03) its Failed during `Running LEF LVS`

The difference i see wrt., `netgen` update.
logs/signoff/33-user_project_wrapper.lef.log as follows:
```
(no matching pin)                          |io_oeb[3]                                  
(no matching pin)                          |io_oeb[2]                                  
(no matching pin)                          |io_oeb[1]                                  
(no matching pin)                          |io_oeb[0]                                  
---------------------------------------------------------------------------------------
Cell pin lists for user_project_wrapper and user_project_wrapper altered to match.
Cells failed matching, or top level cell failed pin matching.
```

## Expected behavior

Flow should pass all stages.

## Environment

```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: OL_Tag_09_03-dev (Testing OpenLane tag: [2022.09.03](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.03))
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---

```

## Reproduction Material

Required files for running `netgen` can be downloaded from here: https://jenkins.openroad.tools/job/OpenLane-MPW-CI-Public/job/production-small-3/job/PR-77/lastFailedBuild/artifact/OpenLane/designs/yonga_modbus_controller/runs/RUN_2022.09.03_22.22.18/
## Logs

```
[STEP 33]
[INFO]: Running LEF LVS...
[ERROR]: There are LVS errors in the design: See 'logs/signoff/33-user_project_wrapper.lvs.lef.log' for details.
[INFO]: Saving current set of views in 'results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
[INFO]: The failure may have been because of the following warnings:
[WARNING]: Skipping Tap/Decap Insertion.

    while executing
"flow_fail"
    (procedure "quit_on_lvs_error" line 12)
    invoked from within
"quit_on_lvs_error -log $count_lvs_log"
    (procedure "run_lvs" line 79)
    invoked from within
"run_lvs"
    (procedure "run_lvs_step" line 10)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 52)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 401)
make: *** [Makefile:175: quick_run] Error 1
make: Leaving directory '/tmp/workspace/_Public_production-small-3_PR-77@2/OpenLane'
```

kareefardi, 2023-01-31:
@vijayank88 is this still an issue?

vijayank88, 2023-02-01:
@kareefardi 
Still facing LVS error:
```
                               
(no matching pin)                          |io_oeb[37]                                  
.
.
(no matching pin)                          |io_oeb[6]                                  
(no matching pin)                          |io_oeb[5]                                  
(no matching pin)                          |io_oeb[4]                                  
(no matching pin)                          |io_oeb[3]                                  
(no matching pin)                          |io_oeb[2]                                  
(no matching pin)                          |io_oeb[1]                                  
(no matching pin)                          |io_oeb[0]
```

closed 2023-02-01 by vijayank88

d-m-bailey, 2023-02-01:
@vijayank88 You could try adding this to the magic extraction command.
```
ext2spice short resistor
```

Without this, only one of the shorted pins remains in the extracted netlist. With this, all the pins remain with a virtual resistor between shorted pins (which matches the verilog netlist).

This won't work with SRAM designs though.

vijayank88, 2023-02-02:
@d-m-bailey 

For `assign io_oeb = {`MPRJ_IO_PADS{wb_rst_i}};`

In generated netlist 
```
assign io_oeb = { wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i, wb_rst_i };
```
All reset pin connected to io_oeb. But why LVS throws error as unmatched pins?

d-m-bailey, 2023-02-02:
In the layout, all the `io_oeb[*]` pins are shorted to one net. 
Without `ext2spice short resistor`, only one of the shorted nets remains in the final extracted netlist - the pins connected to the shorted nets are removed causing the unmatched pins error.

vijayank88, 2023-02-02:
Ok. I will try to include above option and let you know. thanks

vijayank88, 2023-02-02:
@d-m-bailey Thanks. After updating `ext2spice short resistor` LVS issue got fixed.
Is that ok to add that variable permanently  or its need to be added only specific scenarios?

d-m-bailey, 2023-02-02:
It doesn't work with openram, but any design other than that might be ok.