The-OpenROAD-Project/OpenLane #1320: "Net zero_ of signal type GROUND is not routable by TritonRoute. Move to special nets" error in detailed routing

mbaykenar, 2022-09-06:
## Description

In a design for MPW7 shuttle, I hardened my macros and then in the top module (user_project_wrapper) connected them by instantiation. There are also assignment code, in which an input signal is directly assigned to some output signals. Other than, it is all module instantiation and connections.

openlane quit at Detailed Routing stage by complaining "Net zero_ of signal type GROUND is not routable by TritonRoute. Move to special nets"

It is similar to issue #1185 However, I don't understand the solution.
## Expected behavior

Getting rid of the error
## Environment

```
Kernel: Linux v5.4.0-124-generic
Distribution: ubuntu 18.04
Python: v3.6.9 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)
```

## Reproduction Material

You can find issue_reproducible.zip, openlane.log, errors.log, cmds.log and config.tcl file in the attachment
Also my github repo is: https://github.com/mbaykenar/mpw7_yonga_soc
## Logs

```
YOUR LOGS HERE
```

[cmds.log](https://github.com/The-OpenROAD-Project/OpenLane/files/9493424/cmds.log)
[errors.log](https://github.com/The-OpenROAD-Project/OpenLane/files/9493426/errors.log)
[openlane.log](https://github.com/The-OpenROAD-Project/OpenLane/files/9493427/openlane.log)
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9493433/issue_reproducible.zip)
[config.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9493434/config.zip)

donn, 2022-09-06:
A fix was pulled in to a later version of OpenLane than the one verified for OpenMPW. I would file this same issue in https://github.com/efabless/caravel_user_project. They need to update their OpenLane.

donn, 2022-09-06:
Let me keep this open for posterity, actually.

mbaykenar, 2022-09-06:
I added these lines after initialize_floorplan in floorplan.tcl of MPW7 openlane version as stated in issue #1185 and it passed at least Detailed Routing stage:

set tielo_cell [lindex $::env(SYNTH_TIELO_PORT) 0]
set tielo_port [lindex $::env(SYNTH_TIELO_PORT) 1]
set tiehi_cell [lindex $::env(SYNTH_TIEHI_PORT) 0]
set tiehi_port [lindex $::env(SYNTH_TIEHI_PORT) 1]
insert_tiecells "$tielo_cell/$tielo_port" -prefix "TIE_ZERO_"
insert_tiecells "$tiehi_cell/$tiehi_port" -prefix "TIE_ONE_"

donn, 2022-09-06:
Well, yeah, but that violates the principle that you should be using the version of OpenLane caravel tells you to. It helps Efabless make sense of your results if something goes wrong.

kareefardi, 2024-07-17:
This is probably fixed by now

closed 2024-07-17 by kareefardi