The-OpenROAD-Project/OpenLane #1321: Verilog code execution in OpenLane

Shrihari01, 2022-09-06:
How to insert src file and config.json in designs folder of required verilog program ?

To run and test the verilog code

vijayank88, 2022-09-06:
Have you referred built-in examples? https://github.com/The-OpenROAD-Project/OpenLane/blob/master/designs/APU/config.json

Shrihari01, 2022-09-06:
How to run the code in terminal?

vijayank88, 2022-09-06:
Fill `issue_template` properly, so i can know which version of OpenLane currently used

vijayank88, 2022-09-06:
## Environment

```
YOUR SURVEY HERE
```

Shrihari01, 2022-09-06:
WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v5.15.0-41-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: d609968cca9ebc6aae6ba8ada9c16b1ed3ce4d0a
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

d609968 2022-09-06T04:48:37+02:00 Replace mismatches' environment variables with flags (#1316) - Mohamed Gaber -  (HEAD -> master, origin/master, origin/HEAD)
2d97f88 2022-09-04T12:29:33+02:00 [BOT] Update magic (#1308) - Openlane Bot -  (tag: 2022.09.05)
460d3d8 2022-09-02T11:20:08+02:00 [BOT] Update yosys (#844) - Openlane Bot -  (tag: 2022.09.03)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)

vijayank88, 2022-09-06:
Refer this document https://github.com/The-OpenROAD-Project/OpenLane/blob/master/designs/README.md to create new design.

Place your verilog code insde `OpenLane/designs/<design name>/src/`
Update `config.json` for `CLOCK_PORT` and `CLOCK_PERIOD`
Execute `./flow.tcl -design <design name>` will run yosys and synthesize your verilog code.

Shrihari01, 2022-09-06:
How to Update config.json for CLOCK_PORT and CLOCK_PERIOD ?

vijayank88, 2022-09-06:
in slack DM me: D026ENCEDK3

vijayank88, 2022-09-06:
share your verilog code and top module name. 
I will share you `config.json` and use the same at your end

Shrihari01, 2022-09-06:
// top module name-alu.v

module alu32(in_a,in_b, 
c_opcode,o_result);
input [31:0]in_a,in_b;
input [2:0]c_opcode;
output [31:0]o_result;
reg [31:0]o_result;
always@(in_a,in_b,c_opcode)
begin
case(c_opcode)
3'b000: o_result = in_a + in_b;
3'b001: o_result = in_a - in_b;
3'b010: o_result = in_a + 1;
3'b011: o_result = in_a - 1;
3'b100: o_result = in_a?1:0; // true
3'b101: o_result = ~in_a; // Complement
3'b110: o_result = in_a & in_b; //AND Bitwise
3'b111: o_result = in_a | in_b; // OR Bitwise
default : o_result = 32'bx;
endcase
end
endmodule

vijayank88, 2022-09-06:
[alu32.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9494498/alu32.zip)
Unzip attached file and place `alu32` directory under `OpenLane/designs/alu32`
Run following command, it will complete RTL-GDS2 flow.
```
cd OpenLane
make mount
./flow.tcl -design alu32
```

vijayank88, 2022-09-06:
This page specific to report tool related issues. 
Please read the document and try to understand the flow.
Already built-in examples there how to use clock less designs: https://github.com/The-OpenROAD-Project/OpenLane/blob/master/designs/inverter/config.json

Learn from examples and implement your own design.
Thanks

closed 2022-09-06 by vijayank88