The-OpenROAD-Project/OpenLane #1325: Missing variables documentation

kareefardi, 2022-09-06:
### Prompt
The following list are variables being used by the flow but not documented. These variables could be potentially used by a user but I am still not confident whether this is the case for all of them. I need to review them again. I am opening this to start a discussion about these.

```
CARRY_SELECT_ADDER_MAP
CELLS_LEF_OPT
CORE_AREA
CVC_SCRIPTS_DIR
DESIGN_DIR
DFF_LIB_SYNTH
DIODE_CELL
DIODE_CELL_PIN
DRC_EXCLUDE_CELL_LIST_OPT
ECO_ENABLE
ECO_FINISH
ECO_FIX_FILE
ECO_ITER
ECO_SKIP_PIN
EXTRA_LIBS
FAKEDIODE_CELL
FP_CORE_MARGIN
FP_PDN_ENABLE_GLOBAL_CONNECTIONS
FP_PDN_HSPACING
FP_PDN_LOWER_LAYER
FP_PDN_RAILS_LAYER
FP_PDN_UPPER_LAYER
FP_PDN_VSPACING
FULL_ADDER_MAP
GDS_FILES_OPT
GPIO_PADS_LEF_CORE_SIDE
GPIO_PADS_VERILOG
KLAYOUT_DRC_TECH_SCRIPT
LAYERS_RC
LIB_SYNTH_COMPLETE_NO_PG
LIB_SYNTH_NO_PG
LIB_SYNTH_OPT
LVS_EXTRA_GATE_LEVEL_VERILOG
LVS_EXTRA_STD_CELL_LIBRARY
METAL_LAYER_NAMES
PDN_CFG
PL_RESIZER_TIE_SEPERATION
PL_TARGET_DENSITY_CELLS
QUIT_ON_HOLD_VIOLATIONS
QUIT_ON_SETUP_VIOLATIONS
QUIT_ON_TIMING_VIOLATIONS
RCX_CC_MODEL
RCX_CONTEXT_DEPTH
RCX_CORNER_COUNT
RCX_COUPLING_THRESHOLD
RCX_MAX_RESISTANCE
RIPPLE_CARRY_ADDER_MAP
STA_REPORT_POWER
STD_CELL_GROUND_PINS
STD_CELL_LIBRARY_CDL
STD_CELL_LIBRARY_OPT_CDL
STD_CELL_POWER_PINS
SYNTH_DEFINES
SYNTH_EXPLORE
SYNTH_MUX_MAP
SYNTH_SCRIPT
SYNTH_TOP_LEVEL
TECH_LEF_OPT
TECH_METAL_LAYERS
USE_GPIO_ROUTING_LEF
VIAS_RC
```

### Proposal
Document these variables.

donn, 2022-09-06:
A lot of these are internally used variables that should NOT be documented `DESIGN_DIR`, `ECO_[^E].+`, but others are.

kareefardi, 2022-09-06:
Yes. Although `DESIGN_DIR` can be used in a `config.tcl` file. But Yes I don't disagree. Let's try and narrow them down to the ones that do need documentation

kareefardi, 2022-09-06:
I already eliminated a lot of variables but it's hard to know them all.

donn, 2022-09-06:
Since it is an undocumented variable, when I took over I've made an assumption that DESIGN_DIR is necessarily the directory where `config.tcl` is. Too late to change that.

donn, 2022-09-06:
`config.json`, technically. I want people to stop using `config.tcl` :)

kareefardi, 2022-09-06:
If you already know some that need to be eliminated from this list other than `ECO_*`, please mention them

kareefardi, 2023-08-01:
This is handled in a series of PRs https://github.com/The-OpenROAD-Project/OpenLane/pull/1764, https://github.com/The-OpenROAD-Project/OpenLane/pull/1891. It was also revisited in https://github.com/The-OpenROAD-Project/OpenLane/issues/1889

closed 2023-08-01 by kareefardi