The-OpenROAD-Project/OpenLane #1329: Please don't kill local install

anthonyjwall, 2022-09-06:
## Description
I'm still using local install, please don't kill it!

## Expected behavior
local_install not to be deprecated 

## Environment
N/A

## Reproduction Material
N/A

Reporting this issue upon receiving the alert telling me to do so when running 
`env.py local-install`

## Logs

Thanks,
Anthony

donn, 2022-09-06:
Thanks for your input! I left that message to gauge how many people were using the local installer (We don't have analytics set up.)

Please see my comment in #1300 for more info.

anthonyjwall, 2022-09-06:
Thank you - I understand and agree with your point on environment variable chaos. I like to maintain my own tools and pdk, partly out of stubbornness (I'm scared of containers), and partly because I use the analog tools also: I want all the tools, including analog, to sit in the same place, e.g. /tools so I have a predictable path to each tool:
/tools/xschem/install/bin
/tools/OpenROAD/install/bin
etc etc