The-OpenROAD-Project/OpenLane #1330: Add MPW CI as a badge

tspyrou, 2022-09-06:
https://github.com/The-OpenROAD-Project/OpenLane/blob/master/README.md should have a badge pointing to the MPW CI's created by Vijayan.

### Prompt
A clear and concise description of what shortcoming you feel OpenLane has.

### Proposal
A clear and concise description of what you want to happen.

donn, 2022-09-06:
Why.

tspyrou, 2022-09-06:
@donn Tim had asked me to create improved CI to help avoid OpenLane needing to use a release branch but instead work from the head. This can be one step in that direction. We could start with just one set of the MPW designs and set them to be required to pass when a PR is made.

donn, 2022-09-07:
I'm not against the idea, but we need to discuss it first.

closed 2024-09-12 by donn