The-OpenROAD-Project/OpenLane #1331: Crash in PDN step

ratul619, 2022-09-06:
Hi , I have  a macro based design and placed the macro almost at the middle - but the PDN step crashes. Please help ASAP

[INFO]: Running Tap/Decap Insertion...
[INFO]: Power planning with power {VPWR} and ground {VGND}...
[STEP 7]
[INFO]: Generating PDN...
[ERROR]: during executing openroad script /openlane/scripts/openroad/pdn.tcl
[ERROR]: Exit code: 1
[ERROR]: full log: designs/rram_wrapper_16x16/runs/latest_new/logs/floorplan/7-pdn.log
[ERROR]: Last 10 lines:
33# 0x00007FCEBB577F1E in /lib64/libtcl8.5.so
34# Tcl_EvalEx in /lib64/libtcl8.5.so
35# Tcl_Eval in /lib64/libtcl8.5.so
36# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
37# ord::tclAppInit(Tcl_Interp*) in openroad
38# Tcl_Main in /lib64/libtcl8.5.so
39# main in openroad
40# __libc_start_main in /lib64/libc.so.6
41# 0x0000000001028727 in openroad
child process exited abnormally

[INFO]: Saving runtime environment...
[INFO]: Creating reproducible...

or_issue.py OpenROAD Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/rram_wrapper_16x16/runs/latest_new/openroad_issue_reproducible…
[FIN] Done.
[INFO]: Reproducible packaged: Please tarball and upload designs/rram_wrapper_16x16/runs/latest_new/openroad_issue_reproducible if you're going to submit an issue.
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/rram_wrapper_16x16/runs/latest_new/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/rram_wrapper_16x16/runs/latest_new/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

ratul619, 2022-09-06:
[](url)
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9499761/openroad_issue_reproducible.tar.gz)

maliberty, 2022-09-06:
There is something wrong with the packaging of this test case:
Error: pdn.tcl, 23 cannot read file ./designs/rram_wrapper_16x16/1T1R_16x16_W7u_xschem_with_16ADC_16CSA_06_09_layout.lib.

(@donn any ideas?)

maliberty, 2022-09-06:
It looks like EXTRA_LIBS isn't packaged

donn, 2022-09-06:
I mean, without the issue template I don't know what version of OpenLane this is, so…

closed 2022-09-06 by donn