The-OpenROAD-Project/OpenLane #1341: Congestion in groute

ratul619, 2022-09-08:
Hi , getting congestion issue in groute.... Seems cells are getting clustered at one place although design is empty ...

please check and help regarding same....
[openroad_issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9527831/openroad_issue_reproducible.tar.gz)

ratul619, 2022-09-08:
hi matt, let me know if you need anythuing to be done at my end,,,thanks a lot

ratul619, 2022-09-08:
also i had to manually copy the .lib file in the area , since by default it was not copying

donn, 2022-09-08:
You didn't follow the issue template. I need the issue template so I can help you.

ratul619, 2022-09-08:
Where can ui find the issue template

vijayank88, 2022-09-09:
@ratul619 
While creating the issue, you could've seen this template. But you missed to fill it.

## Environment
\

\```
YOUR SURVEY HERE
\```

vijayank88, 2022-09-09:
As per slack discussion, user `config.tcl` as follows
```
set ::env(DESIGN_NAME) "RRAM_CONTROLLER"
set ::env(PDK) {sky130B}
set ::env(VERILOG_FILES) [glob  ./designs/RRAM_CONTROLLER/src/top_0709.v]
set ::env(CLOCK_PERIOD) "10.000"
set ::env(CLOCK_PORT) "clk"
set ::env(CELL_PAD) 4
set ::env(EXTRA_LIBS) "./designs/RRAM_CONTROLLER/RRAM_ANALOG.lib ./designs/RRAM_CONTROLLER/sync_fifo_16x16.lib ./designs/RRAM_CONTROLLER/sync_fifo_32x64.lib"
set ::env(EXTRA_LEFS) { ./designs/RRAM_CONTROLLER/RRAM_ANALOG.lef ./designs/RRAM_CONTROLLER/sync_fifo_16x16.lef ./designs/RRAM_CONTROLLER/sync_fifo_32x64.lef}
set ::env(DIE_AREA) {0.0 0.0 2500 2500  }set ::env(PL_TARGET_DENSITY) {0.20}
set ::env(FP_SIZING) "absolute"
#set ::env(TAP_DECAP_INSERTION) {0}
#set ::env(FILL_INSERTION) {0}
#set ::env(QUIT_ON_TR_DRC) {0}#set ::env(FP_PDN_ENABLE_MACROS_GRID) {0}
#set ::env(ROUTING_OPT_ITERS) {10}
set ::env(EXTRA_GDS_FILES)  {./designs/RRAM_CONTROLLER/RRAM_ANALOG.GDS ./designs/RRAM_CONTROLLER/sync_fifo_16x16.magic.gds ./designs/RRAM_CONTROLLER/sync_fifo_32x64.magic.gds}
#set ::env(FP_PIN_ORDER_CFG) $::env(OPENLANE_ROOT)/designs/rram_wrapper_16x16/pin_order.cfg
set ::env(MACRO_PLACEMENT_CFG) [glob $::env(DESIGN_DIR)/macro.cfg]
set ::env(SYNTH_BUFFERING) {0}set ::env(SYNTH_NO_FLAT) {1}
set ::env(PL_BASIC_PLACEMENT) {1}
#set ::env(TAP_DECAP_INSERTION) {0}
#set ::env(FP_PDN_CHECK_NODES) 0
set ::env(FP_PDN_ENABLE_MACROS_GRID) {1}
set ::env(FP_PDN_MACRO_HOOKS) "sync_fifo_16x16 VPWR VGND VPWR VGND,sync_fifo_32x64 VPWR VGND VPWR VGND"
set ::env(VDD_PIN) {VPWR vdd}
set ::env(GND_PIN) {VGND gnd}
set filename ./designs/$::env(DESIGN_NAME)/$::env(PDK)_$::env(STD_CELL_LIBRARY)_config.tcl
if { [file exists $filename] == 1} {
	source $filename
}
set ::env(DESIGN_IS_CORE) 0
```

`set ::env(PL_BASIC_PLACEMENT) {1}` this variable causing the issue. Disable this variable and re-run will resolve congestion issue. @ratul619  confirm by updating above variable.

closed 2022-09-09 by donn

ratul619, 2022-09-09:
Kernel: Linux v5.15.0-46-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: UNKNOWN vUNKNOWN (UNSUPPORTED)
OpenLane Git Version: 63b5966c2874da8bf01a831f0ee9dc68382a4958
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

63b5966 2022-09-07T14:06:24+02:00 Documentation Fixes (#1333) - Mohamed Gaber -  (HEAD -> master, tag: 2022.09.08, origin/master, origin/HEAD)
4dd0170 2022-09-06T14:28:07+02:00 Fix Magic Reproducibles (#1324) - Mohamed Gaber -  (tag: 2022.09.07)
826bb8f 2022-09-06T12:10:03+02:00 Installation, Quickstart Guide and Sphynx Overhaul (#1259) - Arman Avetisyan -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)

ratul619, 2022-09-09:
'''Kernel: Linux v5.15.0-46-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: UNKNOWN vUNKNOWN (UNSUPPORTED)
OpenLane Git Version: 63b5966c2874da8bf01a831f0ee9dc68382a4958
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

63b5966 2022-09-07T14:06:24+02:00 Documentation Fixes (#1333) - Mohamed Gaber -  (HEAD -> master, tag: 2022.09.08, origin/master, origin/HEAD)
4dd0170 2022-09-06T14:28:07+02:00 Fix Magic Reproducibles (#1324) - Mohamed Gaber -  (tag: 2022.09.07)
826bb8f 2022-09-06T12:10:03+02:00 Installation, Quickstart Guide and Sphynx Overhaul (#1259) - Arman Avetisyan -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
'''