The-OpenROAD-Project/OpenLane #1345: congestion issue over analog -

ratul619, 2022-09-09:
'''WARNING: issue-survey appears to be running inside the OpenLane
container.

This makes it difficult to rule out issues with your
environment.

Unless instructed specifically to do so, please run this command
outside the OpenLane container.
---

Kernel: Linux v5.15.0-46-generic
Distribution: centos 7
Python: v3.6.8 (OK)
OpenLane Git Version: 63b5966c2874da8bf01a831f0ee9dc68382a4958
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

63b5966 2022-09-07T14:06:24+02:00 Documentation Fixes (#1333) - Mohamed Gaber -  (HEAD -> master, tag: 2022.09.08, origin/master, origin/HEAD)
4dd0170 2022-09-06T14:28:07+02:00 Fix Magic Reproducibles (#1324) - Mohamed Gaber -  (tag: 2022.09.07)
826bb8f 2022-09-06T12:10:03+02:00 Installation, Quickstart Guide and Sphynx Overhaul (#1259) - Arman Avetisyan -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
'''

ratul619, 2022-09-09:
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9535034/issue_reproducible.tar.gz)

ratul619, 2022-09-09:
Hi , congestion issue showing above analog block , this is causing tool to stop

vijayank88, 2022-09-09:
@maliberty 
Getting congestion inside analog block. Congestion report shows `unknown sources`. Is that possible to fix?

maliberty, 2022-09-12:
This block has bad pins.  For example IN1_WL[2]:

[image]

that sliver of OBS blocks all planar access to the pin and should not be there.

closed 2022-09-19 by donn