The-OpenROAD-Project/OpenLane #1347: Openlane flow fails in LVS Stage

dineshannayya, 2022-09-10:
## Description
This block is picked from caravel design and I am facing LVS Issue is specific to this block. I have check the LVS in lef and gds option, both are showing similar failure

LVS reports:
    net count difference = 175
    device count difference = 65
    unmatched nets = 21
    unmatched devices = 48
    unmatched pins = 0
    property failures = 0

Total errors = 309

## Expected behavior
Expected clean LVS

## Environment
<pre><font color="#4E9A06"><b>dinesha@lenovo-i3-10100-07IMB05</b></font>:<font color="#3465A4"><b>~/workarea/efabless/MPW-7/OpenLane</b></font>$ python3 ./env.py issue-survey
Kernel: Linux v5.15.0-46-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: b6bacc9d1ab469917fda7ceea61ea3a18984b818
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

b6bacc9 2022-08-24T21:21:29+02:00 Build Script Fixes (#1283) - Mohamed Gaber -  (HEAD -&gt; master)
f0f6055 2022-08-24T19:36:18+02:00 Remove `-echo` flags from `read_sdc` (#1284) - Anton Blanchard -  ()
9452129 2022-08-20T13:41:59+02:00 Fix &apos;run_yosys&apos; invocation during -synth_explore (#1278) - meta -  ()
---
Git Remotes

origin	https://github.com/efabless/OpenLane.git (fetch)
origin	https://github.com/efabless/OpenLane.git (push)
</pre>
## Reproduction Material
Here is small test case 
[test.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9539522/test.tar.gz)

[34-digital_pll.lef.log](https://github.com/The-OpenROAD-Project/OpenLane/files/9539523/34-digital_pll.lef.log)
[34-digital_pll.lvs.lef.log](https://github.com/The-OpenROAD-Project/OpenLane/files/9539524/34-digital_pll.lvs.lef.log)

cd test/openlane 
make digital_pll

## Logs
<pre>[WARNING]: Skipping Tap/Decap Insertion.
    while executing
&quot;flow_fail&quot;
    (procedure &quot;quit_on_lvs_error&quot; line 12)
    invoked from within
&quot;quit_on_lvs_error -log $count_lvs_log&quot;
    (procedure &quot;run_lvs&quot; line 79)
    invoked from within
&quot;run_lvs&quot;
    (procedure &quot;run_lvs_step&quot; line 10)
    invoked from within
&quot;[lindex $step_exe 0] [lindex $step_exe 1] &quot;
    (procedure &quot;run_non_interactive_mode&quot; line 52)
    invoked from within
&quot;run_non_interactive_mode {*}$argv&quot;
    invoked from within
&quot;if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor...&quot;
    (file &quot;/openlane/flow.tcl&quot; line 401)
make: *** [Makefile:44: digital_pll] Error 1
</pre>```

d-m-bailey, 2022-09-10:
@dineshannayya There are no `fill`,  `decap`, or `vpwrvgndtap` cells in the layout.

dineshannayya, 2022-09-10:
@d-m-bailey  Sorry, I wrongly copied some of the constraints from top-level.  We can close this issue

closed 2022-09-10 by dineshannayya

hello-eternity, 2022-09-10:
@dineshannayya @d-m-bailey  Sorry，I met the same error and I have little experience of this situation. I check the detail log of lvs, it certainly because the decap cells in the layout, but how can I interfere with this process, please give me some specific methods. thank you very much. Do I need to check some specific parameter or check my design.

d-m-bailey, 2022-09-10:
@hello-eternity Can you post your log files on the slack #lvs channel?