The-OpenROAD-Project/OpenLane #1348: DRC failed with ```in area of metal1 holes > 0.14um^2```

whutddk, 2022-09-10:
## Description
I failed at DRC stage now, and got 
```
[ERROR]: There are violations in the design after Magic DRC.
[ERROR]: Total Number of violations is 3
```

## Expected behavior

DRC clean
## Environment

```
Kernel: Linux v5.10.102.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.4 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)

```

## Reproduction Material

## Logs

```
[STEP 31]
[INFO]: Running Multi-Corner Static Timing Analysis at the nom process corner...
[STEP 32]
[INFO]: Running Magic to generate various views...
[INFO]: Streaming out GDS-II with Magic...
[INFO]: Generating MAGLEF views...
[STEP 33]
[INFO]: Streaming out GDS-II with Klayout...
[STEP 34]
[INFO]: Running XOR on the layouts using Klayout...
[STEP 35]
[INFO]: Running Magic Spice Export from LEF...
[STEP 36]
[INFO]: Writing Powered Verilog...
[STEP 37]
[INFO]: Writing Verilog...
[STEP 38]
[INFO]: Running LEF LVS...
[STEP 39]
[INFO]: Running Magic DRC...
[INFO]: Converting Magic DRC Violations to Magic Readable Format...
[INFO]: Converting Magic DRC Violations to Klayout XML Database...
[ERROR]: There are violations in the design after Magic DRC.
[ERROR]: Total Number of violations is 3
[INFO]: Saving current set of views in '../mnt/r/work/Rift2Go_2330/openlane/user_proj_nsram/runs/22_09_10_07_31/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at '../mnt/r/work/Rift2Go_2330/openlane/user_proj_nsram/runs/22_09_10_07_31/reports/manufacturability.rpt'.
[INFO]: Created metrics report at '../mnt/r/work/Rift2Go_2330/openlane/user_proj_nsram/runs/22_09_10_07_31/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "quit_on_magic_drc" line 13)
    invoked from within
"quit_on_magic_drc -log $::env(drc_prefix).tr"
    (procedure "run_magic_drc" line 26)
    invoked from within
"run_magic_drc"
    (procedure "run_drc_step" line 9)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 57)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "/openlane/flow.tcl" line 389)
make[1]: *** [Makefile:72: user_proj_nsram] Error 1
make[1]: Leaving directory '/mnt/r/work/Rift2Go_2330/openlane'
make: *** [Makefile:73: user_proj_nsram] Error 2

```

whutddk, 2022-09-10:
[39-drc.log](https://github.com/The-OpenROAD-Project/OpenLane/files/9539785/39-drc.log)

vijayank88, 2022-09-10:
@whutddk 
Its know issue with mpw-7a tag and its because of `open_pdks`. Update OpenLane to lateset and run the flow will resolve this issue.

closed 2022-09-11 by whutddk