The-OpenROAD-Project/OpenLane #1353: LVS failed, SRAM power mismatch

whutddk, 2022-09-12:
## Description

My flow always failed at LVS, when I used SRAM macro.

`Circuit 1 contains 1462 nets,    Circuit 2 contains 1446 nets. *** MISMATCH ***`
when I used 8 sram macro, there will be 16 net mismatch. 

## Expected behavior

Pass LVS

## Environment

```
YOUR SURVEY HERE
```

## Reproduction Material

## Logs

```
YOUR LOGS HERE
```

[log.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9544023/log.zip)

whutddk, 2022-09-12:
Is there any mistake I make?

d-m-bailey, 2022-09-12:
Your macros are not connected to power.

From your config.tcl file
```
set ::env(FP_PDN_MACRO_HOOKS) "\
        i_Rift2Wrap vccd1 vssd1 vccd1 vssd1 \
        i_icacheTag_0 vccd1 vssd1 \
        i_icacheTag_1 vccd1 vssd1 \
        i_icacheDat_0 vccd1 vssd1 \
        i_icacheDat_1 vccd1 vssd1 \
        i_dcacheTag_0 vccd1 vssd1 \
        i_dcacheTag_1 vccd1 vssd1 \
        i_dcacheDat_0 vccd1 vssd1 \
        i_dcacheDat_1 vccd1 vssd1 "
```
This should probably be (don't forget the commas)
```
set ::env(FP_PDN_MACRO_HOOKS) "\
        i_Rift2Wrap vccd1 vssd1 vccd1 vssd1, \
        i_icacheTag_0 vccd1 vssd1 vccd1 vssd1, \
        i_icacheTag_1 vccd1 vssd1 vccd1 vssd1, \
        i_icacheDat_0 vccd1 vssd1 vccd1 vssd1, \
        i_icacheDat_1 vccd1 vssd1 vccd1 vssd1, \
        i_dcacheTag_0 vccd1 vssd1 vccd1 vssd1, \
        i_dcacheTag_1 vccd1 vssd1 vccd1 vssd1, \
        i_dcacheDat_0 vccd1 vssd1 vccd1 vssd1, \
        i_dcacheDat_1 vccd1 vssd1 vccd1 vssd1 "
```
From `openlane/configuration/README.md`
```
FP_PDN_MACRO_HOOKS	
Specifies explicit power connections of internal macros to the top level power grid. Comma separated list of macro instance names, power domain vdd and ground net names, and macro vdd and ground pin names: <instance_name> <vdd_net> <gnd_net> <vdd_pin> <gnd_pin>
```
Is there something in your run logs about not being able to find power pins?
Maybe grep `Power planning the following nets` and look around there.

whutddk, 2022-09-12:
Thank you, I pass LVS now. But I got DRC Violations now, just next step! I will close this issue and open a new one.

closed 2022-09-12 by whutddk