The-OpenROAD-Project/OpenLane #1354: DRC failed with `Local interconnect spacing < 0.17um (li.3)`

whutddk, 2022-09-12:
## Description

I got DRC failed when harden the wrapper. and got `Local interconnect spacing < 0.17um (li.3)`, should I ignore it and submit it to MPW-7?

## Expected behavior

DRC clean
## Environment

```
YOUR SURVEY HERE
```

## Reproduction Material

## Logs

[config.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9544845/config.zip)

vijayank88, 2022-09-12:
Issue template is mandatory for filing github issue to understand your current OpenLane setup:
```
 This part is incredibly important:

    Please run the following shell command in the OpenLane root folder:

        python3 ./env.py issue-survey

    Then copy and paste the ENTIRE output between the triple-backticks below. (```)

    If the command does not succeed, you are using an out-of-date version of
    OpenLane, and it is recommended that you update.
```

vijayank88, 2022-09-12:
Its continuation of #1353.

You've to add `GRT_OBS` for the SRAM macros.

whutddk, 2022-09-12:
How to set `GRT_OBS`?
I didn't find it in [doc](https://openlane-docs.readthedocs.io/en/rtd-develop/configuration/README.html)

vijayank88, 2022-09-12:
> Issue template is mandatory for filing github issue to understand your current OpenLane setup:
>
> ```
>  This part is incredibly important:
>
>     Please run the following shell command in the OpenLane [...]

complete this issue template to know which OpenLane currently you're using.

whutddk, 2022-09-12:
i have updated，haven't I？

vijayank88, 2022-09-12:
https://openlane-docs.readthedocs.io/en/rtd-develop/configuration/README.html#routing 
`GLB_RT_OBS`

whutddk, 2022-09-12:
I failed again by adding 
```
set ::env(GLB_RT_OBS)  "
	                li1   40 3150  361  3308,\
	                met1  40 3150  361  3308,\
	                met2  40 3150  361  3308,\
                    met3  40 3150  361  3308,\
                    met4  40 3150  361  3308,\

	                li1   2560 3150 2881 3308,\
	                met1  2560 3150 2881 3308,\
	                met2  2560 3150 2881 3308,\
                    met3  2560 3150 2881 3308,\
                    met4  2560 3150 2881 3308,\

                    li1   430 3150 1434 3372,\
                    met1  430 3150 1434 3372,\
                    met2  430 3150 1434 3372,\
                    met3  430 3150 1434 3372,\
                    met4  430 3150 1434 3372,\

	                li1   1470 3150 2474 3372,\
	                met1  1470 3150 2474 3372,\
	                met2  1470 3150 2474 3372,\
                    met3  1470 3150 2474 3372,\
                    met4  1470 3150 2474 3372,\

	                li1   40  80  361 238,\
	                met1  40  80  361 238,\
	                met2  40  80  361 238,\
                    met3  40  80  361 238,\
                    met4  40  80  361 238,\

                    li1   2560  80 2881 238,\
                    met1  2560  80 2881 238,\
                    met2  2560  80 2881 238,\
                    met3  2560  80 2881 238,\
                    met4  2560  80 2881 238,\

                    li1   430  80  1434 302,\
	                met1  430  80  1434 302,\
	                met2  430  80  1434 302,\
                    met3  430  80  1434 302,\
                    met4  430  80  1434 302,\

                    li1   1470  80 2474 302,\
                    met1  1470  80 2474 302,\
                    met2  1470  80 2474 302,\
                    met3  1470  80 2474 302,\
                    met4  1470  80 2474 302
	                "
```
I see it report that li.3 is interconnect
```
user_project_wrapper
----------------------------------------
Local interconnect spacing < 0.17um (li.3)
----------------------------------------
 162.450um 180.550um 162.605um 180.635um
 162.620um 180.465um 162.775um 180.550um
 162.450um 180.465um 162.605um 180.635um
```
but I cannot see li3 at openroad gui mode, I only see li1.
So, what should I modify next?
@vijayank88

vijayank88, 2022-09-12:
As per slack channel discussion if you're using `SRAM` macros, user will skip magic DRC check with `set ::env(QUIT_ON_MAGIC_DRC_ERROR) 0`.

Still you need to attach test case for this error to debug. 

Wait for @RTimothyEdwards to look into this issue.

whutddk, 2022-09-12:
I upload the whole project to [github](https://github.com/whutddk/Rift2Go_2301)
```
make user_proj_sramout2
make user_project_wrapper
```
can see the DRC failed. 
However, it will cost you too much time, is there any faster ways?

whutddk, 2022-09-12:
skip magic DRC check will failed the MPW-preCheck

closed 2022-09-13 by whutddk

LvmhZhD, 2022-09-28:
> skip magic DRC check will failed the MPW-preCheck

hi,have you got the solution?

whutddk, 2022-09-29:
@LvmhZhD Not yet. For the MPW-7 is closed, I am focused on improving my design now. But I have found a solution in [slack chn](https://open-source-silicon.slack.com/archives/C016UL7AQ73/p1633441652163500?thread_ts=1633433046.163200&cid=C016UL7AQ73). I am not sure whether will it work, and whether will it work for self-design SRAM. I will try that when MPW-8 opened.

LvmhZhD, 2022-09-29:
> @LvmhZhD Not yet. For the MPW-7 is closed, I am focused on improving my design now. But I have found a solution in [slack [...]

Ok,  thank you