The-OpenROAD-Project/OpenLane #1358: pdngen failing for gf180mcuC

proppy, 2022-09-13:
## Description

Trying to run:
```
export PDK_ROOT=/usr/local/share/pdk
export PDK=gf180mcuC
export STD_CELL_LIBRARY=gf180mcu_fd_sc_mcu7t5v0
export STD_CELL_LIBRARY_OPT=gf180mcu_fd_sc_mcu7t5v0

flow.tcl -ignore_mismatches -design inverter
```
with a manual install of https://github.com/efabless/open_pdks trigger the following error in PDNGEN:
```
[ERROR PDN-0190] Unable to determine the pitch of the rows.
```

## Expected behavior

Flow proceed to global placement.

## Environment

```
```

## Reproduction Material

https://colab.research.google.com/gist/proppy/ef19351b5b9a5db95166d705af7bae39/gf180mcu-openlane-playground.ipynb#scrollTo=5hzngAqWcNlW

## Logs
```
[ERROR]: during executing openroad script /content/OpenLane/scripts/openroad/pdn.tcl
[ERROR]: Log: runs/RUN_2022.09.13_07.23.39/logs/floorplan/6-pdn.log
[ERROR]: Last 10 lines:
[INFO ODB-0226] Finished LEF file:  /content/runs/RUN_2022.09.13_07.23.39/tmp/merged.nom.lef
[INFO ODB-0127] Reading DEF file: /content/runs/RUN_2022.09.13_07.23.39/results/floorplan/inverter.def
[INFO ODB-0128] Design: inverter
[INFO ODB-0130]     Created 2 pins.
[INFO ODB-0131]     Created 1 components and 4 component-terminals.
[INFO ODB-0133]     Created 2 nets and 2 connections.
[INFO ODB-0134] Finished DEF file: /content/runs/RUN_2022.09.13_07.23.39/results/floorplan/inverter.def
[ERROR PDN-0190] Unable to determine the pitch of the rows.
Error: pdn_cfg.tcl, 139 PDN-0190
child process exited abnormally
```

vijayank88, 2022-09-13:
@arlpetergadfort
Can you please check this?

kareefardi, 2022-09-13:
The def doesn't have rows which indicates an issue at an earlier stage. I don't think this has to do with pdn specifically

vijayank88, 2022-09-13:
@proppy 
Check `PLACE_SITE` variable in PDK `config.tcl`

proppy, 2022-09-13:
@vijayank88 thanks!

adding those definitions similar to what's in https://github.com/RTimothyEdwards/open_pdks/blob/master/gf180mcu/openlane/gf180mcu_fd_sc_mcu7t5v0/config.tcl (and not in the efabless' fork: https://github.com/efabless/open_pdks/blob/master/gf180mcu/openlane/gf180mcu_fd_sc_mcu7t5v0/config.tcl):
```
# Placement site for core cells
# This can be found in the technology lef
set ::env(PLACE_SITE) "GF018hv5v_mcu_sc7"
set ::env(PLACE_SITE_WIDTH) 0.56
set ::env(PLACE_SITE_HEIGHT) 3.92
```
allows me to get past floorplan/pdn.

Then added this to pass global placement:
```
set ::env(GPL_CELL_PADDING) {0}
set ::env(DPL_CELL_PADDING) {4}
```

It then fails on detailed placement:
```
[ERROR]: during executing openroad script /content/OpenLane/scripts/openroad/resizer.tcl
[ERROR]: Log: runs/RUN_2022.09.13_13.44.51/logs/placement/8-resizer.log
[ERROR]: Last 10 lines:
 TAP_183 overlaps TAP_177
 TAP_184 overlaps TAP_179
 TAP_559 overlaps TAP_554
 TAP_561 overlaps TAP_555
 TAP_563 overlaps TAP_556
 TAP_565 overlaps TAP_557
 TAP_567 overlaps TAP_558
[ERROR DPL-0033] detailed placement checks failed.
DPL-0033
child process exited abnormally
```

proppy, 2022-09-13:
will open a new issue to discuss detailed placement.

closed 2022-09-13 by proppy