The-OpenROAD-Project/OpenLane #1362: hot to work with multiple verilog files

IngYordiDelgado, 2022-09-14:
### Discussed in https://github.com/The-OpenROAD-Project/OpenLane/discussions/1346

<div type='discussions-op-text'>

<sup>Originally posted by **IngYordiDelgado** September  9, 2022</sup>
i tried to test a project with multiple verilog files, i put this files on a directory called rtl in the configuration file i set the verilog file to the path of my top level file and its name and im getting the following error 
[ERROR]: during executing: "yosys -c /openlane/scripts/yosys/synth.tcl -l /openlane/designs/gcd/runs/RUN_2022.09.10_00.07.58/logs/synthesis/1-synthesis.log |& tee /dev/null"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:

2. Generating Graphviz representation of design.
Writing dot description to `/openlane/designs/gcd/runs/RUN_2022.09.10_00.07.58/tmp/synthesis/hierarchy.dot'.
Dumping module gcd to page 1.

3. Executing HIERARCHY pass (managing design hierarchy).

3.1. Analyzing design hierarchy..
ERROR: Module `\gcd_control' referenced in module `\gcd' in cell `\GCDctrl0' is not part of the design.
child process exited abnormally

</div>

vijayank88, 2022-09-15:
@IngYordiDelgado 

Follow issue template guidelines. Missing many details here, without it won't get resolved.

`GCDctrl0` module missing in the file list. Share your `config.tcl`

IngYordiDelgado, 2022-09-15:
this is my configuration file `{
    "DESIGN_NAME": "gcd",
    "VERILOG_FILES": "dir::rtl/src/gcd.v",
    "CLOCK_PORT": "clk",
    "SYNTH_STRATEGY": "DELAY 4",
    "SYNTH_SHARE_RESOURCES": false,
    "FP_IO_VLENGTH": null,
    "FP_IO_HLENGTH": null,
    "FP_IO_VTHICKNESS_MULT": null,
    "FP_IO_HTHICKNESS_MULT": null,
    "FP_IO_MIN_DISTANCE": null,
    "FP_SIZING": "absolute",
    "pdk::asap7": {
        "scl::asap7sc7p5t": {
            "RUN_MAGIC": false,
            "RUN_MAGIC_DRC": false,
            "RUN_LVS": false,
            "PRIMARY_SIGNOFF_TOOL": "klayout",
            "CLOCK_PERIOD": "420",
            "DIE_AREA": "0 0 16.2 16.2",
            "PLACE_DENSITY": 0.35,
            "LEFT_MARGIN_MULT": 20,
            "RIGHT_MARGIN_MULT": 20
        }
    },
    "pdk::sky130*": {
        "scl::sky130_fd_sc_hd": {
            "CLOCK_PERIOD": 10,
            "FP_SIZING": "absolute",
            "DIE_AREA": "0 0 279.96 280.128",
            "PLACE_DENSITY": 0.35,
            "LEFT_MARGIN_MULT": 22,
            "RIGHT_MARGIN_MULT": 22,
            "TOP_MARGIN_MULT": 3,
            "FP_PDN_VPITCH": 27.14,
            "FP_PDN_HPITCH": 27.20,
            "FP_PDN_VOFFSET": 13.57,
            "FP_PDN_HOFFSET": 13.60
        },
        "scl::sky130_fd_sc_hs": {
            "CLOCK_PERIOD": 2.4,
            "FP_SIZING": "absolute",
            "DIE_AREA": "0 0 279.96 280.128",
            "PLACE_DENSITY": 0.35,
            "LEFT_MARGIN_MULT": 22,
            "RIGHT_MARGIN_MULT": 22,
            "TOP_MARGIN_MULT": 3,
            "FP_PDN_VPITCH": 27.14,
            "FP_PDN_HPITCH": 27.20,
            "FP_PDN_VOFFSET": 13.57,
            "FP_PDN_HOFFSET": 13.60,
            "DIODE_INSERTION_STRATEGY": 0
        }
    }
}`

closed 2022-09-19 by donn