The-OpenROAD-Project/OpenLane #1364: DPL fails for gf180mcuC

proppy, 2022-09-15:
## Description

Trying to run:
```
export PDK_ROOT=/usr/local/share/pdk
export PDK=gf180mcuC
export STD_CELL_LIBRARY=gf180mcu_fd_sc_mcu7t5v0
export STD_CELL_LIBRARY_OPT=gf180mcu_fd_sc_mcu7t5v0

flow.tcl -ignore_mismatches -design spm
```
with a manual install of https://github.com/efabless/open_pdks with https://github.com/efabless/open_pdks/pull/3/files applied, trigger the following error in DPL:
```
[ERROR]: during executing openroad script /content/OpenLane/scripts/openroad/resizer.tcl
[ERROR]: Log: OpenLane/designs/spm/runs/RUN_2022.09.15_05.14.56/logs/placement/8-resizer.log
[ERROR]: Last 10 lines:
[INFO DPL-0035]  _557_
[INFO DPL-0035]  _559_
[INFO DPL-0035]  _547_
[INFO DPL-0035]  _540_
[INFO DPL-0035]  _554_
[INFO DPL-0035]  _552_
[INFO DPL-0035]  _551_
[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 76 DPL-0036
child process exited abnormally
```

## Expected behavior

Flow proceed to routing.

## Environment

```
Kernel: Linux v5.10.133+
Distribution: ubuntu 18.04
Python: v3.7.13 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: 37faafee20ec76a349fb817d7a75ed26d94be904
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

37faafe 2022-09-12T14:58:32+02:00 Startup Fixes (#1344) - Mohamed Gaber -  (grafted, HEAD -> master, tag: 2022.09.13, origin/master, origin/HEAD)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

## Reproduction Material

https://colab.sandbox.google.com/gist/proppy/ef19351b5b9a5db95166d705af7bae39/gf180mcu-openlane-playground.ipynb#scrollTo=5hzngAqWcNlW
[issue_reproducible-spm-gf180mcuC.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9572050/issue_reproducible-spm-gf180mcuC.tar.gz)

## Logs
```
OpenROAD da4a559551560a0b9a4875d6bb9d73a9567e6c9a 
This program is licensed under the BSD-3 license. See the LICENSE file for details.
Components of this program may be licensed under more restrictive licenses which must be honored.
[INFO ODB-0222] Reading LEF file: /content/OpenLane/designs/spm/runs/RUN_2022.09.15_05.14.56/tmp/merged.nom.lef
[INFO ODB-0223]     Created 13 technology layers
[INFO ODB-0224]     Created 60 technology vias
[INFO ODB-0225]     Created 229 library cells
[INFO ODB-0226] Finished LEF file:  /content/OpenLane/designs/spm/runs/RUN_2022.09.15_05.14.56/tmp/merged.nom.lef
[INFO ODB-0127] Reading DEF file: /content/OpenLane/designs/spm/runs/RUN_2022.09.15_05.14.56/tmp/placement/7-global.def
[INFO ODB-0128] Design: spm
[INFO ODB-0130]     Created 38 pins.
[INFO ODB-0131]     Created 785 components and 2616 component-terminals.
[INFO ODB-0132]     Created 2 special nets and 1570 connections.
[INFO ODB-0133]     Created 384 nets and 1046 connections.
[INFO ODB-0134] Finished DEF file: /content/OpenLane/designs/spm/runs/RUN_2022.09.15_05.14.56/tmp/placement/7-global.def
[INFO]: Setting RC values...
[INFO RSZ-0027] Inserted 34 input buffers.
[INFO RSZ-0028] Inserted 1 output buffers.
[INFO RSZ-0058] Using max wire length 22815um.
[INFO RSZ-0039] Resized 157 instances.
[INFO DPL-0034] Detailed placement failed on:
[INFO DPL-0035]  _594_
[INFO DPL-0035]  _598_
[INFO DPL-0035]  _589_
[INFO DPL-0035]  _590_
[INFO DPL-0035]  _587_
[INFO DPL-0035]  _544_
[INFO DPL-0035]  _579_
[INFO DPL-0035]  _577_
[INFO DPL-0035]  _568_
[INFO DPL-0035]  _570_
[INFO DPL-0035]  _543_
[INFO DPL-0035]  _576_
[INFO DPL-0035]  _574_
[INFO DPL-0035]  _588_
[INFO DPL-0035]  _545_
[INFO DPL-0035]  _553_
[INFO DPL-0035]  _539_
[INFO DPL-0035]  _583_
[INFO DPL-0035]  _575_
[INFO DPL-0035]  _593_
[INFO DPL-0035]  _565_
[INFO DPL-0035]  _541_
[INFO DPL-0035]  _546_
[INFO DPL-0035]  _571_
[INFO DPL-0035]  _578_
[INFO DPL-0035]  _581_
[INFO DPL-0035]  _573_
[INFO DPL-0035]  _596_
[INFO DPL-0035]  _600_
[INFO DPL-0035]  _566_
[INFO DPL-0035]  _542_
[INFO DPL-0035]  _602_
[INFO DPL-0035]  _580_
[INFO DPL-0035]  _585_
[INFO DPL-0035]  _550_
[INFO DPL-0035]  _595_
[INFO DPL-0035]  _582_
[INFO DPL-0035]  _556_
[INFO DPL-0035]  _555_
[INFO DPL-0035]  _601_
[INFO DPL-0035]  _584_
[INFO DPL-0035]  _567_
[INFO DPL-0035]  _564_
[INFO DPL-0035]  _592_
[INFO DPL-0035]  _597_
[INFO DPL-0035]  _560_
[INFO DPL-0035]  _599_
[INFO DPL-0035]  _549_
[INFO DPL-0035]  _562_
[INFO DPL-0035]  _569_
[INFO DPL-0035]  _572_
[INFO DPL-0035]  _591_
[INFO DPL-0035]  _558_
[INFO DPL-0035]  _586_
[INFO DPL-0035]  _561_
[INFO DPL-0035]  _563_
[INFO DPL-0035]  _548_
[INFO DPL-0035]  _557_
[INFO DPL-0035]  _559_
[INFO DPL-0035]  _547_
[INFO DPL-0035]  _540_
[INFO DPL-0035]  _554_
[INFO DPL-0035]  _552_
[INFO DPL-0035]  _551_
[ERROR DPL-0036] Detailed placement failed.
Error: resizer.tcl, 76 DPL-0036
```

maliberty, 2022-09-15:
The ff are quite large but the look placeable:
[image]

kareefardi, 2022-09-15:
the rows have half the site height which messes up a lot of things. one of them if the tap and decap placement which triggers the error you are seeing above, but this isn't the true issue. the behavior that is causing wrong rows definition is having the `SITE` at the very top of the tech lef. my guess is the `UNITS` `DATABASE MICRONS 2000` set after the `SITE` definition messes up the calculation done by openroad's floorplanner (unless there is a lef specification for that). I would try again with a techlef where the `SITE` definition comes up after the `UNITS` definition until we can confirm that this is a bug with openroad

kareefardi, 2022-09-15:
The lef/def reference states the following:
```
The UNITS statement is optional and, when used, must precede the LAYER statements.
```
It doesn't look like it a specification.

proppy, 2022-09-16:
@kareefardi thanks!

applying the changes you suggested in https://github.com/The-OpenROAD-Project/OpenLane/issues/1364#issuecomment-1247751515 (moving `SITE` def after `UNITS`) get me up to DRT:
https://github.com/The-OpenROAD-Project/OpenLane/issues/1367

proppy, 2022-09-16:
fixed with https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu9t5v0/pull/18 and https://github.com/google/globalfoundries-pdk-libs-gf180mcu_fd_sc_mcu7t5v0/pull/21, but as @kareefardi pointed out it would be good to have either better validation or tolerance to field ordering in OpenROAD.

closed 2022-09-16 by proppy

jjcherry56, 2022-09-18:
Tolerating UNITS after SITE makes no sense because the SITE dimensions use the units value and it would violate the LEF spec (LEF/DEF 5.8 Language Reference pg 12). The LEF parser should error when SITE appears before UNITS.

kareefardi, 2022-09-18:
Missed that in the reference. Thanks

proppy, 2022-09-18:
@jjcherry56 would it make sense to file a validation in OpenLane, or should that be in OpenROAD?

jjcherry56, 2022-09-18:
The LEF parser is in OR, so that is what should be doing the check.

maliberty, 2022-09-18:
@jjcherry56 in the Si2 LEF parser: https://github.com/The-OpenROAD-Project/OpenROAD/blob/b7631451350809842e4fb0c635c3f3ed7f6b270f/src/odb/src/lef/lef/lef.y#L599

shows that it is legal though I don't know what it is supposed to mean.  It seems very odd...

jjcherry56, 2022-09-18:
That parser makes no sense and violates the spec for LEF 5.6, 5.7 and 5.8. I think the author may have gotten the comparison backwards.

maliberty, 2022-09-19:
Let's just remove the version check as I don't think we ever want to accept it.