The-OpenROAD-Project/OpenLane #1366: Placement resizer timing not met with OpenLane tag 2022.09.13 and above

vijayank88, 2022-09-16:
## Description
MPW design which met timing till OpenLane tag 2022.09.08. But failed to meet timing OpenLane tag 2022.09.13 and above.

## Expected behavior
Fix timing issue.

## Environment

```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 60f0172cfb39445062dd519749605b50c66e24ef
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

60f0172 2022-09-08T11:55:49+05:30 OpenLane stable tag update 2022.09.05 (#80) - vijayan -  (HEAD -> main, origin/main, origin/HEAD)
2232885 2022-08-31T08:39:40-03:00 Update OpenLane stable tag -> 2022.08.25 (#78) - vijayan -  ()
437eb8e 2022-08-22T17:11:06-03:00 OpenLane tag update to 2022.08.21 (#76) - vijayan -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI.git (push)

```

## Reproduction Material

MPW CI passed: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Fproduction-large-1/detail/main/31/pipeline/456
Test case 1: Timing met or less TNS 
[issue_reproducible_pass.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9583935/issue_reproducible_pass.zip)

MPW CI Failed: https://jenkins.openroad.tools/blue/organizations/jenkins/OpenLane-MPW-CI-Public%2Fproduction-large-1/detail/PR-83/1/pipeline/467
Test case 2: Timing not met  
[issue_reproducible_fail.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9583923/issue_reproducible_fail.zip)

## Logs

```
===========================================================================
 report_tns
============================================================================
tns -13.13
tns_report_end
wns_report

===========================================================================
 report_wns
============================================================================
wns -3.83
wns_report_end
worst_slack

===========================================================================
 report_worst_slack -max (Setup)
============================================================================
worst slack -3.83

===========================================================================
 report_worst_slack -min (Hold)
============================================================================
worst slack -0.48
worst_slack_end
clock_skew

```

The difference iI found is `DONT_USE_CELLS` list.

vijayank88, 2022-09-17:
@jjcherry56 
whether resizer can fix this timing violation or needs update in `DONT_USE_CELLS`?

antonblanchard, 2022-09-17:
Why are we preventing the resizer from using delay cells? Sounds like an Openlane bug to me.

vijayank88, 2022-09-17:
https://github.com/The-OpenROAD-Project/OpenLane/commit/37faafee20ec76a349fb817d7a75ed26d94be904 this commit changes the optimization library list. @antonblanchard

antonblanchard, 2022-09-18:
> [37faafe](https://github.com/The-OpenROAD-Project/OpenLane/commit/37faafee20ec76a349fb817d7a75ed26d94be904) this commit changes the optimization library list. @antonblanchard

@donn was this change meant to prevent the resizer from using delay buffers?

maliberty, 2022-09-19:
@antonblanchard where in that commit is the change to the delay buffers?  I'm not seeing it

antonblanchard, 2022-09-19:
@maliberty I'm going off the screen shot that @vijayank88 pasted into the bug, where `DONT_USE_CELLS` now includes `sky_fd_sc_hd__dly*`

`DONT_USE_CELLS` is defined as `The list of cells to not use during resizer optimizations.`

maliberty, 2022-09-19:
@vijayank88 is which commit do you see the change in DONT_USE_CELLS?

vijayank88, 2022-09-19:
> [37faafe](https://github.com/The-OpenROAD-Project/OpenLane/commit/37faafee20ec76a349fb817d7a75ed26d94be904) this commit changes the optimization library list. @antonblanchard

@maliberty

maliberty, 2022-09-19:
@vijayank88 what line in that commit are you referring to when you say the DONT_USE list changed?

vijayank88, 2022-09-19:
`scripts/config/extract_opt_variables.py`  and `scripts/tcl_commands/all.tcl` are updated. 
i'm not tested in depth. Just compared two versions resizer logs and found changes wrt. `DONT_USE_CELLS`
I think @donn can give quick overview on the changes.

maliberty, 2022-09-19:
@vijayank88 I just tried with 2022.09.16 and I don't see those cells in my dont use list for spm:
```
set ::env(DONT_USE_CELLS) {sky130_fd_sc_hd__a2111oi_0 sky130_fd_sc_hd__a21boi_0 sky130_fd_sc_hd__and2_0 sky130_fd_sc_hd__buf_16 sky130_fd_sc_hd__clkdlybuf4s15_1 sky130_fd_sc_hd__clkdlybuf4s18_1 sky130_fd_sc_hd__fa_4 sky130_fd_sc_hd__lpflow_bleeder_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_16 sky130_fd_sc_hd__lpflow_clkbufkapwr_2 sky130_fd_sc_hd__lpflow_clkbufkapwr_4 sky130_fd_sc_hd__lpflow_clkbufkapwr_8 sky130_fd_sc_hd__lpflow_clkinvkapwr_1 sky130_fd_sc_hd__lpflow_clkinvkapwr_16 sky130_fd_sc_hd__lpflow_clkinvkapwr_2 sky130_fd_sc_hd__lpflow_clkinvkapwr_4 sky130_fd_sc_hd__lpflow_clkinvkapwr_8 sky130_fd_sc_hd__lpflow_decapkapwr_12 sky130_fd_sc_hd__lpflow_decapkapwr_3 sky130_fd_sc_hd__lpflow_decapkapwr_4 sky130_fd_sc_hd__lpflow_decapkapwr_6 sky130_fd_sc_hd__lpflow_decapkapwr_8 sky130_fd_sc_hd__lpflow_inputiso0n_1 sky130_fd_sc_hd__lpflow_inputiso0p_1 sky130_fd_sc_hd__lpflow_inputiso1n_1 sky130_fd_sc_hd__lpflow_inputiso1p_1 sky130_fd_sc_hd__lpflow_inputisolatch_1 sky130_fd_sc_hd__lpflow_isobufsrc_1 sky130_fd_sc_hd__lpflow_isobufsrc_16 sky130_fd_sc_hd__lpflow_isobufsrc_2 sky130_fd_sc_hd__lpflow_isobufsrc_4 sky130_fd_sc_hd__lpflow_isobufsrc_8 sky130_fd_sc_hd__lpflow_isobufsrckapwr_16 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_2 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_4 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_4 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_1 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_2 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_4 sky130_fd_sc_hd__mux4_4 sky130_fd_sc_hd__o21ai_0 sky130_fd_sc_hd__o311ai_0 sky130_fd_sc_hd__or2_0 sky130_fd_sc_hd__probe_p_8 sky130_fd_sc_hd__probec_p_8 sky130_fd_sc_hd__xor3_1 sky130_fd_sc_hd__xor3_2 sky130_fd_sc_hd__xor3_4 sky130_fd_sc_hd__xnor3_1 sky130_fd_sc_hd__xnor3_2 sky130_fd_sc_hd__xnor3_4  }
```

vijayank88, 2022-09-20:
@maliberty 
Have you compared my attachment `issue_reproducible`?

maliberty, 2022-10-25:
Sorry this one seems to have gotten lost.  Is there still an issue?  I don't see any CI failures.

vijayank88, 2022-11-08:
This MPW design got updated and `version2` submitted. No more part of MPW CI regression.

closed 2022-11-08 by vijayank88