The-OpenROAD-Project/OpenLane #1371: [Global Routing Resizer Timing Optimizations] Stage does not fix hold violation, when a input has setup and hold violation in different end point

dineshannayya, 2022-09-18:
## Description
Global Routing Resizer Timing Optimizations - Stage does not fix hold violation, when a input has both setup and hold violation in different end point.
Below example: core_dmem_req_ack => _7275_/D has Hold Violation: -0.34  & core_dmem_req_ack => core2_dmem_req_ack has setup violation: -3.10 .   

<pre>Startpoint: core_dmem_req_ack (input port clocked by core_clk)
Endpoint: _7275_ (rising edge-triggered flip-flop clocked by core_clk)
Path Group: core_clk
Path Type: min

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock core_clk (rise edge)
   0.00    0.00   clock network delay (propagated)
   2.00    2.00 ^ input external delay
   0.01    2.01 ^ core_dmem_req_ack (in)
   0.16    2.17 ^ input687/X (sky130_fd_sc_hd__dlymetal6s2s_1)
   0.00    2.17 ^ _7275_/D (sky130_fd_sc_hd__dfxtp_1)
           2.17   data arrival time

   0.00    0.00   clock core_clk (rise edge)
   2.31    2.31   clock network delay (propagated)
   0.25    2.56   clock uncertainty
   0.00    2.56   clock reconvergence pessimism
           2.56 ^ _7275_/CLK (sky130_fd_sc_hd__dfxtp_1)
  -0.06    2.51   library hold time
           2.51   data required time
---------------------------------------------------------
           2.51   data required time
          -2.17   data arrival time
---------------------------------------------------------
          -0.34   slack (VIOLATED)
</pre>

<pre>Startpoint: core_dmem_req_ack (input port clocked by core_clk)
Endpoint: core2_dmem_req_ack (output port clocked by core_clk)
Path Group: core_clk
Path Type: max

  Delay    Time   Description
---------------------------------------------------------
   0.00    0.00   clock core_clk (rise edge)
   0.00    0.00   clock network delay (propagated)
   6.00    6.00 v input external delay
   0.00    6.00 v core_dmem_req_ack (in)
   0.18    6.18 v input687/X (sky130_fd_sc_hd__dlymetal6s2s_1)
   0.52    6.70 v _3415_/X (sky130_fd_sc_hd__or4_1)
   0.21    6.91 v wire2226/X (sky130_fd_sc_hd__buf_4)
   0.20    7.11 v _3471_/X (sky130_fd_sc_hd__buf_2)
   0.17    7.29 v _3759_/X (sky130_fd_sc_hd__buf_6)
   0.22    7.51 v wire2168/X (sky130_fd_sc_hd__clkbuf_8)
   0.37    7.88 v _3771_/X (sky130_fd_sc_hd__a22o_2)
   0.22    8.10 v output1158/X (sky130_fd_sc_hd__buf_2)
   0.00    8.10 v core2_dmem_req_ack (out)
           8.10   data arrival time

  10.00   10.00   clock core_clk (rise edge)
   0.00   10.00   clock network delay (propagated)
  -0.50    9.50   clock uncertainty
   0.00    9.50   clock reconvergence pessimism
  -4.50    5.00   output external delay
           5.00   data required time
---------------------------------------------------------
           5.00   data required time
          -8.10   data arrival time
---------------------------------------------------------
          -3.10   slack (VIOLATED)
</pre>

## Expected behavior
Tool should have fixed Hold Violation at _7275_/D without impacting setup violation other end point. 

## Environment
<pre><font color="#4E9A06"><b>dinesha@lenovo-i3-10100-07IMB05</b></font>:<font color="#3465A4"><b>~/workarea/efabless/MPW-7/OpenLane</b></font>$ python3 ./env.py issue-survey
Kernel: Linux v5.15.0-46-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: b6bacc9d1ab469917fda7ceea61ea3a18984b818
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

b6bacc9 2022-08-24T21:21:29+02:00 Build Script Fixes (#1283) - Mohamed Gaber -  (HEAD -&gt; master)
f0f6055 2022-08-24T19:36:18+02:00 Remove `-echo` flags from `read_sdc` (#1284) - Anton Blanchard -  ()
9452129 2022-08-20T13:41:59+02:00 Fix &apos;run_yosys&apos; invocation during -synth_explore (#1278) - meta -  ()
---
Git Remotes

origin	https://github.com/efabless/OpenLane.git (fetch)
origin	https://github.com/efabless/OpenLane.git (push)
</pre>

## Reproduction Material
[issue_reproducible.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9594534/issue_reproducible.tar.gz)

## Logs

[16-resizer.log](https://github.com/The-OpenROAD-Project/OpenLane/files/9594575/16-resizer.log)

antonblanchard, 2022-09-18:
The resizer is unable to resolve all the setup violations, and hold fixup is being told not to create setup violations and so can't make progress. Increasing the clock period from 10 to 16 so setup repair can pass fixes the issue.

To get the resizer to ignore setup issues when doing hold repair, you can set:

```
set ::env(GLB_RESIZER_ALLOW_SETUP_VIOS) 1
```

Even with that option, hold repair gives up after a couple of passes. It's not obvious to me why it gave up though.

antonblanchard, 2022-09-19:
> Even with that option, hold repair gives up after a couple of passes. It's not obvious to me why it gave up though.

I see why, because the path doesn't have any place where we have enough setup margin to resolve the hold issues. This fixes it, but I'm not sure it's the best solution. I moved the `allow_setup_violations` checks to ignore all the checks:

```
diff --git a/src/rsz/src/RepairHold.cc b/src/rsz/src/RepairHold.cc
index b52ce9909..9800ed331 100644
--- a/src/rsz/src/RepairHold.cc
+++ b/src/rsz/src/RepairHold.cc
@@ -391,18 +391,18 @@ RepairHold::repairEndHold(Vertex *end_vertex,
             Slew buffer_slews[RiseFall::index_count];
             resizer_->bufferDelays(buffer_cell, load_cap, dcalc_ap,
                                    buffer_delays, buffer_slews);
-            // setup_slack > -hold_slack
-            if (slacks[rise_index_][max_index_] - setup_margin
-                > -(slacks[rise_index_][min_index_] - hold_margin)
-                && slacks[fall_index_][max_index_] - setup_margin
-                > -(slacks[fall_index_][min_index_] - hold_margin)
-                // enough slack to insert the buffer
-                // setup_slack > buffer_delay
-                && (allow_setup_violations
-                    || ((slacks[rise_index_][max_index_] - setup_margin)
-                        > buffer_delays[rise_index_]
-                        && (slacks[fall_index_][max_index_] - setup_margin)
-                        > buffer_delays[fall_index_]))) {
+            if (allow_setup_violations
+                // setup_slack > -hold_slack
+                || ((((slacks[rise_index_][max_index_] - setup_margin)
+                    > (-(slacks[rise_index_][min_index_] - hold_margin)))
+                   && ((slacks[fall_index_][max_index_] - setup_margin)
+                    > (-(slacks[fall_index_][min_index_] - hold_margin))))
+                    // enough slack to insert the buffer
+                    // setup_slack > buffer_delay
+                   && (((slacks[rise_index_][max_index_] - setup_margin)
+                        > buffer_delays[rise_index_])
+                        && ((slacks[fall_index_][max_index_] - setup_margin)
+                        > buffer_delays[fall_index_])))) {
               Vertex *path_load = expanded.path(i + 1)->vertex(sta_);
               Point path_load_loc = db_network_->location(path_load->pin());
               Point drvr_loc = db_network_->location(path_vertex->pin());
```

If this isn't acceptable, perhaps a better solution is to do two passes. The first pass looking for a place to resolve the hold issue without impacting any setup issues, then if `allow_setup_violations` is set, a second pass that just finds a place to insert the hold buffer without any regard to setup violations it might cause.

@jjchery56 any thoughts? To summarize, this test case is unable to resolve all the setup violations but also fails to resolve hold violations even when `-allow_setup_violations` is set. Ideally the SDC should be fixed so that setup fixup can pass, but it seems like we should try harder to resolve the hold violations.

dineshannayya, 2022-09-19:
I feel issue is script is trying to fix hold based on margin in output of the cell. If the output of cell has two path, One have Hold violation and Other path as setup violation. Then better to fix hold at input of the end point.
In My case 
Hold violation:  input687/X  => _7275_/D  : -0.34 
Setup violation: input687/X  => core2_dmem_req_ack :  -3.10 
Better fix is to add Hold buffer at _7275_/D, which will not create any additional setup violation in the other branch.

vijayank88, 2022-09-19:
I think its similar to this issue: https://github.com/The-OpenROAD-Project/OpenLane/issues/1366

Updating `DONT_USE_CELLS` will fix timing violations.

vijayank88, 2022-09-19:
@dineshannayya 
Can you please share timing logs for placement resizer stage?
Possible attach test case for placement stage to confirm the same.

dineshannayya, 2022-09-19:
@vijayank88 My Don't cell list is
set ::env(DONT_USE_CELLS) {sky130_fd_sc_hd__a2111oi_0 sky130_fd_sc_hd__a21boi_0 sky130_fd_sc_hd__and2_0 sky130_fd_sc_hd__buf_16 sky130_fd_sc_hd__clkdlybuf4s15_1 sky130_fd_sc_hd__clkdlybuf4s18_1 sky130_fd_sc_hd__fa_4 sky130_fd_sc_hd__lpflow_bleeder_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_1 sky130_fd_sc_hd__lpflow_clkbufkapwr_16 sky130_fd_sc_hd__lpflow_clkbufkapwr_2 sky130_fd_sc_hd__lpflow_clkbufkapwr_4 sky130_fd_sc_hd__lpflow_clkbufkapwr_8 sky130_fd_sc_hd__lpflow_clkinvkapwr_1 sky130_fd_sc_hd__lpflow_clkinvkapwr_16 sky130_fd_sc_hd__lpflow_clkinvkapwr_2 sky130_fd_sc_hd__lpflow_clkinvkapwr_4 sky130_fd_sc_hd__lpflow_clkinvkapwr_8 sky130_fd_sc_hd__lpflow_decapkapwr_12 sky130_fd_sc_hd__lpflow_decapkapwr_3 sky130_fd_sc_hd__lpflow_decapkapwr_4 sky130_fd_sc_hd__lpflow_decapkapwr_6 sky130_fd_sc_hd__lpflow_decapkapwr_8 sky130_fd_sc_hd__lpflow_inputiso0n_1 sky130_fd_sc_hd__lpflow_inputiso0p_1 sky130_fd_sc_hd__lpflow_inputiso1n_1 sky130_fd_sc_hd__lpflow_inputiso1p_1 sky130_fd_sc_hd__lpflow_inputisolatch_1 sky130_fd_sc_hd__lpflow_isobufsrc_1 sky130_fd_sc_hd__lpflow_isobufsrc_16 sky130_fd_sc_hd__lpflow_isobufsrc_2 sky130_fd_sc_hd__lpflow_isobufsrc_4 sky130_fd_sc_hd__lpflow_isobufsrc_8 sky130_fd_sc_hd__lpflow_isobufsrckapwr_16 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_1 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_2 sky130_fd_sc_hd__lpflow_lsbuf_lh_hl_isowell_tap_4 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_4 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_1 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_2 sky130_fd_sc_hd__lpflow_lsbuf_lh_isowell_tap_4 sky130_fd_sc_hd__mux4_4 sky130_fd_sc_hd__o21ai_0 sky130_fd_sc_hd__o311ai_0 sky130_fd_sc_hd__or2_0 sky130_fd_sc_hd__probe_p_8 sky130_fd_sc_hd__probec_p_8 sky130_fd_sc_hd__xor3_1 sky130_fd_sc_hd__xor3_2 sky130_fd_sc_hd__xor3_4 sky130_fd_sc_hd__xnor3_1 sky130_fd_sc_hd__xnor3_2 sky130_fd_sc_hd__xnor3_4  }

vijayank88, 2022-09-19:
@dineshannayya 
Sorry., I've misunderstood the OpenLane tags. this is not same as mine.

jjcherry56, 2022-09-19:
@dineshannayya is correct. There is a better strategy that fixes the hold violation without affecting the setup violation. Setup repair splits non-violating loads from violating loads and hold repair could do the same. 
@antonblanchard you are correct that the allow_setup_violations needs to move up but you don't need all those extra parens around && clauses.

jjcherry56, 2022-10-11:
fixed in openroad 79313e90d

closed 2022-11-09 by donn