The-OpenROAD-Project/OpenLane #1380: LVS fails for fails for gf180mcuC

proppy, 2022-09-21:
## Description

Trying to run:
```
export PDK_ROOT=/usr/local/share/pdk
export PDK=gf180mcuC
export STD_CELL_LIBRARY=gf180mcu_fd_sc_mcu7t5v0
export STD_CELL_LIBRARY_OPT=gf180mcu_fd_sc_mcu7t5v0

flow.tcl -ignore_mismatches -design spm
```
with https://github.com/proppy/volare-gf180mcu-wip/releases/download/gf180mcu-82d61e2_a897aa3_3d1aed6/default.tar.xz and https://github.com/The-OpenROAD-Project/OpenLane/commit/3ee8cf0da0b5292557e3f3ae398194d65970a6d2 reverted to workaround #1379 fails with the following error in the *LVS* step:
```
[ERROR]: There are LVS errors in the design: See 'OpenLane/designs/spm/runs/RUN_2022.09.21_01.23.13/logs/signoff/35-spm.lvs.lef.log' for details.
```

## Expected behavior

Flow proceed to gds streaming.

## Environment

```
Kernel: Linux v5.10.133+
Distribution: ubuntu 18.04
Python: v3.7.13 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: d375db931edfb03fc42859d590eb3ec799bc6c9e
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

d375db9 2022-09-15T13:35:19+02:00 [BOT] Update magic (#1318) - Openlane Bot -  (HEAD, tag: 2022.09.16)
37faafe 2022-09-12T14:58:32+02:00 Startup Fixes (#1344) - Mohamed Gaber -  (tag: 2022.09.13)
cbb9e6b 2022-09-12T14:49:15+02:00 Documentation Restructure (#1337) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

## Reproduction Material

https://colab.research.google.com/gist/proppy/2c2576e8df29631ed98ac1eac760a458/gf180mcu-openlane-playground.ipynb#scrollTo=5hzngAqWcNlW

[gf180mcu-spm-runs-lvs-issue.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9612329/gf180mcu-spm-runs-lvs-issue.zip)

## Logs

```
LVS reports:
    net count difference = 0
    device count difference = 0
    unmatched nets = 0
    unmatched devices = 0
    unmatched pins = 392
    property failures = 0

Total errors = 392
```

proppy, 2022-09-21:
/cc @mithro

proppy, 2022-09-22:
@RTimothyEdwards do you think this could be related to recent netgen changes? I don't get any LVS errors when using an older version of netgen `1.5.229_0_g153ce0e`:
https://colab.sandbox.google.com/gist/proppy/5322c64a317a584aee81fb77f930152f/gf180mcu-openlane-playground.ipynb
```
[STEP 29]
[INFO]: Running LVS (log: OpenLane/designs/spm/runs/RUN_2022.09.22_05.46.35/logs/signoff/29-lvs.lef.log)...
```
/content/OpenLane/designs/spm/runs/RUN_2022.09.22_05.46.35/logs/signoff/29-lvs.lef.log:
```
Netgen 1.5.229 compiled on Wed Sep 14 16:04:31 UTC 2022
Warning: netgen command 'format' use fully-qualified name '::netgen::format'
Warning: netgen command 'global' use fully-qualified name '::netgen::global'
Generating JSON file result
Reading netlist file /content/OpenLane/designs/spm/runs/RUN_2022.09.22_05.46.35/results/signoff/spm.spice
Reading netlist file /content/OpenLane/designs/spm/runs/RUN_2022.09.22_05.46.35/tmp/signoff/26-spm.pnl.v
Warning:  A case-insensitive file has been read and so the	verilog file must be treated case-insensitive to match.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__clkbuf_1.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__nand2_1.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__nor2_1.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__oai21_1.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__xnor3_1.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__clkinv_1.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__xor2_1.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__dffrnq_1.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__endcap.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__filltie.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__clkbuf_4.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__clkbuf_16.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__antenna.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__fillcap_32.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__fill_1.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__fillcap_8.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__fill_2.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__fillcap_4.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__fillcap_64.
Creating placeholder cell definition for module gf180mcu_fd_sc_mcu7t5v0__fillcap_16.
Reading setup file /content/pdks/gf180mcuC/libs.tech/netgen/gf180mcuC_setup.tcl
Comparison output logged to file /content/OpenLane/designs/spm/runs/RUN_2022.09.22_05.46.35/logs/signoff/29-spm.lef.log
Logging to file "/content/OpenLane/designs/spm/runs/RUN_2022.09.22_05.46.35/logs/signoff/29-spm.lef.log" enabled
Circuit gf180mcu_fd_sc_mcu7t5v0__filltie contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__antenna contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__fill_2 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__nor2_1 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__clkinv_1 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__nand2_1 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__fill_1 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__fillcap_8 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__fillcap_4 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__oai21_1 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__xnor3_1 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__clkbuf_1 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__dffrnq_1 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__fillcap_16 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__clkbuf_16 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__fillcap_32 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__fillcap_64 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__clkbuf_4 contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__endcap contains no devices.
Circuit gf180mcu_fd_sc_mcu7t5v0__xor2_1 contains no devices.

Contents of circuit 1:  Circuit: 'spm'
Circuit spm contains 2512 device instances.
  Class: gf180mcu_fd_sc_mcu7t5v0__dffrnq_1 instances:  64
  Class: gf180mcu_fd_sc_mcu7t5v0__nand2_1 instances:  64
  Class: gf180mcu_fd_sc_mcu7t5v0__clkbuf_1 instances:  96
  Class: gf180mcu_fd_sc_mcu7t5v0__clkbuf_4 instances:   1
  Class: gf180mcu_fd_sc_mcu7t5v0__clkinv_1 instances:  65
  Class: gf180mcu_fd_sc_mcu7t5v0__filltie instances: 143
  Class: gf180mcu_fd_sc_mcu7t5v0__clkbuf_16 instances:   9
  Class: gf180mcu_fd_sc_mcu7t5v0__xnor3_1 instances:  31
  Class: gf180mcu_fd_sc_mcu7t5v0__oai21_1 instances:  31
[... 20 lines cut ...]
  Class: gf180mcu_fd_sc_mcu7t5v0__clkbuf_16 instances:   9
  Class: gf180mcu_fd_sc_mcu7t5v0__xnor3_1 instances:  31
  Class: gf180mcu_fd_sc_mcu7t5v0__oai21_1 instances:  31
  Class: gf180mcu_fd_sc_mcu7t5v0__fillcap_4 instances: 190
  Class: gf180mcu_fd_sc_mcu7t5v0__fillcap_8 instances:  77
  Class: gf180mcu_fd_sc_mcu7t5v0__fillcap_64 instances:   1
  Class: gf180mcu_fd_sc_mcu7t5v0__fillcap_16 instances:  37
  Class: gf180mcu_fd_sc_mcu7t5v0__fill_1 instances: 350
  Class: gf180mcu_fd_sc_mcu7t5v0__fill_2 instances: 723
  Class: gf180mcu_fd_sc_mcu7t5v0__nor2_1 instances:  31
[... cut ...] 07
  Class: gf180mcu_fd_sc_mcu7t5v0__fillcap_32 instances:  13
  Class: gf180mcu_fd_sc_mcu7t5v0__xor2_1 instances:   1
  Class: gf180mcu_fd_sc_mcu7t5v0__endcap instances:  78
Circuit contains 430 nets.

Circuit 1 contains 2512 devices, Circuit 2 contains 2512 devices.
Circuit 1 contains 430 nets,    Circuit 2 contains 430 nets.

Final result: 
Circuits match uniquely.
.
Logging to file "/content/OpenLane/designs/spm/runs/RUN_2022.09.22_05.46.35/logs/signoff/29-spm.lef.log" disabled
LVS Done.
```

proppy, 2022-09-22:
I think that means it could related to one of those three changes:
- https://github.com/RTimothyEdwards/netgen/commit/9297090dc1f4896f724c4b5beacdc60aab51df22
- https://github.com/RTimothyEdwards/netgen/commit/4fa4d046c893c3e31c44402e13d499021336f642
- https://github.com/RTimothyEdwards/netgen/commit/cc5f6d929f8f1f71cc4c3411943f0806ee19c95b

Curious if it would make sense to test them in isolation, given that they are referring to the same feature?

RTimothyEdwards, 2022-09-30:
@proppy :  I don't think it's directly related to the proxy pin issue but could be caused by the same code modification, but I think it is most likely that the code modification changed the text output of the comparison, which broke the `count_lvs.py` script, which makes various assumptions about the format of the output text.

I am going to revert the netgen modification and redo it in a different way;  there is a very useful bit of code there that handles cases where a pin gets disconnected from an internal net, and reports the internal net name with "not a pin" instead of just reporting "no matching pin" (i.e., it gives much more useful information to the user to help in debugging the issue).  So I want to implement that code, only without breaking every existing LVS script.

d-m-bailey, 2022-09-30:
@RTimothyEdwards Can I have a day to see if I can come up with a solution to the implicit verilog pin problem? For people doing LEF LVS, we could tell them to use 1.5.228.

RTimothyEdwards, 2022-09-30:
@d-m-bailey :  I'm reverting back to the state before I modified MatchPins() for now (version 1.5.234).  If you want to work on a different solution, you can start from 1.5.233 if you like, and if that works I can always do another revert back to there and then apply your new commit.

kareefardi, 2023-01-31:
I believe this is fixed now

closed 2023-01-31 by kareefardi