The-OpenROAD-Project/OpenLane #1393: [OpenSTA] Need command to show the statistic on inter-connect Timing arc/parasitic load by the the Spef

dineshannayya, 2022-09-26:
This request raised based on Slack discussion with @tspyrou 

### Prompt
OpenSTA tool has option to show the annotated status for SDF reading, But Similar details are not available for SPEF reading.

% report_annotated_check -list_annotated
                                                          Not   
Check type                        Total    Annotated   Annotated
----------------------------------------------------------------
cell setup arcs                     2746           0        2746
cell hold arcs                        2746           0        2746
cell recovery arcs                 1078           0        1078
cell removal arcs                  1078           0        1078
----------------------------------------------------------------
                                             7648           0        7648

### Proposal
Looking for STA command to show the SPEF annotation status shows number of timing arc loaded/not loaded.

tspyrou, 2022-09-26:
@dineshannayya I opened https://github.com/The-OpenROAD-Project/OpenSTA/issues/120 for this issue. I am closing this one since we can track it there.

tspyrou, 2022-09-26:
I am reopening this since github automatically links the issues.

jjcherry56, 2022-10-11:
openroad commit 91b326968 updates opensta to include the report_parasitic_annotation command

closed 2022-10-11 by jjcherry56