The-OpenROAD-Project/OpenLane #1394: Diode Insertion fails if routing obstruction enabled

vijayank88, 2022-09-26:
## Description
If configuration file has `GRT_OBS` and `diode_insertion` stage fails during `run_routing` stage

## Expected behavior
Complete the flow successfully

## Environment

```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: c40bcaca582bb192606265d49a582fb9ee3ba1be
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

c40bcac 2022-09-22T07:26:28+02:00 [BOT] Update PDK (#1389) - Openlane Bot -  (HEAD -> master, tag: 2022.09.23, origin/master, origin/HEAD)
c8f917e 2022-09-22T07:23:38+02:00 Correctly decorate wrapper in click_odb (#1381) - Matt Liberty -  ()
f81bd81 2022-09-22T07:21:39+02:00 Correct the installation link (#1384) - Arman Avetisyan -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

## Reproduction Material

1. Clone MPW CI: https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI
2. `cd OpenLane-MPW-CI`
3. Update `openlane_tag ` [2022.09.23](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.23)
4. To install latest OpenLane: `./scripts/setup-ci.sh`
5. To reproduce the issue run the design: `./scripts/run-design.sh yonga-100m-ethernet`
Check MPW CI for error: https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI/pull/91/checks?check_run_id=8532111552
Flow will fail during diode insertion stage with following error

## Logs

```
[STEP 13]
[INFO]: Running Diode Insertion (log: logs/routing/13-obs.log)...
invalid command name "set_db"
    while executing
"set_db $save_db"
    (procedure "apply_route_obs" line 15)
    invoked from within
"apply_route_obs"
    (procedure "add_route_obs" line 3)
    invoked from within
"add_route_obs"
    (procedure "run_routing" line 20)
    invoked from within
"run_routing"
    (procedure "run_routing_step" line 8)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 52)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 401)
make: *** [Makefile:175: quick_run] Error 1
make: Leaving directory '/tmp/workspace/_Public_production-small-3_PR-91@2/OpenLane'
```

vijayank88, 2022-09-30:
@donn 
Its blocking MPW CI regression test for OpenLane tag update.
Can you please priorities this issue?
Thanks....

donn, 2022-09-30:
I am afraid I am currently at capacity. If you would like to prioritize this task, please email @shalan who will assign a priority to this issue.

maliberty, 2022-10-03:
@shalan please update with an ETA

PR #1408 (2022-10-05): Fix `set_odb` invocation after adding obstructions
Resolves #1394
closed 2022-10-05 by donn