The-OpenROAD-Project/OpenLane #1403: issue_reproducible -

McLucifer2646, 2022-10-02:
## Description

I am trying to complete the OpenLane flow for my sequence detector design... The design runs pretty well when I go about the ```./flow.tcl -interactive``` method and runs perfectly well till routing but is failing when I try to run the complete flow using ```./flow.tcl -design iiitb_sd```. Here iiitb_sd is the name of my design file that contains netlist and other related files.

## Expected behavior

Complete the flow successfully till step number 43 and have assertive feedback saying "Flow Complete"

## Environment

## Python Installation

```
$   sudo apt install -y build-essential python3 python3-venv python3-pip
```

## Docker Installation
```
$   sudo apt-get update

$   sudo apt-get install docker docker-compose

$   sudo systemctl start docker

$   sudo docker run hello-world
```

YOUR SURVEY HERE

## Reproduction Material

1. git clone https://github.com/McLucifer2646/iiitb_sd.git
2. cd iiitb_sd
3. Install OpenLane as per the following commands
```
git clone https://github.com/The-OpenROAD-Project/OpenLane.git
cd OpenLane/
make 
make test
```
4. Once you are in the OpenLane folder, ```cd designs``` 
5. ```mkdir iiitb_sd``` and follow the next steps.
6. Follow the following steps for src folder generation and config.json file as mentioned in the LAYOUT section, using the  following link https://github.com/McLucifer2646/iiitb_sd#layout 
7. Instead of running ```./flow.tcl -interactive```, run ```./flow.tcl -design iiitb_sd```

## Logs

```
YOUR LOGS HERE
```
```
[STEP 27]
[INFO]: Running Multi-Corner Static Timing Analysis at the min process corner (log: designs/iiitb_sd/runs/RUN_2022.10.02_12.24.11/logs/signoff/27-parasitics_multi_corner_sta.min.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/sta_multi_corner.tcl
[ERROR]: Log: designs/iiitb_sd/runs/RUN_2022.10.02_12.24.11/logs/signoff/27-parasitics_multi_corner_sta.min.log
[ERROR]: Last 10 lines:
36# 0x00007F00FBB23F1E in /lib64/libtcl8.5.so
37# Tcl_EvalEx in /lib64/libtcl8.5.so
38# Tcl_Eval in /lib64/libtcl8.5.so
39# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
40# ord::tclAppInit(Tcl_Interp*) in openroad
41# Tcl_Main in /lib64/libtcl8.5.so
42# main in openroad
43# __libc_start_main in /lib64/libc.so.6
44# 0x0000000000C17807 in openroad
child process exited abnormally

[ERROR]: Creating issue reproducible...
[INFO]: Saving runtime environment...
OpenLane TCL Issue Packager

EFABLESS CORPORATION AND ALL AUTHORS OF THE OPENLANE PROJECT SHALL NOT BE HELD
LIABLE FOR ANY LEAKS THAT MAY OCCUR TO ANY PROPRIETARY DATA AS A RESULT OF USING
THIS SCRIPT. THIS SCRIPT IS PROVIDED ON AN "AS IS" BASIS, WITHOUT WARRANTIES OR
CONDITIONS OF ANY KIND.

BY USING THIS SCRIPT, YOU ACKNOWLEDGE THAT YOU FULLY UNDERSTAND THIS DISCLAIMER
AND ALL IT ENTAILS.

Parsing config file(s)…
Setting up /openlane/designs/iiitb_sd/runs/RUN_2022.10.02_12.24.11/issue_reproducible…
Found $::env(CLOCK_PORT)…
Found $::env(CLOCK_PORT)…
Found $::env(DESIGN_NAME)…
Found $::env(LIB_FASTEST)…
Found $::env(LIB_SLOWEST)…
Found $::env(LIB_TYPICAL)…
Found $::env(STA_REPORT_POWER)…
Found $::env(CURRENT_DEF)…
Found $::env(CURRENT_DEF)…
Found $::env(CURRENT_NETLIST)…
Found $::env(CURRENT_SDC)…
Found $::env(CURRENT_SPEF)…
Found $::env(CURRENT_SPEF)…
Found $::env(CURRENT_SPEF)…
Found $::env(STA_LEF)…
Found $::env(STA_PRE_CTS)…
Done.
[INFO]: Reproducible packaged: Please tarball and upload 'designs/iiitb_sd/runs/RUN_2022.10.02_12.24.11/issue_reproducible' if you're going to submit an issue.
[INFO]: Saving current set of views in 'designs/iiitb_sd/runs/RUN_2022.10.02_12.24.11/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'designs/iiitb_sd/runs/RUN_2022.10.02_12.24.11/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'designs/iiitb_sd/runs/RUN_2022.10.02_12.24.11/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
[INFO]: The failure may have been because of the following warnings:
[WARNING]: Current core area is too small for a power grid. The power grid will be minimized.

    while executing
"flow_fail"
    (procedure "run_tcl_script" line 89)
    invoked from within
"run_tcl_script -tool openroad -no_consume {*}$args"
    (procedure "run_openroad_script" line 2)
    invoked from within
"run_openroad_script $::env(SCRIPTS_DIR)/openroad/sta_multi_corner.tcl  -indexed_log $log"
    (procedure "run_sta" line 38)
    invoked from within
"run_sta -lef $::env($lef) -log $log_name -process_corner $process_corner -multi_corner"
    (procedure "run_parasitics_sta" line 44)
    invoked from within
"run_parasitics_sta"
    (procedure "run_parasitics_sta_step" line 9)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 52)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 401)
```

vijayank88, 2022-10-03:
@McLucifer2646 
I confirm the attached test case crash STA run.
OpenLane tag: 2022.08.25 is month back commit. Can update OpenLane commit to latest and confirm the same issue repeat again?

vijayank88, 2022-10-03:
with latest OpenROAD commit its after timing report its crashing.
```
======================== Slowest Corner ==================================

Clock clock
Latency      CRPR       Skew
_18_/CLK ^
   0.57
_20_/CLK ^
   0.51     -0.03       0.02

======================= Typical Corner ===================================

Clock clock
Latency      CRPR       Skew
_18_/CLK ^
   0.33
_20_/CLK ^
   0.30     -0.02       0.01

======================= Fastest Corner ===================================

Clock clock
Latency      CRPR       Skew
_18_/CLK ^
   0.23
_20_/CLK ^
   0.21     -0.01       0.01

clock_skew_end
power_report

===========================================================================
 report_power
============================================================================

======================= Slowest Corner =================================

Stack trace:
 0# 0x0000000000C879F6 in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 1# 0x00007F9136C7D400 in /lib64/libc.so.6
 2# sta::LibertyCell::findPgPort(char const*) const in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 3# sta::Power::findSwitchingPower(sta::LibertyCell*, sta::LibertyPort const*, sta::PwrActivity&, float, sta::Corner const*, sta::PowerResult&) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 4# sta::Power::power(sta::Instance const*, sta::LibertyCell*, sta::Corner const*, sta::PowerResult&) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 5# sta::Power::power(sta::Corner const*, sta::PowerResult&, sta::PowerResult&, sta::PowerResult&, sta::PowerResult&, sta::PowerResult&) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 6# design_power(sta::Corner const*) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 7# 0x0000000000E8B820 in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
 8# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
 9# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
10# TclObjInterpProcCore in /lib64/libtcl8.5.so
11# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
12# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
13# TclObjInterpProcCore in /lib64/libtcl8.5.so
14# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
15# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
16# 0x00007F9139A17647 in /lib64/libtcl8.5.so
17# TclEvalObjEx in /lib64/libtcl8.5.so
18# 0x00007F9139A5227F in /lib64/libtcl8.5.so
19# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
20# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
21# 0x00007F9139A17647 in /lib64/libtcl8.5.so
22# TclEvalObjEx in /lib64/libtcl8.5.so
23# 0x00007F91399D21D0 in /lib64/libtcl8.5.so
24# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
25# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
26# 0x00007F9139A17647 in /lib64/libtcl8.5.so
27# TclEvalObjEx in /lib64/libtcl8.5.so
28# 0x00007F91399D5F00 in /lib64/libtcl8.5.so
29# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
30# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
31# TclObjInterpProcCore in /lib64/libtcl8.5.so
32# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
33# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
34# TclObjInterpProcCore in /lib64/libtcl8.5.so
35# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
36# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
37# 0x00007F9139A17647 in /lib64/libtcl8.5.so
38# TclEvalObjEx in /lib64/libtcl8.5.so
39# 0x00007F9139A5227F in /lib64/libtcl8.5.so
40# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
41# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
42# 0x00007F9139A17647 in /lib64/libtcl8.5.so
43# TclEvalObjEx in /lib64/libtcl8.5.so
44# 0x00007F91399D21D0 in /lib64/libtcl8.5.so
45# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
46# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
47# 0x00007F9139A17647 in /lib64/libtcl8.5.so
48# TclEvalObjEx in /lib64/libtcl8.5.so
49# 0x00007F91399D5F00 in /lib64/libtcl8.5.so
50# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
51# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
52# TclObjInterpProcCore in /lib64/libtcl8.5.so
53# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
54# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
55# TclObjInterpProcCore in /lib64/libtcl8.5.so
56# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
57# 0x00007F91399CBF1E in /lib64/libtcl8.5.so
58# Tcl_EvalEx in /lib64/libtcl8.5.so
59# TclEvalObjEx in /lib64/libtcl8.5.so
60# 0x00007F91399D2A65 in /lib64/libtcl8.5.so
61# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
62# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
63# 0x00007F9139A17647 in /lib64/libtcl8.5.so
64# TclEvalObjEx in /lib64/libtcl8.5.so
65# 0x00007F91399D21D0 in /lib64/libtcl8.5.so
66# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
67# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
68# 0x00007F9139A17647 in /lib64/libtcl8.5.so
69# TclEvalObjEx in /lib64/libtcl8.5.so
70# 0x00007F9139A5227F in /lib64/libtcl8.5.so
71# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
72# 0x00007F9139A0F36C in /lib64/libtcl8.5.so
73# TclObjInterpProcCore in /lib64/libtcl8.5.so
74# 0x00007F91399CAEB2 in /lib64/libtcl8.5.so
75# 0x00007F91399CBF1E in /lib64/libtcl8.5.so
76# Tcl_EvalEx in /lib64/libtcl8.5.so
77# Tcl_Eval in /lib64/libtcl8.5.so
78# ord::tclAppInit(Tcl_Interp*) in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
79# Tcl_Main in /lib64/libtcl8.5.so
80# main in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
81# __libc_start_main in /lib64/libc.so.6
82# 0x0000000000C834D7 in /home/vijayan/OPENROAD_FLOW/OpenROAD_FLOW/OpenROAD/build/src/openroad
```
@jjcherry56 can you please check this issue?

jjcherry56, 2022-10-03:
The underlying problem is that the fast and slow liberty files are missing the sky130_fd_sc_hd__inv_2 cell.
The libraries have to be exactly congruent for multi-corner analysis. I will work on adding checks for broken
libraries, but in the mean time you should get the libraries fixed. There are no doubt other cells with the same 
problem.

McLucifer2646, 2022-10-04:
WELL, the problem got fixed, I changed the fast and slow liberty files as advised by @jjcherry56 and I got an affirmative green check saying ***"Flow Complete"***. 

Thank you @maliberty @jjcherry56 for helping me out with my bug and special Thanks to @vijayank88 who taught me how to report an issue on GitHub leading to the solution.

closed 2022-10-04 by McLucifer2646