The-OpenROAD-Project/OpenLane #1404: Flow crashes during writing timing model

vijayank88, 2022-10-03:
## Description
With [2022.09.29](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.29) OpenLane flow fails during writing timing model for the MPW design which passed with [2022.09.23](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.23) successfully.

## Expected behavior

Crash should not happen.

## Environment

```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 2022.09.29
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---

```

## Reproduction Material

1.     Clone MPW CI: https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI
2.     cd OpenLane-MPW-CI
3.     Update `openlane-tag` [2022.09.29](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.29)
4.     To install latest OpenLane: ./scripts/setup-ci.sh
5.     To reproduce the issue run the design: `./scripts/run-design.sh logic_bist LBIST_mbist_top1`
6.     Check MPW CI for error: https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI/pull/91/checks?check_run_id=8648880990
7.     Flow will crash during STA run with `writing timing model` step.

## Logs

```
area_report_end
[WARNING] Did not save OpenROAD database!
Writing SDF to /openlane/designs/LBIST_mbist_top1/runs/RUN_2022.10.03_06.29.53/results/synthesis/mbist_top1.sdf...
Writing timing model to /openlane/designs/LBIST_mbist_top1/runs/RUN_2022.10.03_06.29.53/results/synthesis/mbist_top1.lib...
Stack trace:
 0# 0x0000000000C78EE6 in openroad
 1# 0x00007FF1D804A400 in /lib64/libc.so.6
 2# sta::MakeTimingModel::makeSetupHoldTimingArcs(sta::Pin const*, std::map<sta::ClockEdge*, sta::RiseFallMinMax, std::less<sta::ClockEdge*>, std::allocator<std::pair<sta::ClockEdge* const, sta::RiseFallMinMax> > > const&) in openroad
 3# sta::MakeTimingModel::findTimingFromInputs() in openroad
 4# sta::MakeTimingModel::makeTimingModel(char const*, char const*) in openroad
 5# sta::Sta::writeTimingModel(char const*, char const*, sta::Corner const*) in openroad
 6# 0x0000000000E4CD8A in openroad
 7# 0x00007FF1DAD97EB2 in /lib64/libtcl8.5.so
 8# 0x00007FF1DADDC36C in /lib64/libtcl8.5.so
 9# TclObjInterpProcCore in /lib64/libtcl8.5.so
10# 0x00007FF1DAD97EB2 in /lib64/libtcl8.5.so
11# 0x00007FF1DADDC36C in /lib64/libtcl8.5.so
12# TclObjInterpProcCore in /lib64/libtcl8.5.so
13# 0x00007FF1DAD97EB2 in /lib64/libtcl8.5.so
14# 0x00007FF1DADDC36C in /lib64/libtcl8.5.so
15# 0x00007FF1DADE4647 in /lib64/libtcl8.5.so
16# TclEvalObjEx in /lib64/libtcl8.5.so
17# 0x00007FF1DAE1F27F in /lib64/libtcl8.5.so
18# 0x00007FF1DAD97EB2 in /lib64/libtcl8.5.so
19# 0x00007FF1DADDC36C in /lib64/libtcl8.5.so
20# 0x00007FF1DADE4647 in /lib64/libtcl8.5.so
21# TclEvalObjEx in /lib64/libtcl8.5.so
22# 0x00007FF1DAD9F1D0 in /lib64/libtcl8.5.so
23# 0x00007FF1DAD97EB2 in /lib64/libtcl8.5.so
24# 0x00007FF1DADDC36C in /lib64/libtcl8.5.so
25# 0x00007FF1DADE4647 in /lib64/libtcl8.5.so
26# TclEvalObjEx in /lib64/libtcl8.5.so
27# 0x00007FF1DADA2F00 in /lib64/libtcl8.5.so
28# 0x00007FF1DAD97EB2 in /lib64/libtcl8.5.so
29# 0x00007FF1DADDC36C in /lib64/libtcl8.5.so
30# TclObjInterpProcCore in /lib64/libtcl8.5.so
31# 0x00007FF1DAD97EB2 in /lib64/libtcl8.5.so
32# 0x00007FF1DADDC36C in /lib64/libtcl8.5.so
33# TclObjInterpProcCore in /lib64/libtcl8.5.so
34# 0x00007FF1DAD97EB2 in /lib64/libtcl8.5.so
35# 0x00007FF1DAD98F1E in /lib64/libtcl8.5.so
36# Tcl_EvalEx in /lib64/libtcl8.5.so
37# Tcl_Eval in /lib64/libtcl8.5.so
38# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
39# ord::tclAppInit(Tcl_Interp*) in openroad
40# Tcl_Main in /lib64/libtcl8.5.so
41# main in openroad
42# __libc_start_main in /lib64/libc.so.6
43# 0x0000000000C75967 in openroad
```

jjcherry56, 2022-10-03:
you are going to have to package this as an openroad command file before I will look at it.

vijayank88, 2022-10-06:
@jjcherry56 
Attached test case: 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9721624/issue_reproducible.zip)

Respective log:
```
OpenROAD 4174c3ad802d2ac1d04d387d2c4b883903f6647e
```
```
Setting global connections for newly added cells...
[WARNING] Did not save OpenROAD database!
Writing SDF files for all corners...
Writing SDF for the ff corner to /openlane/designs/LBIST_mbist_top1/runs/RUN_2022.10.05_16.37.32/results/routing/process_corner_min/mbist_top1.ff.sdf...
Writing SDF for the ss corner to /openlane/designs/LBIST_mbist_top1/runs/RUN_2022.10.05_16.37.32/results/routing/process_corner_min/mbist_top1.ss.sdf...
Writing SDF for the tt corner to /openlane/designs/LBIST_mbist_top1/runs/RUN_2022.10.05_16.37.32/results/routing/process_corner_min/mbist_top1.tt.sdf...
Writing timing models for all corners...
Writing timing models for the ff corner to /openlane/designs/LBIST_mbist_top1/runs/RUN_2022.10.05_16.37.32/results/routing/process_corner_min/mbist_top1.ff.lib...
Stack trace:
 0# 0x0000000000C78EE6 in openroad
 1# 0x00007FE8A6023400 in /lib64/libc.so.6
 2# sta::MakeTimingModel::makeSetupHoldTimingArcs(sta::Pin const*, std::map<sta::ClockEdge*, sta::RiseFallMinMax, std::less<sta::ClockEdge*>, std::allocator<std::pair<sta::ClockEdge* const, sta::RiseFallMinMax> > > const&) in openroad
 3# sta::MakeTimingModel::findTimingFromInputs() in openroad
 4# sta::MakeTimingModel::makeTimingModel(char const*, char const*) in openroad
 5# sta::Sta::writeTimingModel(char const*, char const*, sta::Corner const*) in openroad
 6# 0x0000000000E4CD8A in openroad
 7# 0x00007FE8A8D70EB2 in /lib64/libtcl8.5.so
 8# 0x00007FE8A8DB536C in /lib64/libtcl8.5.so
 9# TclObjInterpProcCore in /lib64/libtcl8.5.so
10# 0x00007FE8A8D70EB2 in /lib64/libtcl8.5.so
11# 0x00007FE8A8DB536C in /lib64/libtcl8.5.so
12# TclObjInterpProcCore in /lib64/libtcl8.5.so
13# 0x00007FE8A8D70EB2 in /lib64/libtcl8.5.so
14# 0x00007FE8A8DB536C in /lib64/libtcl8.5.so
15# 0x00007FE8A8DBD647 in /lib64/libtcl8.5.so
16# TclEvalObjEx in /lib64/libtcl8.5.so
17# 0x00007FE8A8DF827F in /lib64/libtcl8.5.so
18# 0x00007FE8A8D70EB2 in /lib64/libtcl8.5.so
19# 0x00007FE8A8DB536C in /lib64/libtcl8.5.so
20# 0x00007FE8A8DBD647 in /lib64/libtcl8.5.so
21# TclEvalObjEx in /lib64/libtcl8.5.so
22# 0x00007FE8A8D781D0 in /lib64/libtcl8.5.so
23# 0x00007FE8A8D70EB2 in /lib64/libtcl8.5.so
24# 0x00007FE8A8DB536C in /lib64/libtcl8.5.so
25# 0x00007FE8A8DBD647 in /lib64/libtcl8.5.so
26# TclEvalObjEx in /lib64/libtcl8.5.so
27# 0x00007FE8A8D7BF00 in /lib64/libtcl8.5.so
28# 0x00007FE8A8D70EB2 in /lib64/libtcl8.5.so
29# 0x00007FE8A8DB536C in /lib64/libtcl8.5.so
30# TclObjInterpProcCore in /lib64/libtcl8.5.so
31# 0x00007FE8A8D70EB2 in /lib64/libtcl8.5.so
32# 0x00007FE8A8DB536C in /lib64/libtcl8.5.so
33# TclObjInterpProcCore in /lib64/libtcl8.5.so
34# 0x00007FE8A8D70EB2 in /lib64/libtcl8.5.so
35# 0x00007FE8A8D71F1E in /lib64/libtcl8.5.so
36# Tcl_EvalEx in /lib64/libtcl8.5.so
37# Tcl_Eval in /lib64/libtcl8.5.so
38# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
39# ord::tclAppInit(Tcl_Interp*) in openroad
40# Tcl_Main in /lib64/libtcl8.5.so
41# main in openroad
42# __libc_start_main in /lib64/libc.so.6
43# 0x0000000000C75967 in openroad
```

vijayank88, 2022-10-07:
Test case generated with latest OpenROAD commit: `OpenROAD v2.0-5118-gbf529ad4a`
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9731019/issue_reproducible.zip)

Logs:
```
[STEP 20]
[INFO]: Running Multi-Corner Static Timing Analysis at the min process corner (log: logs/signoff/20-rcx_mcsta.min.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/sta_multi_corner.tcl
[ERROR]: Log: designs/LBIST_mbist_top1/runs/RUN_2022.10.07_05.37.57/logs/signoff/20-rcx_mcsta.min.log
[ERROR]: Last 10 lines:
35# 0x00007F06A37AAF1E in /lib64/libtcl8.5.so
36# Tcl_EvalEx in /lib64/libtcl8.5.so
37# Tcl_Eval in /lib64/libtcl8.5.so
38# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
39# ord::tclAppInit(Tcl_Interp*) in openroad
40# Tcl_Main in /lib64/libtcl8.5.so
41# main in openroad
42# __libc_start_main in /lib64/libc.so.6
43# 0x0000000000C91827 in openroad
child process exited abnormally
```

jjcherry56, 2022-10-07:
I fixed this in OpenSTA but who knows how long it will take to trickle up to openroad, which is 3 months behind on opensta and is currently blocked by CI metric failures

maliberty, 2022-10-08:
@jjcherry56 is there a PR pending for the update?  Is there something you need in order to resolve the metrics issue?

vijayank88, 2022-10-18:
Any update on this, around 15 stable MPW CI design got failed due to this error.

maliberty, 2022-10-18:
sta is no longer three months behind and is now on a commit from Oct 5.  It looks like this fix is after that date.  I've started a PR to update sta in OR.  (FYI @jjcherry56 )

maliberty, 2022-10-18:
https://github.com/The-OpenROAD-Project/OpenROAD/pull/2380

maliberty, 2022-10-18:
merged

vijayank88, 2022-10-21:
I would like to keep this open till respective commit merged into OpenLane master.

maliberty, 2022-10-21:
Ok I'll leave you in charge of closing it then.

commit 1f923eb2 (2022-11-01): [BOT] Update openroad_app (#1435)

Supposedly resolves #1404
closed 2022-11-01 by donn