The-OpenROAD-Project/OpenLane #1405: Flow fails during healing antenna violators

vijayank88, 2022-10-03:
## Description
With OpenLane tag [2022.09.20](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.09.20) and above fails during ` Healing Antenna Violators...` step.

## Expected behavior

Flow should pass successfully.

## Environment

```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 2022.10.03
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---

```

## Reproduction Material

- Clone MPW CI: https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI
- `cd OpenLane-MPW-CI`
- Update `openlane-tag ` to [2022.10.03](https://github.com/The-OpenROAD-Project/OpenLane/releases/tag/2022.10.03)
- To install latest OpenLane: `./scripts/setup-ci.sh`
- To reproduce the issue run the design: `./scripts/run-design.sh qf100 qf_mkQF100Fabric`
- Check MPW CI for error: https://github.com/The-OpenROAD-Project/OpenLane-MPW-CI/pull/91/checks?check_run_id=8648881163
- Flow will fail during `[STEP 30] [INFO]: Healing Antenna Violators...`

## Logs

```
[STEP 30]
[INFO]: Healing Antenna Violators...
[ERROR]: during executing: "openroad -python /openlane/scripts/odbpy/diodes.py replace_fake --violations-file /openlane/designs/qf_mkQF100Fabric/runs/RUN_2022.09.30_22.28.47/reports/signoff/29-antenna_violators.rpt --fake-diode sky130_ef_sc_hd__fakediode_2 --true-diode sky130_fd_sc_hd__diode_2 --input-lef /openlane/designs/qf_mkQF100Fabric/runs/RUN_2022.09.30_22.28.47/tmp/merged.nom.lef --output-def /openlane/designs/qf_mkQF100Fabric/runs/RUN_2022.09.30_22.28.47/results/routing/mkQF100Fabric.def --output /openlane/designs/qf_mkQF100Fabric/runs/RUN_2022.09.30_22.28.47/results/routing/mkQF100Fabric.odb /openlane/designs/qf_mkQF100Fabric/runs/RUN_2022.09.30_22.28.47/results/routing/mkQF100Fabric.odb |& tee /dev/null /dev/null"
[ERROR]: Exit code: 1
[ERROR]: Last 10 lines:
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1395, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 754, in invoke
    return __callback(*args, **kwargs)
  File "/openlane/scripts/odbpy/reader.py", line 76, in wrapper
    function(**kwargs)
  File "/openlane/scripts/odbpy/diodes.py", line 393, in replace_fake
    diode = [sc for sc in reader.block.getMasters() if sc.getName() == true_diode]
TypeError: getMasters() missing 1 required positional argument: 'masters'
child process exited abnormally

[INFO]: Saving current set of views in 'results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.

    while executing
"flow_fail"
    (procedure "try_catch" line 17)
    invoked from within
"try_catch $::env(OPENROAD_BIN) -python  {*}$args  --input-lef $::env(MERGED_LEF)  --output-def $arg_values(-output_def)  --output $arg_values(-output)..."
    (procedure "manipulate_layout" line 19)
    invoked from within
"manipulate_layout $::env(SCRIPTS_DIR)/odbpy/diodes.py replace_fake -output_def $::env(CURRENT_DEF) --violations-file $::env(ANTENNA_VIOLATOR_LIST) --f..."
    (procedure "heal_antenna_violators" line 15)
    invoked from within
"heal_antenna_violators"
    (procedure "run_diode_insertion_2_5_step" line 9)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 52)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 401)
```

vijayank88, 2022-10-18:
@donn 
What will be the expected time to get resolve this?
If its take too long, I'll update MPW CI designs by changing the `DIODE_INSERTION_STRATEGY' to make CI green as workaround.

donn, 2023-04-10:
Should be fixed now.

closed 2023-04-10 by donn