The-OpenROAD-Project/OpenLane #1412: [PDN] How to connect two Power Region of same power nets

dineshannayya, 2022-10-08:
## Description
I am looking for way to connect two power region which are sharing same power nets.
I have using example in PDN directory test case region_secondary_nets.tcl  and made changes to include top-level region with all the power nets.

Top-level has Power Ring : VREG1,VREG2,VDD,VSS
Also Sub Region also has same power Strips

[image]

## Expected behavior
My expectation was as sub-region power net names are same as top-level nets, PDN should have connected both the region with met-4 strip.  But I don't see any Power net connectivity between these region.  What are additional command need to force the tool to connect these regions

## Environment
Openroad Git log

<pre><font color="#C4A000">commit d919ca2d4f57ad35c095156f0b426fb89eef2430 (</font><font color="#06989A"><b>HEAD -&gt; </b></font><font color="#4E9A06"><b>master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/master</b></font><font color="#C4A000">, </font><font color="#CC0000"><b>origin/HEAD</b></font><font color="#C4A000">)</font>
Merge: b3023a4e7 8d0ad148d
Author: Eder Monteiro &lt;43450810+eder-matheus@users.noreply.github.com&gt;
Date:   Sat Oct 8 01:28:48 2022 -0300

    Merge pull request #2170 from eder-matheus/grt_derating

<font color="#C4A000">commit b3023a4e7f5f5cfb244c69b2e483974cdcf6b998</font>
Merge: 252626617 ec1ec7f4e
Author: Matt Liberty &lt;mliberty@eng.ucsd.edu&gt;
Date:   Fri Oct 7 21:03:31 2022 -0700

    Merge pull request #2340 from macd/fixCoverity
    
    Fix Coverity issues

</pre>

## Reproduction Material

[exp10.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9738512/exp10.tar.gz)

Run command : run_cmd

OR openroad -gui region_secondary_nets.tcl

## Logs

maliberty, 2022-10-10:
@arlpetergadfort any thoughts?

arlpetergadfort, 2022-10-10:
@maliberty I would suggest making the subregion rings in M3 and M4 and place straps in the core grid on M5 that connect to M4, that should connect them.

dineshannayya, 2022-10-11:
@arlpetergadfort  I am tried to change the Sub Region Ring to M3 and M4,  Still i see Sub region power domain does not connect top-level ring

[image]

[region_secondary_nets.tcl.txt](https://github.com/The-OpenROAD-Project/OpenLane/files/9750937/region_secondary_nets.tcl.txt)

Let me know any additional command need to connect two region.

arlpetergadfort, 2022-10-11:
@dineshannayya you will also need to add straps in M5 as part of the core grid to make the connections. Also you do not need to move the straps in the subregion.

dineshannayya, 2022-10-25:
@arlpetergadfort Thanks for your suggestion, I am able to connect two region with M5 straps

closed 2022-10-25 by dineshannayya