The-OpenROAD-Project/OpenLane #1419: SDF file, FF timing arcs: all bc, nc, wc entries have the same value

cloudxcc, 2022-10-10:
## Description
SDF timing arcs of each FF seem to be unconditional (having the same value fo bc, nc, wc conditions). Random example: 

(CELL
  (CELLTYPE "sky130_fd_sc_hd__dfrtp_1")
  (INSTANCE _29240_)
  (DELAY
   (ABSOLUTE
    (IOPATH CLK Q (0.390:0.390:0.390) (0.406:0.406:0.406))
    (IOPATH RESET_B Q () (0.000:0.000:0.000))
   )
  )
  (TIMINGCHECK
    (REMOVAL (posedge RESET_B) (posedge CLK) (0.337:0.337:0.337))       <---------- values should vary
    (RECOVERY (posedge RESET_B) (posedge CLK) (-0.211:-0.211:-0.211))
    (HOLD (posedge D) (posedge CLK) (-0.058:-0.058:-0.058))
    (HOLD (negedge D) (posedge CLK) (-0.074:-0.074:-0.074))
    (SETUP (posedge D) (posedge CLK) (0.086:0.086:0.086))
    (SETUP (negedge D) (posedge CLK) (0.144:0.144:0.144))
  )
 )

Each FF has different values, so that looks okay.

## Expected behavior
Bc, nc and wc values should vary.

## Environment
Latest OpenLane. I checked designs using older versions of OL, but the problem seems to be there ever since.

## Reproduction Material
Any digital design (running Caravel flow) of your choice.

## Logs
Serach for "_df" in any given SDF file and you'll find the problem.

jjcherry56, 2022-10-11:
SDF written by OpenSTA is targeted for timing analysis and the min/max values are for a single process corner specified by the -corner argument to write_sdf. They are NOT different process corners.

closed 2022-10-11 by jjcherry56

cloudxcc, 2022-10-11:
For all other timing arcs (non FF and interconnects) the bc, nc, wc values differ. So which one is the correct one for a - let's say - buffer or an interconnect.

There is only one final SDF file in the sdf/ directory. Which process corner does it cover ?

cloudxcc, 2022-10-11:
@jjcherry56 

For all other timing arcs (non FF and interconnects) the bc, nc, wc values differ. So which one is the correct one for a - let's say - buffer or an interconnect and what is the meaning of the 3 different values ?

Why are there "conditional timing arcs or slew merging" for all other timing arcs but not - let's say - for the CLK -> Q timing arc of a FF ?

The only remaining sdf file is the nom process corner, as the other ones get overwritten, right ?

maliberty, 2022-10-11:
There is only one path through clk->q typically.  I don't see OL passing any corner flag so I don't think it write sdf for more than one corner.

cloudxcc, 2022-10-11:
Sure, but there is also only one path through buffers, inverters, ...  etc. and there all bc, nc, wc values differ.

maliberty, 2022-10-11:
Sorry I don't mean one path through, I mean one path to it.

cloudxcc, 2022-10-12:
@jjcherry56, @maliberty 

1)	Only sta.tcl writes one (I guess nom) SDF file. sta_multic_orner.tcl does not write corner specific SDF files. But it would be nice.

2)	I was wrong with stating that buffers, … have different bc, nc, wc values. They have identical values, too.

3)	Based o Matt's help I guess I slowly get what "conditional timing arcs or slew merging" means. But based on a previous issue I raised, I thought conditional timings are handled with different timing paths as well:

https://github.com/The-OpenROAD-Project/OpenLane/issues/1192