The-OpenROAD-Project/OpenLane #1421: klayout is not able to open GUI into docker container

CristopherA96, 2022-10-10:
## Description

Hi, I found a typo error in the full documentation at Step 8. Optional: Viewing Test Design Outputs. Also, by running klayout as it is specified in the documentation is not able to open the gds file into container. Outside container it is able to open gds.

What happens is that a path is needed where the sky130A.lyt files and the sky130A.lyp file are located. That provokes an error when I try to run klayout with the GDSII.

The path should be:
klayout -e -nn $PDK_ROOT/sky130A/libs.tech/klayout/tech/sky130A.lyt \
   -l $PDK_ROOT/sky130A/libs.tech/klayout/tech/sky130A.lyp \
   ./designs/spm/runs/openlane_test/results/final/gds/spm.gds

## Expected behavior

Writing the correct path for sky130A.lty and sky130A.lyp hope to open the gds file with klayout but a warning is shown:

Warning: QStandardPaths: XDG_RUNTIME_DIR not set, defaulting to '/tmp/runtime-'
process 9: The last reference on a connection was dropped without closing the connection. This is a bug in an application. See dbus_connection_unref() documentation for details.

I tried by using magic instead of klayout and it works.

So, it will be helpful if klayout works well into the container, but I do not have problems using klayout outside container.

## Environment

I am using a machine with Ubuntu 20.04.

atorkmabrains, 2022-10-10:
@CristopherA96 I don't believe any GUI application will open in Docker unless you start docker with GUI support. Please take a look at:
https://www.google.com/url?sa=t&rct=j&q=&esrc=s&source=web&cd=&ved=2ahUKEwiawZeR7db6AhWER_EDHcO7CxYQFnoECBEQAQ&url=https%3A%2F%2Fmedium.com%2Fgeekculture%2Frun-a-gui-software-inside-a-docker-container-dce61771f9&usg=AOvVaw2Had8GimnSKiAydgEq2SZE

and 
https://www.howtogeek.com/devops/how-to-run-gui-applications-in-a-docker-container/

CristopherA96, 2022-10-10:
Thanks @atorkmabrains, I was thinking in the option shown in second link. 

I think that documentation will be update it, because of this missing part.

ChenLiangChi, 2022-10-24:
Hi,
I am having the same issue and this is how I solved it.

**First**, simply adding "DOCKER_OPTIONS += --privileged" in the Makefile within the OpenLane repository. 
**Second**, "make mount" again.
**Last**, run the klayout command again with the correct files locations:
klayout -e -nn $PDK_ROOT/sky130A/libs.tech/klayout/tech/sky130A.lyt \
-l $PDK_ROOT/sky130A/libs.tech/klayout/tech/sky130A.lyp \
./designs/spm/runs/openlane_test/results/final/gds/spm.gds

Then it should works.

Hope this helps you.

armleo, 2022-10-24:
1. The environment information is not filled properly.
2. Need the PDK_ROOT's content (to see if its available to user)

AFAIK you should not use DOCKER_OPTIONS += --privileged. It may create different problems.

donn, 2022-11-09:
"I am using a machine with Ubuntu 20.04," while helpful, is unfortunately not a substitute for the issue survey.

closed 2022-11-09 by donn