The-OpenROAD-Project/OpenLane #1423: Control the PDN when it's minimised

mattvenn, 2022-10-12:
### Prompt
I'm working on squeezing a design very narrow. Got down to 30um. This is for an experiment for TinyTapeout.com

[image]

I should be able to go even smaller, which would be good, but then the PDN vertical straps start to touch.
There isn't really a need for so many.

### Proposal

When the PDN gets minimised, allow a way to control the number of straps in use.

Example repo: https://github.com/TinyTapeout/scanchain

vijayank88, 2022-10-12:
@mattvenn 
Repo not found

mattvenn, 2022-10-12:
? Link works for me!

mattvenn, 2022-10-12:
apologies - it was set to private for some reason