The-OpenROAD-Project/OpenLane #1426: be able to set io length to less than 4

mattvenn, 2022-10-12:
## Description
FP_IO_VLENGTH & FP_IO_HLENGTH don't work for setting less than 4

## Expected behavior
be able to set less than 4

## Environment

```

Kernel: Linux v5.15.0-48-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)

```

## Reproduction Material

https://github.com/TinyTapeout/scanchain

## Logs
[image]

maliberty, 2022-10-12:
I don't see any setting of those variables in https://github.com/TinyTapeout/scanchain/blob/main/src/config.tcl

What are you trying to set it to?  The value is snapped to the manufacturing grid.

Are you setting FP_IO_[VH]EXTEND ?

mattvenn, 2022-10-12:
https://github.com/TinyTapeout/scanchain/blob/main/src/config.tcl#L32

setting HLENGTH to 2, and it remains 4.

maliberty, 2022-10-12:
You have to set both H&V by the current script:
https://github.com/The-OpenROAD-Project/OpenLane/blob/32da932761213af689f10088d671e1e3dc38f273/scripts/openroad/ioplacer.tcl#L26

maliberty, 2022-10-12:
Actually it looks like a copy paste error as it is checking the same var twice...

mattvenn, 2022-10-12:
Yes, and I've tried setting both and it still remains 4.

mattvenn, 2022-10-12:
I've tried setting both and it remains 4

maliberty, 2022-10-12:
@eder-matheus any thoughts?

eder-matheus, 2022-10-12:
@maliberty I think OpenLane scripts are incorrect. These three blocks:

```
if {$::env(FP_IO_HLENGTH) != "" && $::env(FP_IO_HLENGTH) != ""} {
	set_pin_length -hor_length $::env(FP_IO_HLENGTH) \
		-ver_length $::env(FP_IO_VLENGTH)
}

if {$::env(FP_IO_HLENGTH) != "" && $::env(FP_IO_HLENGTH) != ""} {
	set_pin_length_extension -hor_extension $::env(FP_IO_HEXTEND) \
		-ver_extension $::env(FP_IO_VEXTEND)
}

if {$::env(FP_IO_HLENGTH) != "" && $::env(FP_IO_HLENGTH) != ""} {
	set_pin_thick_multiplier -hor_multiplier $::env(FP_IO_HTHICKNESS_MULT) \
		-ver_multiplier $::env(FP_IO_VTHICKNESS_MULT)
}
```
are checking for the `FP_IO_HLENGTH`, where they should be checking for `FP_IO_HEXTEND` and `FP_IO_HTHICKNESS_MULT`. Not sure if these variables have random content on it.

@mattvenn  Does OpenLane generate a reproducible in a .tar file? It's easier to debug with this.

maliberty, 2022-10-12:
Yes with https://github.com/The-OpenROAD-Project/OpenLane/blob/master/docs/source/for_developers/using_or_issue.md or by forcing a flow failure.  @mattvenn please package a test case

mattvenn, 2022-10-13:
I tried making a reproducible, but in the process of making it, the pads are now the correct length. I think I hadn't saved the config after setting both VLENGTH & HLENGTH.

I am only testing horizontal IOs.

If only FP_IO_VLENGTH = 2, then pads are 4
if only FP_IO_HLENGTH = 2, then pads are 4.
if both are set, then pads are 2.

maliberty, 2022-10-13:
I think that is a script problem in OL not an OR issue.  @donn

kareefardi, 2023-02-05:
Fixed by attached PR

closed 2023-02-05 by kareefardi