The-OpenROAD-Project/OpenLane #1432: PDK_ROOT is not specified.

ashkanr65, 2022-10-17:
## Description

When I run 
`sudo make test`
The OpenLane works and do the test and pass through all the steps.
But when I run `sudo ./flow.tcl -design spm` (or any other designs). it returns this error:
```
[ERROR]: PDK_ROOT is not specified. Please make sure you have it set.

    while executing
"prep {*}$args"
    (procedure "run_non_interactive_mode" line 11)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
	if { [info exists arg_values(-file)] } {
		run_file [file normalize $a..."
    (file "./flow.tcl" line 434)
```
In the flow.tcl, the only required items that is needed is -design name.

## Expected behavior

## Environment

```
Kernel: Linux v5.15.0-48-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: UNKNOWN vUNKNOWN (UNSUPPORTED)
OpenLane Git Version: 0dc6fb79c91082b94f8ded78d70f8bacbab96bf2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

0dc6fb7 2022-05-17T08:24:25-07:00 Enable PL_ROUTABILITY_DRIVEN and PL_TIME_DRIVEN (#1092) - Anton Blanchard -  (HEAD -> master, tag: 2022.05.18_02.12.32, origin/master, origin/HEAD)
fba1ad0 2022-05-17T14:45:07+02:00 Remove unnecessary calls to `read_verilog` (#1084) - Anton Blanchard -  ()
9d463c0 2022-05-17T14:43:10+02:00 Add `FP_IO_UNMATCHED_ERROR` flag (#1086) - Kareem Farid -  ()

```

## Reproduction Material

## Logs

```
YOUR LOGS HERE
```

vijayank88, 2022-10-17:
@ashkanr65 
you're using very old commit from `2022-05-17` version. Please update to latest one and try again.
https://github.com/The-OpenROAD-Project/OpenLane

ashkanr65, 2022-10-17:
I ran:

`git pull --depth 1 https://github.com/The-OpenROAD-Project/OpenLane.git master
make
`
But I still get the same error

vijayank88, 2022-10-17:
Can you confirm `docker --version`?

Try following commands in `OpenLane` directory
```
git checkout master
git pull
make
```

ashkanr65, 2022-10-17:
The 
`
docker --version
`
returns
`
Docker version 20.10.19, build d85ef84
`
By
`git pull `
I got this error:
`fatal: refusing to merge unrelated histories
`
But
`Sudo make test
`
actually works.
But this command still not working:
`sudo ./flow.tcl -design spm
`
and returns same error

vijayank88, 2022-10-17:
Create new directory and start fresh installation, by following below stepsL
```
git clone --depth 1 https://github.com/The-OpenROAD-Project/OpenLane.git
cd OpenLane/
make
make test 
```

ashkanr65, 2022-10-17:
@vijayank88 

Now I get this error:
```

[ERROR]: Failed to compare PDKs.
'NoneType' object has no attribute 'startswith'
Traceback (most recent call last):
  File "/home/ashkan/OpenLane/dependencies/verify_versions.py", line 68, in verify_versions
    if pdk.startswith("sky130"):
AttributeError: 'NoneType' object has no attribute 'startswith'

[ERROR]: Please update your environment. OpenLane will now quit.
```

vijayank88, 2022-10-17:
`python3 --version`
Try following commands and paste complete log.
```
cd OpenLane
make mount
./flow.tcl -design spm -verbose 99
```

ashkanr65, 2022-10-17:
```
python3 --version
Python 3.8.10

```
[openlane.log](https://github.com/The-OpenROAD-Project/OpenLane/files/9799857/openlane.log)

vijayank88, 2022-10-17:
```
[ERROR]: Failed to compare PDKs.
'NoneType' object has no attribute 'startswith'
Traceback (most recent call last):
  File "/home/ashkan/OpenLane/dependencies/verify_versions.py", line 68, in verify_versions
    if pdk.startswith("sky130"):
AttributeError: 'NoneType' object has no attribute 'startswith'

[ERROR]: Please update your environment. OpenLane will now quit.
```
Attached `openlane.log` shows  `Flow Complete`, not found above error.

closed 2022-10-17 by vijayank88