The-OpenROAD-Project/OpenLane #1434: can't turn off output buffers

mattvenn, 2022-10-17:
## Description

outputs have buffers on them. This should be controllable with PL_RESIZER_BUFFER_OUTPUT_PORTS
with it set to 0, outputs should not be buffered.

## Expected behavior

turning of the buffered output ports should turn off the buffers. 

## Environment

```
Kernel: Linux v5.13.0-28-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.7 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)

```

## Reproduction Material

[buffertest.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9804281/buffertest.tar.gz)

mattvenn, 2022-10-17:
maybe it's to do with PL_RESIZER_BUFFER_INPUT_PORTS.
yes that's it. clkbufs are used for input ports,
but then there are these 8 bufs inserted between the ins and the outs - maybe that's expected / desired?

antonblanchard, 2022-10-17:
I was looking at something similar recently. You have inputs and outputs directly connected:

```
module buffer_test (
    input [7:0] ins,
    output [7:0] outs);

    assign outs = ins;
endmodule
```

I don't think the issue is with `PL_RESIZER_BUFFER_INPUT_PORTS/PL_RESIZER_BUFFER_OUTPUT_PORTS` (ie `buffer_ports`).

I think the issue is that yosys/abc adds a buffer to each of these nets, and `remove_buffers` doesn't remove them:

```
        // Do not remove buffers connected to input/output ports
        // because verilog netlists use the net name for the port.
        && !bufferBetweenPorts(buffer)) {
```

closed 2022-10-18 by mattvenn