The-OpenROAD-Project/OpenLane #1454: Generate congestion report

maliberty, 2022-10-27:
### Prompt
global_route now has a  -congestion_report_file option.  This will generate a report file that shows where the congestion is failing when grt fails.  The resulting file can be loaded in the drc viewer in the OR gui.  See https://github.com/The-OpenROAD-Project/OpenROAD-flow-scripts/blob/2ec8eca9d0a06154111224774a5c7f17fb8b0538/flow/scripts/global_route.tcl#L19 for an example usage.

### Proposal
Generate the report and document it.

vijayank88, 2023-03-28:
@donn 
Any update on this?

It is very useful to designers to find out high congestion hot spots and can be useful to trace the overflow also.

kareefardi, 2023-05-29:
@vijayank88 @maliberty I am looking into adding `-congestion_report_file`. It says load the report file with DRC viewer. I can't find the DRC viewer. Also, how can I programmatically load the report file?

vijayank88, 2023-05-29:
@kareefardi 
Just use `-congestion_report_file` argument part of global routing tcl script and path to store report file.
If there is any congestion, it automatically dumped into report file. User can manually load into OpenROAD GUI to view and analyze the same.

kareefardi, 2023-05-31:
@vijayank88 https://github.com/The-OpenROAD-Project/OpenLane/pull/1824 included `-congestion_report_file`. I noticed that OpenROAD prints a message notifying the user about the congestion report file. There is no need for additional documentation in OpenLane, is there?

vijayank88, 2023-06-01:
> @vijayank88 #1824 included `-congestion_report_file`. I noticed that OpenROAD prints a message notifying the user about the congestion report file. There is no need for additional documentation in OpenLane, is there?

If needed `GRT_CONGESTION_REPORT_FILE` add to flow variable list with description as `Report file to view/analyze in OpenROAD GUI to check congestion during global routing`