The-OpenROAD-Project/OpenLane #1456: ECO Flow fails

vijayank88, 2022-10-28:
## Description

As I'm facing hold violation with one of MPW designs, trying to use `ECO_ENABLE` to fix it. But flow failed to find eco `parasitics_multi_corner_sta` logs.

## Expected behavior

`ECO_ENABLE` should complete the flow.

## Environment

```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: 41ed0036c3dc77961a34005fc7e4494881f0a07f
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

41ed003 2022-10-25T20:04:52+02:00 Fix get_design_path (#1450) - Liu Yihua -  (HEAD -> master, tag: 2022.10.26, origin/master, origin/HEAD)
63a6d02 2022-10-25T10:36:13+02:00 Fix Synthesis Exploration (#1437) - vijayan -  ()
9197534 2022-10-25T09:36:14+02:00 fix usage command in docs (#1449) - Liu Yihua -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)
```

## Reproduction Material

Update `designs/spm/config.json` with `ECO_ENABLE` true.
```
 "ECO_ENABLE": true,
```

1. make mount
2. ./flow.tcl -design spm

## Logs

```
[INFO]: [ECO: 0] Checking output...
no files matched glob pattern "*parasitics_multi_corner_sta*log"
    while executing
"glob -directory $::env(signoff_logs) *parasitics_multi_corner_sta*log"
    (procedure "eco_gen_buffer" line 3)
    invoked from within
"eco_gen_buffer"
    (procedure "eco_output_check" line 4)
    invoked from within
"eco_output_check"
    (procedure "run_eco_flow" line 25)
    invoked from within
"run_eco_flow"
    (procedure "run_eco_step" line 3)
    invoked from within
"[lindex $step_exe 0] [lindex $step_exe 1] "
    (procedure "run_non_interactive_mode" line 52)
    invoked from within
"run_non_interactive_mode {*}$argv"
    invoked from within
"if { [info exists flags_map(-interactive)] || [info exists flags_map(-it)] } {
    if { [info exists arg_values(-file)] } {
        run_file [file nor..."
    (file "./flow.tcl" line 411)

```

aliazak6, 2022-11-03:
I am facing the same issue.

donn, 2022-11-09:
@msaligane Are there any plans to continue maintaining the ECO flow?

aliazak6, 2022-11-10:
set files [glob -directory $::env(signoff_logs) * rcx_mcsta * log]
set lines [split [cat "$::env(routing_tmpfiles)/eco_fix.tcl"] "\n"]
set ::env(ECO_FIX_FILE) $::env(routing_tmpfiles)/eco_fix.tcl

changing eco.tcl file lines[ 59,81 and 98] solves the issue. But eco_fix.tcl doesnt include any violations even if you have them so eco flow still does not work.

PR #1685 (2023-03-21): Exit Cleanup
~ Move `QUIT_ON` variables outside of checker functions
~ Move final timing checks to a flow step: `run_timing_check_step`
~ Replace a couple of loose `file exists` with `assert_files_exist` calls
~ Replace all `flow_fail` across the flow with `throw_error`, which behaves accordingly:
    * If running an interactive script, `flow_fail` is called
    * Else, the errors are propagated upwards, where `flow.tcl` is to catch it
~ Rename `try_catch` to `try_exec` as it now propagates the error, with translation behavior
- Remove UMich experimental ECO flow
---
Resolves #1079
Resolves #1 [... cut]
closed 2023-03-21 by donn