The-OpenROAD-Project/OpenLane #1458: run_designs not supporting JSON configuration file

yihuajack, 2022-10-30:
## Description

Run
```bash
python3 run_designs.py spm
```
as instructed in https://openlane.readthedocs.io/en/latest/usage/exploration_script.html
it failed with errors (see Log)

## Expected behavior

In Line 338 of `run_designs.py`, it wants to call `gen_base_config` method in `config.py`, and in that function, it tries to `copyfile` from `config_file` to `base_config_file`. However, the parameter `base_config_file` is the argument `base_config_path`, which is `base_path + "base_config.tcl"`. The code tries to copy `config.tcl` to `base_config.tcl`. However, for designs using JSON configuration files, there is only `config.json`, no `config.tcl`, so the errors occur.

## Environment

```
YOUR SURVEY HERE
```
> Kernel: Linux v5.10.102.1-microsoft-standard-WSL2
Distribution: ubuntu 22.04
Python: v3.10.6 (OK)
Container Engine: docker v20.10.20 (OK)
OpenLane Git Version: 41ed0036c3dc77961a34005fc7e4494881f0a07f
pip: INSTALLED
python-venv: INSTALLED

> PDK Version Verification Status: OK

> Git Log (Last 3 Commits)

> 41ed003 2022-10-25T20:04:52+02:00 Fix get_design_path (#1450) - Liu Yihua -  (HEAD -> master, tag: 2022.10.26, origin/master, origin/HEAD)
63a6d02 2022-10-25T10:36:13+02:00 Fix Synthesis Exploration (#1437) - vijayan -  ()
9197534 2022-10-25T09:36:14+02:00 fix usage command in docs (#1449) - Liu Yihua -  ()

> Git Remotes

> origin  https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin  https://github.com/The-OpenROAD-Project/OpenLane.git (push)

## Reproduction Material

## Logs

```
YOUR LOGS HERE
```
> [2022-10-30 11:04:21,783 -  INFO] [START] spm
[2022-10-30 11:04:21,897 - ERROR] [FAIL ] spm                 : Check designs/spm/runs/config_regression/openlane.log
[2022-10-30 11:04:21,897 -  INFO] [DONE ] spm                 : Writing report...
grep: /mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/config.tcl: No such file or directory
grep: /mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/config.tcl: No such file or directory
grep: /mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/config.tcl: No such file or directory
grep: /mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/config.tcl: No such file or directory
grep: /mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/config.tcl: No such file or directory
grep: /mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/config.tcl: No such file or directory
grep: /mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/config.tcl: No such file or directory
grep: /mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/config.tcl: No such file or directory
grep: /mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/config.tcl: No such file or directory
grep: /mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/config.tcl: No such file or directory
grep: /mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/config.tcl: No such file or directory
Failed to extract runtime info for spm/config_regression: [Errno 2] No such file or directory: '/mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/runtime.yaml'
Exception in thread Thread-1 (run_design):
Traceback (most recent call last):
  File "/usr/lib/python3.10/threading.py", line 1016, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.10/threading.py", line 953, in run
    self._target(*self._args, **self._kwargs)
  File "/mnt/d/Resources/OpenLane/run_designs.py", line 271, in run_design
    with open(f"{run_path}/report.csv", "w") as report_file:
FileNotFoundError: [Errno 2] No such file or directory: '/mnt/d/Resources/OpenLane/designs/spm/runs/config_regression/report.csv'
[2022-10-30 11:04:22,273 -  INFO] Exiting thread 0...
[2022-10-30 11:04:22,273 -  INFO] Getting top results...
Traceback (most recent call last):
  File "/mnt/d/Resources/OpenLane/run_designs.py", line 423, in <module>
    cli()
  File "/home/yihua/.local/lib/python3.10/site-packages/click/core.py", line 1130, in __call__
    return self.main(*args, **kwargs)
  File "/home/yihua/.local/lib/python3.10/site-packages/click/core.py", line 1055, in main
    rv = self.invoke(ctx)
  File "/home/yihua/.local/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/yihua/.local/lib/python3.10/site-packages/click/core.py", line 760, in invoke
    return __callback(*args, **kwargs)
  File "/mnt/d/Resources/OpenLane/run_designs.py", line 395, in cli
    utils.add_computed_statistics(report_file_name + ".csv")
  File "/mnt/d/Resources/OpenLane/scripts/utils/utils.py", line 92, in add_computed_statistics
    row["suggested_clock_frequency"] = 1000 / suggest_clock_period
ZeroDivisionError: float division by zero

mattvenn, 2022-11-23:
run designs is very useful, so please get this supported asap.

mattvenn, 2022-12-20:
any progress on this? very useful tool

donn, 2022-12-28:
Something of a misdiagnosed issue- this is failing for other reasons, not because of the JSON file. I just ran the JSON file fine.

yihuajack, 2023-01-02:
> Something of a misdiagnosed issue- this is failing for other reasons, not because of the JSON file. I just ran the JSON file fine.

Yes, the problem is that `run_designs.py` just cannot handle `config.json`. It only tries to look for `config.tcl`.

donn, 2023-01-02:
@yihuajack This `config.tcl` is a separate file *inside* the run folder and is generated whether a JSON or Tcl file is sourced.

The problem here is that the run was not done at all, so no `config.tcl` was generated.

donn, 2023-01-02:
There is a separate `base_config_file` issue- but that's not what your log is running into

PR #1597 (2023-01-02): `run_designs` fixes
+ Add JSON support to "matrix" mode
~ Rename "regression mode" to "matrix", as it more accurately reflects a combination of variables
~ Replace bizzare "regression script" with a JSON file achieving basically the same thing
~ `run_designs` now clears run paths before it starts
~ `run_designs` no longer attempts to generated reports for runs that fail at startup
~ `verify_versions` no longer attempts to check the PDK if the environment variable PDK is not set
~ `scripts/config/config.py:get_config` -> `get_config_for_run` (more accurate)
~ `scripts/config/tcl.py:read_tcl_env [... cut]
commit e570a6a5 (2023-01-02): `run_designs` fixes (#1597)

+ Add JSON support to "matrix" mode
~ Rename "regression mode" to "matrix", as it more accurately reflects a combination of variables
~ Replace bizzare "regression script" with a JSON file achieving basically the same thing
~ `run_designs` now clears run paths before it starts
~ `run_designs` no longer attempts to generated reports for runs that fail at startup
~ `verify_versions` no longer attempts to check the PDK if the environment variable PDK is not set
~ `scripts/config/config.py:get_config` -> `get_config_for_run` (more accurate)
~ `scripts/config/tcl.py:rea [... cut]
closed 2023-01-02 by kareefardi