The-OpenROAD-Project/OpenLane #1467: Segfault during writing timing model for mutlti-corner

vijayank88, 2022-11-02:
## Description

Its a MPW CI design, facing segfault during step `Running Multi-Corner Static Timing Analysis at the min process corner`

## Expected behavior

Fix segfault error.

## Environment

```
Kernel: Linux v3.10.0-1160.49.1.el7.x86_64
Distribution: centos 7
Python: v3.6.8 (OK)
Container Engine: docker v20.10.16 (OK)
OpenLane Git Version: a0b19fa68b79eff3e29b674106b06d20aab637f1
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a0b19fa 2022-11-01T19:09:25+02:00 Add Wire Length Checker (#1463) - Mohamed Gaber -  (HEAD -> master, tag: 2022.11.02, origin/master, origin/HEAD)
1f923eb 2022-11-01T12:07:33+02:00 [BOT] Update openroad_app (#1435) - Openlane Bot -  ()
41ed003 2022-10-25T20:04:52+02:00 Fix get_design_path (#1450) - Liu Yihua -  (tag: 2022.10.26)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane.git (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane.git (push)

```

## Reproduction Material

Test case attached here: 
[issue_reproducible.zip](https://github.com/The-OpenROAD-Project/OpenLane/files/9920407/issue_reproducible.zip)

## Logs

```
[STEP 21]
[INFO]: Running Multi-Corner Static Timing Analysis at the min process corner (log: logs/signoff/21-rcx_mcsta.min.log)...
[ERROR]: during executing openroad script /openlane/scripts/openroad/sta_multi_corner.tcl
[ERROR]: Log: designs/wb_host/runs/RUN_2022.11.02_05.44.33/logs/signoff/21-rcx_mcsta.min.log
[ERROR]: Last 10 lines:
39# 0x00007FB2539A0F1E in /lib64/libtcl8.5.so
40# Tcl_EvalEx in /lib64/libtcl8.5.so
41# Tcl_Eval in /lib64/libtcl8.5.so
42# sta::sourceTclFile(char const*, bool, bool, Tcl_Interp*) in openroad
43# ord::tclAppInit(Tcl_Interp*) in openroad
44# Tcl_Main in /lib64/libtcl8.5.so
45# main in openroad
46# __libc_start_main in /lib64/libc.so.6
47# 0x0000000000CA05E7 in openroad
child killed: segmentation violation
```

vijayank88, 2022-11-02:
@jjcherry56 
FYI

jjcherry56, 2022-11-04:
fixed in OR commit a237fba34

commit ddcc5c9f (2022-11-09): [BOT] Update openroad_app (#1466)

Allegedly resolves #1467
closed 2022-11-09 by donn