The-OpenROAD-Project/OpenLane #1469: OpenLane only passes when design has been run through yosys

Fraserbc, 2022-11-03:
## Description
I have a design that I'm trying to build using OpenLane, but it only passes once I run the design through a yosys synth first.

The design can be found [here](https://github.com/Fraserbc/tt02-simon/tree/main/src).

When I run Openlane with `./flow.tcl -verbose 2 -overwrite -design /work/src -run_path /work/runs -tag wokwi` on `simon.v` it failed with routing congestion. But if I instead run this yosys script and OpenLane on the resulting `synth.v`, it passes.
```
read_verilog simon.v
hierarchy -top fraserbc_simon
flatten
synth
write_verilog synth.v
```

To me this makes very little sense, as I can't see what this first yosys pass will do that OpenLane doesn't.

## Expected behavior
The design should pass without having to run it through yosys first.

## Environment
The actual flow is being run inside the container, but the script instructed me to run it outside.
```
Kernel: Linux v5.19.9-arch1-1
Distribution: arch 
Python: v3.10.6 (OK)
Container Engine: docker v20.10.21 (OK)
OpenLane Git Version: f9b5781f5ef0bbdf39ab1c2bbd78be8db11b27f2
pip: INSTALLED
pip:venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

f9b5781 2022-07-01T16:04:31+02:00 Fix a bug with `-overwrite` (#1171) - Anton Blanchard -  (grafted, HEAD, tag: 2022.07.02_01.38.08)
```

## Reproduction Material
[design_notpasses.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9927960/design_notpasses.tar.gz)
[design_passes.tar.gz](https://github.com/The-OpenROAD-Project/OpenLane/files/9927961/design_passes.tar.gz)

It's run with the sky130 PDK in the `efabless/openlane:2022.07.02_01.38.08` docker container.

vijayank88, 2022-11-03:
@Fraserbc 
Is that mean you've synthesized netlist and want to use that file directly by skipping yosys stage right?

Fraserbc, 2022-11-03:
@vijayank88 no, I'm running that synthesised netlist through Openlane. I don't want to skip that. But OpenLane only passed when said synthesised netlist is used, if I try and run it on the original verilog, it failed to pass with routing congestion error.

vijayank88, 2022-11-03:
@Fraserbc 
Attach routing stage `issue_reproducible` generated by flow.  We will debug and provide the solution.
Also you're using 3 months old commit. Update OpenLane commit to latest once and try again.

vijayank88, 2022-11-03:
By usign `simon.v` with latest OpenLane flow able to complete the flow.

[image]

closed 2022-11-03 by Fraserbc