The-OpenROAD-Project/OpenLane #1470: Standardcell boundary redefined, flow fails at DRC

w32agobot, 2022-11-03:
## Description
Flow unexpectedly fails with DRC error `Diffusion contact to gate < 0.055um (licon.11)`.  

The boundary boxes in `results/final/gds` are different to the boundary boxes seen in `results/final/mag`. Compareable behavior has been previously mentioned in [Issue #261](https://github.com/google/skywater-pdk/issues/261), it is stated that the `area_sc` identifier should tell the DRC to use different rules in standard cells. I assume the `area_sc` identifier does not cover the whole standard-cell since the bbox has changed, therefore false-positive DRC errors pop up.  

[image]
In the image you can see the smaller bbox and DRC-errors

## Expected behavior
No false-positives in the DRC. Flow not failing with error `Diffusion contact to gate < 0.055um (licon.11)` in standard cells. 

## Environment
[iic-osic-tools](https://github.com/iic-jku/iic-osic-tools) 
OpenLane d375db931edfb03fc42859d590eb3ec799bc6c9e
OpenROAD v2.0-4818-g4174c3ad8
magic 8.3.324

```
python3 ./env.py issue-survey
Kernel: Linux v5.10.16.3-microsoft-standard-WSL2
Distribution: rocky 9.0
Python: v3.9.10 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: d375db931edfb03fc42859d590eb3ec799bc6c9e
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

d375db9 2022-09-15T13:35:19+02:00 [BOT] Update magic (#1318) - Openlane Bot -  (HEAD, tag: 2022.09.16)
37faafe 2022-09-12T14:58:32+02:00 Startup Fixes (#1344) - Mohamed Gaber -  (tag: 2022.09.13)
cbb9e6b 2022-09-12T14:49:15+02:00 Documentation Restructure (#1337) - Mohamed Gaber -  ()
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

## Reproduction Material
```
git clone https://github.com/w32agobot/bbox_issue
cd bbox_issue
flow.tcl -design adc_top -tag doh -overwrite
```

## Logs
flow.tcl:
```
[STEP 44]
[INFO]: Running Magic DRC (log: adc_top/runs/foo/logs/signoff/44-drc.log)...
[INFO]: Converting Magic DRC Violations to Magic Readable Format...
[INFO]: Converting Magic DRC Violations to KLayout XML Database...
[ERROR]: There are violations in the design after Magic DRC.
[ERROR]: Total Number of violations is 37467
[INFO]: Saving current set of views in 'adc_top/runs/foo/results/final'...
[INFO]: Generating final set of reports...
[INFO]: Created manufacturability report at 'adc_top/runs/foo/reports/manufacturability.rpt'.
[INFO]: Created metrics report at 'adc_top/runs/foo/reports/metrics.csv'.
[INFO]: Saving runtime environment...
[ERROR]: Flow failed.
```

logs/signoff/44-drc.log:
```
Reading "sky130_fd_sc_hd__decap_4".
Reading "sky130_fd_sc_hd__buf_4".
Error while reading cell "sky130_fd_sc_hd__buf_4" (byte position 173114): Warning:  Cell sky130_fd_sc_hd__buf_4 boundary was redefined.
Error while reading cell "sky130_fd_sc_hd__buf_4" (byte position 173114): Warning:  Cell sky130_fd_sc_hd__buf_4 boundary was redefined.
Error while reading cell "sky130_fd_sc_hd__buf_4" (byte position 173114): Warning:  Cell sky130_fd_sc_hd__buf_4 boundary was redefined.
Error while reading cell "sky130_fd_sc_hd__buf_4" (byte position 173114): Warning:  Cell sky130_fd_sc_hd__buf_4 boundary was redefined.
Reading "sky130_fd_sc_hd__fill_2".
Reading "sky130_fd_sc_hd__inv_2".
Error while reading cell "sky130_fd_sc_hd__inv_2" (byte position 20984): Warning:  Cell sky130_fd_sc_hd__inv_2 boundary was redefined.
Reading "sky130_fd_sc_hd__inv_4".
Reading "sky130_fd_sc_hd__conb_1".
Reading "sky130_fd_sc_hd__buf_1".
```

logs/signoff/37-gdsii.log
```
Reading "sky130_fd_sc_hd__decap_4".
Warning:  cell sky130_fd_sc_hd__decap_4 already existed before reading GDS!
Reading "sky130_fd_sc_hd__buf_4".
Warning:  cell sky130_fd_sc_hd__buf_4 already existed before reading GDS!
Reading "sky130_fd_sc_hd__and2b_1".
Reading "sky130_fd_sc_hd__dlymetal6s6s_1".
Reading "sky130_fd_sc_hd__fill_2".
```

reports/signoff/drc.rpt:
```
----------------------------------------
Diffusion contact to gate < 0.055um (licon.11)
----------------------------------------
 5.915um 313.570um 5.920um 313.795um
 5.915um 313.095um 5.920um 313.320um
 6.500um 313.740um 6.505um 313.795um
 6.555um 313.570um 6.560um 313.740um
 6.500um 313.265um 6.505um 313.320um
and many more
```

vijayank88, 2022-11-03:
@w32agobot 
With latest OpenLane commit: https://github.com/The-OpenROAD-Project/OpenLane/commit/a0b19fa68b79eff3e29b674106b06d20aab637f1 not facing any DRC errors.

Please update your OpenLane commit and try again.

Logs for reference:
```
DRC style is now "drc(full)"
Loading DRC CIF style.
No errors found.
[INFO]: COUNT: 0
[INFO]: Should be divided by 3 or 4
[INFO]: DRC Checking DONE (/openlane/designs/adc_top/runs/run1/reports/signoff/drc.rpt)
[INFO]: Saving mag view with DRC errors (/openlane/designs/adc_top/runs/run1/results/signoff/adc_top.drc.mag)
[INFO]: Saved
[INFO]: Converting Magic DRC database to various tool-readable formats...
[INFO]: No DRC violations after GDS streaming out.
```

w32agobot, 2022-11-03:
Unfortunately the DRC-errors also occur after pulling iic-osic-tools revision `next` with OpenLane a0b19fa. Thanks for checking, that means it is somehow related to the environment I'm using

``` 
bash-5.1$ python3 ./env.py issue-survey
Kernel: Linux v5.10.16.3-microsoft-standard-WSL2
Distribution: rocky 9.0
Python: v3.9.10 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: a0b19fa68b79eff3e29b674106b06d20aab637f1
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a0b19fa 2022-11-01T19:09:25+02:00 Add Wire Length Checker (#1463) - Mohamed Gaber -  (HEAD, tag: 2022.11.02, origin/master, origin/HEAD, master)
1f923eb 2022-11-01T12:07:33+02:00 [BOT] Update openroad_app (#1435) - Openlane Bot -  ()
41ed003 2022-10-25T20:04:52+02:00 Fix get_design_path (#1450) - Liu Yihua -  (tag: 2022.10.26)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)

```

vijayank88, 2022-11-04:
@w32agobot 
As I shared the `logs` above, not able to reproduce  drc error at my end.
Please share drc stage test case as per openlane command to check it our end.

w32agobot, 2022-11-04:
@vijayank88 flow.tcl must run in the [iic-osic-tools](https://github.com/iic-jku/iic-osic-tools) with `DOCKER_TAG=next` (the latest docker-image is [pushed a month ago](https://hub.docker.com/r/hpretl/iic-osic-tools/tags)). As seen in `logs` above, this is maybe the only way to reproduce the issue. 

I'm closing the issue since I'm in direct contact with the iic-osic-tools devs and we need to check the local environment first

closed 2022-11-04 by w32agobot