The-OpenROAD-Project/OpenLane #1472: `MAGIC_EXT_USE_GDS` if enabled facing LVS issue

w32agobot, 2022-11-05:
## Description
`flow.tcl -design spm` with config `"MAGIC_EXT_USE_GDS": true` fails at LVS with `Top level cell failed pin matching.`

## Expected behavior
LVS with `MAGIC_EXT_USE_GDS` completes without error

## Environment
```
bash-5.1$   python3 ./env.py issue-survey
Kernel: Linux v5.10.16.3-microsoft-standard-WSL2
Distribution: rocky 9.0
Python: v3.9.10 (OK)
Critical Alert: No Docker or Docker-compatible container engine was found.
OpenLane Git Version: a0b19fa68b79eff3e29b674106b06d20aab637f1
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a0b19fa 2022-11-01T19:09:25+02:00 Add Wire Length Checker (#1463) - Mohamed Gaber -  (HEAD, tag: 2022.11.02, origin/master, origin/HEAD, master)
1f923eb 2022-11-01T12:07:33+02:00 [BOT] Update openroad_app (#1435) - Openlane Bot -  ()
41ed003 2022-10-25T20:04:52+02:00 Fix get_design_path (#1450) - Liu Yihua -  (tag: 2022.10.26)
---
Git Remotes

origin	https://github.com/The-OpenROAD-Project/OpenLane (fetch)
origin	https://github.com/The-OpenROAD-Project/OpenLane (push)
```

open_pdks 0059588  
Magic 8.3.331

## Reproduction Material
`openlane/designs/spm`

## Logs
```
results/signoff/spm.gds.spice
.subckt spm VGND VPWR _000_ _001_ _002_ _003_ _004_ _005_ _006_ _007_ _008_ _009_
+ _010_ _011_ _012_ _013_ _014_ _015_ _016_ _017_ _018_ _019_ _020_ _021_ _022_ _023_
+ _024_ _025_ _026_ _027_ _028_ _029_ _030_ _031_ _032_ _033_ _034_ _035_ _036_ _037_
+ _038_ _039_ _040_ _041_ _042_ _043_ _044_ _045_ _046_ _047_ _048_ _049_ _050_ _051_
...
```

Maybe interesting: those "ports" appear as plain text in the final GDS, which is different to a GDS built with an older version
[image]

dineshannayya, 2022-11-09:
@maliberty @vijayank88  I also finding  GDS based LVS check is broken in latest Openlane/OpenRoad Tools set 
with set ::env(MAGIC_EXT_USE_GDS) {1}

<pre><font color="#4E9A06"><b>dinesha@lenovo-i3-10100-07IMB05</b></font>:<font color="#3465A4"><b>~/workarea/efabless/MPW-7/OpenLane</b></font>$ python3 ./env.py issue-survey
Kernel: Linux v5.15.0-50-generic
Distribution: ubuntu 20.04
Python: v3.8.10 (OK)
Container Engine: docker v20.10.17 (OK)
OpenLane Git Version: a0b19fa68b79eff3e29b674106b06d20aab637f1
pip: INSTALLED
python-venv: INSTALLED
---
PDK Version Verification Status: OK
---
Git Log (Last 3 Commits)

a0b19fa 2022-11-01T19:09:25+02:00 Add Wire Length Checker (#1463) - Mohamed Gaber -  (HEAD -&gt; master)
1f923eb 2022-11-01T12:07:33+02:00 [BOT] Update openroad_app (#1435) - Openlane Bot -  ()
41ed003 2022-10-25T20:04:52+02:00 Fix get_design_path (#1450) - Liu Yihua -  ()
---
Git Remotes

origin	https://github.com/efabless/OpenLane.git (fetch)
origin	https://github.com/efabless/OpenLane.git (push)
</pre>

vijayank88, 2022-11-09:
@dineshannayya 
This issue got reproduced in MPW CI for your designs. Currently I disabled `set ::env(MAGIC_EXT_USE_GDS) 0` to pass LVS stage, as already this ticket raised here.

dineshannayya, 2022-11-15:
LVS based on GDS is better tape-out check than LEF based one, Hope there is priority debug on this issue.

dineshannayya, 2022-11-21:
Is there update on this ?

vijayank88, 2022-11-21:
@dineshannayya 
Please attach the test case.
We'll ask Tim to look into this.

dineshannayya, 2022-11-21:
@vijayank88 with latest openlane + OpenRoad Tool set, I see LVS GDS flow is clean

<pre><font color="#C4A000">commit cb59d1f84deb5cedbb5b0a3e3f3b4129a967c988 (</font><font color="#06989A"><b>HEAD -&gt; </b></font><font color="#4E9A06"><b>master</b></font><font color="#C4A000">)</font>
Author: Kareem Farid &lt;kareefardi@users.noreply.github.com&gt;
Date:   Fri Nov 18 18:42:38 2022 +0200

    fix return values in `equally_spaced_sequence` (#1503)

<font color="#C4A000">commit 78ed21795f1bcc46013c9c3a0372140b9254c8ec</font>
Author: Openlane Bot &lt;83099986+openlane-bot@users.noreply.github.com&gt;
Date:   Thu Nov 17 20:29:24 2022 +0200

    [BOT] Update openroad_app (#1487)
    
    Co-authored-by: Donn &lt;me@donn.website&gt;
    
    + Adds or_tools to the OR build process
</pre>

vijayank88, 2022-11-21:
@w32agobot 
Can you confirm is that resolve?

w32agobot, 2022-11-22:
@vijayank88 Yes I can confirm, with latest OpenLane the flow completed without LVS errors

closed 2022-11-23 by w32agobot