The-OpenROAD-Project/OpenLane #1473: How to change the pdk ?

radhe-2022, 2022-11-06:
Can anyone tell help to change the pdk from sky130 to gf180mcu??

maliberty, 2022-11-06:
@proppy

vijayank88, 2022-11-07:
@radhe-2022 
Follow below steps:
```
export PDK_ROOT=<absolute path to GF180 PDK >`
export PDK=gf180mcuC
make mount
```

proppy, 2022-11-07:
Alternatively:
```
export PDK_FAMILY=gf180mcu # set environment variable for volare
export PDK=gf180mcuC # set environment variable for the rest of the toolchain
make
make test
```
should allow volare to resolve and download the new GF180MCU PDK.

See also https://caravel-user-project.readthedocs.io/en/latest/ for instructions on how to set up the caravel_user_project environment to submit a project to the recently announced GF-MPW0 Open MPW shuttle:
https://opensource.googleblog.com/2022/10/announcing-globalfoundries-open-mpw-shuttle-program.html

radhe-2022, 2022-11-07:
What should be the directory of executing all this instructions??

proppy, 2022-11-07:
@radhe-2022 fresh checkout of OpenLane, see https://github.com/The-OpenROAD-Project/OpenLane#installation-the-short-version

radhe-2022, 2022-11-07:
I'm a beginner 
I followed..
launched terminal in home directory
>> git clone https://github.com/The-OpenROAD-Project/OpenLane.git
>> cd OpenLane
>> EXPORT PDK_FAMILY=gf180mcu
>> EXPORT PDK=gf180mcuC
>> sudo make
>> sudo make test

I exported the gf180 pdk but still after make command docker fetches sky130A pdk image

Please help if Im doing any blunder .. 

Thankyou @proppy @vijayank88

vijayank88, 2022-11-07:
@radhe-2022 
Where GF180 pdk installed in your machine?

proppy, 2022-11-07:
> launched terminal in home directory

Can you paste the full command outout?

> EXPORT PDK_FAMILY=gf180mcu
> EXPORT PDK=gf180mcuC

`export` should be in lowercase.

> sudo make
> sudo make test

`sudo` shouldn't be strictly required.

radhe-2022, 2022-11-08:
Done..! Thankyou so much sir @proppy @vijayank88 

Heading towards my first design to get tapeout

closed 2022-11-08 by vijayank88